realtimeradio/souk-firmware #7: PSB missing tones at some frequencies

sr-cdf, 2023-05-19:
The PSB outputs no power for some requested frequencies when using this code:
```
def setfreqpsb(r,freq_hz):
    n_tones=1
    r.output.use_pfs()
    r.set_tone(0, freq_hz/1e6, phase_offset_rads=0)
    shift_stages = int(np.ceil(np.log2(n_tones)))
    shift_schedule = 2**(shift_stages) - 1
    for synth in [r.pfs, r.pfsoffset]:
        synth.set_fftshift(shift_schedule)
    r.sync.arm_sync()
    r.sync.sw_sync() # Not necessary if there is a PPS connected

```    
So far, I've noticed 7 frequencies that dont work.

`[1.86726e+08, 2.41572e+08, 2.47666e+08, 2.72042e+08, 3.87828e+08, 4.42674e+08, 4.48768e+08]`

The output for a good frequency is:

```
In [109]: setfreqpsb(s,448768000.0+2e6)
2023-05-19 10:12:44,477 - souk_mkid_readout.blocks.block:rfsoc4x2 - INFO - chan_select - Setting output 0 to channel 2799
2023-05-19 10:12:44,595 - souk_mkid_readout.blocks.block:rfsoc4x2 - INFO - synth_input_reorder - Setting output 1393 to channel -1
2023-05-19 10:12:44,669 - souk_mkid_readout.blocks.block:rfsoc4x2 - INFO - synth_input_reorder - Setting output 1398 to channel -1
2023-05-19 10:12:44,745 - souk_mkid_readout.blocks.block:rfsoc4x2 - INFO - synth_input_reorder - Setting output 1399 to channel -1
2023-05-19 10:12:44,829 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Using offset filter bank
2023-05-19 10:12:44,829 - souk_mkid_readout.blocks.block:rfsoc4x2 - INFO - synth_offset_input_reorder - Setting output 1399 to channel 0
2023-05-19 10:12:46,100 - souk_mkid_readout.blocks.block:rfsoc4x2 - WARNING - sync - Timed out waiting for sync pulse

```

And for a bad frequency is:

```
In [99]: setfreqpsb(s,448768000.0)
2023-05-19 10:10:48,445 - souk_mkid_readout.blocks.block:rfsoc4x2 - INFO - chan_select - Setting output 0 to channel 2796
2023-05-19 10:10:48,577 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Using centered filter bank
2023-05-19 10:10:48,577 - souk_mkid_readout.blocks.block:rfsoc4x2 - INFO - synth_input_reorder - Setting output 1398 to channel 0
2023-05-19 10:10:49,853 - souk_mkid_readout.blocks.block:rfsoc4x2 - WARNING - sync - Timed out waiting for sync pulse

```

At this point the frequencies seem to be pretty random. I measured the SNR of single tones at a range of frequencies `freqs = np.linspace(10e6,MIXFREQ*2-10e6,401)` and some of them came back with no power:

[image]

For example:

[image]

I will have a closer look around some of these bad frequencies now to see whats going on.

sr-cdf, 2023-05-22:
Hey @jack-h,

So, some of these bad frequencies are repeatable and some are not.

The tone powers are generally constant as a function of frequency, but there are regions where the power suddenly cuts off. Those regions appear to be 600kHz wide. 

I honed right in on one of the cut-off frequencies and found the power was fine up to precisely 185,100,000.0 MHz, but at 1Hz higher it was gone.

Here is a bad tone at 185.700 MHz:
[image]

Here is a good tone at 185.702MHz:
[image]

Here is the tone power and noise as a function of frequency in some bad cases:

[image]
[image]
[image]

And in good cases:

[image]
[image]

I guess the next thing to test is exactly which PSB channels these dead sections correspond to and see if there is there any rhyme or reason to it.

jack-h, 2023-05-22:
I just did this:

```
In [11]: r.initialize(); r.sync.arm_sync();
2023-05-22 14:37:05,383 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Programming with /home/jackh/src/souk-firmware/firmware/src/souk_single_pipeline_4x2/outputs/souk_single_pipeline_4x2_2023-05-06_1316.fpg
2023-05-22 14:44:36,860 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Initializing block (writable): fpga
2023-05-22 14:44:36,860 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Initializing block (writable): rfdc
2023-05-22 14:44:40,735 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Initializing block (writable): sync
2023-05-22 14:44:40,745 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Initializing block (writable): input
2023-05-22 14:44:40,749 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Initializing block (writable): pfb
2023-05-22 14:44:40,754 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Initializing block (writable): pfbtvg
2023-05-22 14:44:40,855 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Initializing block (writable): chanselect
2023-05-22 14:44:41,029 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Initializing block (writable): zoomfft
2023-05-22 14:44:41,036 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Initializing block (writable): zoomacc
2023-05-22 14:44:41,038 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Initializing block (writable): mixer
2023-05-22 14:44:51,925 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Initializing block (writable): pfs_chanselect
2023-05-22 14:44:52,057 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Initializing block (writable): pfs_offset_chanselect
2023-05-22 14:44:52,191 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Initializing block (writable): pfs
2023-05-22 14:44:52,199 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Initializing block (writable): pfsoffset
2023-05-22 14:44:52,206 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Initializing block (writable): autocorr
2023-05-22 14:44:52,208 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Initializing block (writable): accumulator0
2023-05-22 14:44:52,210 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Initializing block (writable): accumulator1
2023-05-22 14:44:52,212 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Initializing block (writable): output
2023-05-22 14:44:52,214 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Initializing block (writable): gen_cordic
2023-05-22 14:44:52,262 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Initializing block (writable): gen_lut
2023-05-22 14:44:53,748 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Performing software global reset
2023-05-22 14:44:54,958 - souk_mkid_readout.blocks.block:rfsoc4x2 - WARNING - sync - Timed out waiting for sync pulse
2023-05-22 14:44:56,581 - souk_mkid_readout.blocks.block:rfsoc4x2 - WARNING - sync - Timed out waiting for sync pulse

In [12]: def setfreqpsb(r,freq_hz):
    ...:     n_tones=1
    ...:     r.output.use_pfs()
    ...:     r.set_tone(0, freq_hz, phase_offset_rads=0)
    ...:     shift_stages = int(np.ceil(np.log2(n_tones)))
    ...:     shift_schedule = 2**(shift_stages) - 1
    ...:     for synth in [r.pfs, r.pfsoffset]:
    ...:         synth.set_fftshift(shift_schedule)
    ...:     r.sync.arm_sync()
    ...:     r.sync.sw_sync() # Not necessary if there is a PPS connected
    ...: 

In [13]: setfreqpsb(r, 185700000)
2023-05-22 14:45:11,123 - souk_mkid_readout.blocks.block:rfsoc4x2 - INFO - chan_select - Setting output 0 to channel 2357
2023-05-22 14:45:11,488 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2 - INFO - Using offset filter bank
2023-05-22 14:45:11,488 - souk_mkid_readout.blocks.block:rfsoc4x2 - INFO - synth_offset_input_reorder - Setting output 1178 to channel 0
2023-05-22 14:45:12,859 - souk_mkid_readout.blocks.block:rfsoc4x2 - WARNING - sync - Timed out waiting for sync pulse
```
(Note I'm working from the latest code which takes freqs in Hz)

And I see a clear tone on DAC_B, but at 185.4 MHz. If I tweak to 185.702, I see a tone at 185.6 MHz. (It's not impossible that my spectrum analyzer is a little off)

Can you confirm that you're using firmware `souk_single_pipeline_4x2_2023-05-06_1316.fpg`  and the latest software?

jack-h, 2023-05-22:
Scratch that -- after this bugfix https://github.com/realtimeradio/souk-firmware/commit/2939db08c3ac713e088ef41a14c3c79d85466680 -- I get good results for both 185.7 MHz and 185.702 MHz

jack-h, 2023-05-22:
PS -- if you are doing your sweeps using the `setfreqpsb` method you included above, I suggest `wait=False` on the `arm_sync` call, to save an unnecessary wait for a nonexistent PPS to pass

sr-cdf, 2023-05-23:
Thanks Jack,

I cannot reproduce this either, now.

Below are plots of SNR of tones placed exactly at PSB bin-centers (to see which bins were playing up). In some places, every other bin was off,  in repetitions of 8.

I repeated again this morning after hard power cycling the board and as you can see  there are no missing tones.

FPG file is `souk_single_pipeline_4x2_2023-05-06_1316.fpg`
SW version is `0.0.1-915e8acd-dirty`

The board had been running for maybe a week without being reset or reprogrammed.

What could have been going on?

PS: Thanks for the wait=False hint, saved me a few thousand seconds there!

Before reset:

[image]

Zoom in:

[image]

After reset:

[image]

jack-h, 2023-05-23:
If it really was running the same firmware before, I don't really understand what could have gone wrong here.
I suppose it is possible that with various playing with things something got written to a register which it shouldn't have and wasn't being correctly reset. If this ever happens again, it would be good to know if issuing a top-level `initialize` call fixes things. That would at least implicate the configuration software rather than something in firmware/hardware

jack-h, 2023-05-23:
> I will let you know if it happens again and if initialise fixes it.
>
> Is it possible that something screwed up when I sent a keyboard interrupt?
> Like when it was taking a while to set 1000 tones. Maybe something got
> stopped at a bad time?

Not impossible, but the failure modes I can imagine would be that a tone
you don't want doesn't get removed from transmission.