realtimeradio/souk-firmware #9: Sync to frequency reference?

sr-cdf, 2023-05-23:
Hey @jack-h,

Is there any way to lock the board's clock to an external frequency reference eg 10MHz?

What do the SYNC_IN and CLK_IN ports do?

Or could there be an option to set a DAC as a 10MHz output so that other devices could be locked to the board?

jack-h, 2023-05-23:
The CLK_IN should be able to achieve this, but possibly I need to compile
some new software for the on-board PLL to use this input rather than the
integrated oscillator.
Relatedly, the PPS_IN is already set up so that this can be used to trigger
a global reset of firmware on multiple boards

sr-cdf, 2023-05-23:
Ok, I don't know how much of a priority this is right now, except it would mean our spectrum analysers would align with our tones better. But in the long run, we probably do want to lock in to a very stable frequency reference so that we don't confuse local clock frequency drift with KID frequency shift.

Does the PPS firmware reset the whole system or does it just reset some timers/counters for synchronising between boards? If the internal clocks of each board are not locked together they will drift apart over time. In the past, a 1PPS signal has been used to increment a timestamp to precisely align our samples with external telemetry.

jack-h, 2023-05-31:
External lock to 10MHz configured in commit 8336447946b . Note that the default configuration has been changed so that a 10MHz reference is now required.

The PPS will be used in the same manner you describe from the past. As to exactly what else is / isn't reset by the PPS (for example, output tone phases) this could go either way, depending on requirements. Currently synth bank tones _are_ reset by the PPS (or "simulated" PPS via the `sync.sw_sync()` method)

closed 2023-06-01 by jack-h