realtimeradio/souk-firmware #14: Fatal git submodule update

sr-cdf, 2023-06-23:
Is this something to be concerned about after running git pull?

$ git submodule update --init --recursive 
Submodule path 'firmware/lib/mlib_devel': checked out '940060ccb87b74c8b6faa541bf1b5316c4d6a604'
Submodule path 'software/casperfpga': checked out 'd3674b5cb6add3f0cf15e505824b591d2a381458'
fatal: No url found for submodule path 'software/casperfpga/src/tftpy-0.6.3-fork-py2.7.egg' in .gitmodules
fatal: Failed to recurse into submodule path 'software/casperfpga'

jack-h, 2023-06-23:
The TFTP library mentioned here isn't relevant for our purposes (it's used
by some other non-RFSoC FPGA boards for comms). But I'm confused as to
where this error is really coming from. I've never done a `--recursive`
submodule init from the top level, but I don't see anything identifying
that tftp directory as a submodule.
I'll try and figure out what the issue is here, but in the meantime, I
suspect (famous last words) it won't cause any problems

sr-cdf, 2023-06-23:
Oh I thought the instruction was to perform the submodule update from the top level as described in the top level README.md:

To open/modify/compile:
Clone this repository
Clone submodules:
git submodule update --init --recursive

jack-h, 2023-06-23:
Heh, I guess it is. I usually only init the submodules I actually use
(casperfpga on the machine I control from; mlib_devel on the machine I
compile firmware with) so I think the truth you have stumbled upon is that
I've never actually followed my own instructions

jack-h, 2023-06-26:
Fixed in fa6b8fdc592c2

closed 2023-06-26 by jack-h