realtimeradio/souk-firmware #15: Board requires power cycle after update firmware from 5.2 to 5.5

sr-cdf, 2023-06-23:
After pulling the lastest verion and updating all the software, `souk_mkid_readout.SoukMkidReadout() `was failing with:

```
In [1]: import souk_mkid_readout

In [2]: s=souk_mkid_readout.SoukMkidReadout('rfsoc4x2',configfile='/home/sam/souk/souk-firmware/software/control_sw/config/souk-single-pipeline-4x2-sam-auto.yaml')
2023-06-23 11:27:31,204 - souk_mkid_readout.blocks.block:rfsoc4x2:p0_cordic_gen - ERROR - Tried to read register block_info which doesn't exist!
2023-06-23 11:27:32,230 - souk_mkid_readout.blocks.block:rfsoc4x2:p0_lut_gen - ERROR - Tried to read register block_info which doesn't exist!
```

After power cycling the board it ran fine.

It's as if the software knew the board was already programmed and was trying to read a register that was not present in the previous version.

There was no way to program the latest bit file because the fpga interface was not being created. 

After the power cycle, the software must not have been able to see the fpga was not programmed and continued as expected.

jack-h, 2023-06-23:
Your astute analysis is correct. I've added some semblance of firmware
version support checking, and I'll try and figure out how to better handle
this.

jack-h, 2023-06-26:
Are you sure the `SoukMkidReadout` interface actually failed to create? Despite the error messages, it should have been constructed to the point that you could run `SoukMkidReadout.program()` to load the new firmware.
In any case, commit 6fb66366446 should be more helpful in surpressing unhelpful issues and highlighting the actual problem

sr-cdf, 2023-06-26:
> *External email to Cardiff University - *Take care when replying/opening
> attachments or links.
> *Nid ebost mewnol o Brifysgol Caerdydd yw hwn - *Cymerwch ofal wrth
> ateb/agor atodiadau neu ddolenni.

I may have assumed that the interface was not created, without actually
testing if it was there.
Thanks anyway for the fix.

closed 2023-06-26 by jack-h