realtimeradio/souk-firmware #16: Tones are working without 10MHz reference connected

sr-cdf, 2023-06-23:
Nothing connected to CLK_IN but tones are visible on spectrum analyser anyway.

LUT, CORDIC and PSB all working.

LMK file "`rfsoc4x2_PL_122M88_REF_245M76_10M_EXTREF.txt`" was copied to the 4x2 with scp. Had to copy to home directory first then sudo copy to `/lib/firmware` due to permissions.

SoukMkidReadout configfile also updated to specify the correct LMK file.

Status report indicates the correct LMK file is in use: 

```
In [24]: s.rfdc.print_status()
ADC0_enabled: True
ADC0_pll: 1
ADC0_state: 15
ADC1_enabled: True
ADC1_pll: 1
ADC1_state: 12
ADC2_enabled: True
ADC2_pll: 1
ADC2_state: 15
ADC3_enabled: True
ADC3_pll: 1
ADC3_state: 12
DAC0_enabled: True
DAC0_pll: 1
DAC0_state: 15
DAC1_enabled: True
DAC1_pll: 1
DAC1_state: 15
DAC2_enabled: False
DAC3_enabled: False
lmkfile: rfsoc4x2_PL_122M88_REF_245M76_10M_EXTREF.txt
lmxfile: rfsoc4x2_LMX_REF_245M76_OUT_491M52.txt

In [25]: s.fpga.print_status()
antname: None
fw_build_time: 2023-06-05T15:04:00
fw_supported: True
fw_type: 2
fw_version: 5.5.0.0
host: astrog119:rfsoc4x2
programmed: True
sw_version: 0.0.1-335a41bf
timestamp: 2023-06-23T12:59:45.237595
```

Should it be working?

sr-cdf, 2023-06-23:
Connecting the REF OUT of my spectrum analyser (sinusoidal 10 MHz 1 V_PP 50 Ohm) to the CLK IN and trying again from a powered off state hasn;t made any difference. I would have expected the frequency of a generated tone appear on the spectrum analyser with no offset. This was not the case.

Is 1Vpp high enough at the CLK_IN port?

sr-cdf, 2023-06-23:
My bad, it did work after the power cycle, i was looking at the wrong number.

So I guess 1Vpp is enough,

But still not sure why it works with no clock connected, does it fall back to internal clock if no external source is present?

sr-cdf, 2023-06-23:
FYI, unplugging the CLK_IN while its still running doesnt crash the system, it must fall back to the internal clock, because the tone just hops back to where it was without the clocks locked together.

sr-cdf, 2023-06-23:
Think I just answered my own issue there, the system falls back to the internal clock if no external clock is present. Happy days.

closed 2023-06-23 by sr-cdf

jack-h, 2023-06-23:
I'm actually pleasantly surprised :)