realtimeradio/souk-firmware #17: Problem with SoukMkidReadout.set_multi_tone()

sr-cdf, 2023-06-26:
Issue when setting multiple tones:

```
Request: ?read p0_mix_lo2_scale 0 0     
Reply: !read fail
```

Heres the ipython session:

```
In [1]: import souk_mkid_readout
   ...: 
   ...: s=souk_mkid_readout.SoukMkidReadout('rfsoc4x2',configfile='/home/sam/souk/souk
   ...: -firmware/software/control_sw/config/souk-single-pipeline-4x2-sam-auto.yaml')
   ...: 
   ...: if not s.fpga.is_programmed():
   ...:     s.program()
   ...: s.initialize()
   ...: 
2023-06-26 14:42:46,272 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): fpga
2023-06-26 14:42:46,272 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): rfdc
2023-06-26 14:42:50,111 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): sync
2023-06-26 14:42:50,119 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): input
2023-06-26 14:42:50,120 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): pfb
2023-06-26 14:42:50,124 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): pfbtvg
2023-06-26 14:42:50,201 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): chanselect
2023-06-26 14:42:50,312 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): zoomfft
2023-06-26 14:42:50,319 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): zoomacc
2023-06-26 14:42:50,320 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): mixer
2023-06-26 14:42:50,489 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): psb_chanselect
2023-06-26 14:42:50,595 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): psb_offset_chanselect
2023-06-26 14:42:50,724 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): psb
2023-06-26 14:42:50,727 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): psboffset
2023-06-26 14:42:50,730 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): autocorr
2023-06-26 14:42:50,732 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): accumulator0
2023-06-26 14:42:50,750 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): accumulator1
2023-06-26 14:42:50,774 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): output
2023-06-26 14:42:50,777 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): gen_cordic
2023-06-26 14:42:50,814 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): gen_lut
2023-06-26 14:42:51,928 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Performing software global reset
2023-06-26 14:42:53,133 - souk_mkid_readout.blocks.block:rfsoc4x2:p0_sync - WARNING - Timed out waiting for sync pulse

In [2]: s.fpga.print_status()

antname: None
fw_build_time: 2023-06-05T15:04:00
fw_supported: True
fw_type: 2
fw_version: 5.5.0.0
host: astrog119:rfsoc4x2
programmed: True
sw_version: 0.0.1-335a41bf
timestamp: 2023-06-26T14:42:55.385793

In [3]: freqs_hz= [ 10e6, 15e6]

In [4]: s.set_multi_tone(freqs_hz)
---------------------------------------------------------------------------
KatcpRequestFail                          Traceback (most recent call last)
Cell In[4], line 1
----> 1 s.set_multi_tone(freqs_hz)

File ~/souk/env/lib/python3.8/site-packages/souk_mkid_readout/souk_mkid_readout.py:469, in SoukMkidReadout.set_multi_tone(self, freqs_hz, phase_offsets_rads, amplitudes)
    467 self.chanselect.set_channel_outmap(chanmap_in)
    468 # Write mixer tones
--> 469 self.mixer.set_freqs(lo_freqs_hz, phase_offsets_rads, amplitudes, self.adc_clk_hz)
    470 # Write output maps
    471 self.psb_chanselect.set_channel_outmap(chanmap_psb)

File ~/souk/env/lib/python3.8/site-packages/souk_mkid_readout/blocks/mixer.py:273, in Mixer.set_freqs(self, freqs_hz, phase_offsets, scaling, sample_rate_hz)
    271 for i in range(self._n_parallel_chans):
    272     regprefix = f'lo{i}'
--> 273     self.write(regprefix + '_scale', scaling[i::self._n_parallel_chans].tobytes())
    274     self.write(regprefix + '_phase_inc', phase_steps[i::self._n_parallel_chans].tobytes())
    275     self.write(regprefix + '_phase_offset', phase_offsets[i::self._n_parallel_chans].tobytes())

File ~/souk/env/lib/python3.8/site-packages/souk_mkid_readout/blocks/block.py:194, in Block.write(self, reg, val, offset, **kwargs)
    187 """
    188 A simple wrapper around CasperFpga.write(), which modifies the
    189 register name to reflect the name of this block in the system
    190 hierarchy. Also add exception handling to skip writes which fail
    191 because a register doesn't exist.
    192 """
    193 try:
--> 194     self.host.write(self.prefix + reg, val, offset=offset, **kwargs)
    195 except:
    196     if reg not in self.listdev():

File ~/souk/env/lib/python3.8/site-packages/casperfpga-0.4.4.dev1349+head.d3674b5-py3.8-linux-x86_64.egg/casperfpga/casperfpga.py:537, in CasperFpga.write(self, device_name, data, offset)
    529 """
    530 Write data, then read it to confirm a successful write.
    531 
   (...)
    534 :param offset: offset at which to write, in bytes
    535 """
    536 self.blindwrite(device_name, data, offset)
--> 537 new_data = self.read(device_name, len(data), offset)
    538 if new_data != data:
    539     # TODO - this error message won't show you the problem if
    540     # it's not in the first word
    541     unpacked_wrdata = struct.unpack('>L', data[0:4])[0]

File ~/souk/env/lib/python3.8/site-packages/casperfpga-0.4.4.dev1349+head.d3674b5-py3.8-linux-x86_64.egg/casperfpga/casperfpga.py:277, in CasperFpga.read(self, device_name, size, offset, **kwargs)
    268 def read(self, device_name, size, offset=0, **kwargs):
    269     """
    270     Read size-bytes of binary data with carriage-return escape-sequenced.
    271 
   (...)
    275     :param kwargs:
    276     """
--> 277     data = self.transport.read(device_name, size, offset, **kwargs)
    278     if self.is_little_endian:
    279         assert ((len(data) % 4) == 0), \
    280             "Can only read multiples of 4 bytes because CasperFpga is doing an endianness flip"

File ~/souk/env/lib/python3.8/site-packages/casperfpga-0.4.4.dev1349+head.d3674b5-py3.8-linux-x86_64.egg/casperfpga/transport_katcp.py:395, in KatcpTransport.read(self, device_name, size, offset)
    386 def read(self, device_name, size, offset=0):
    387     """
    388     Read size-bytes of binary data with carriage-return escape-sequenced.
    389    
   (...)
    393     :return: binary data string
    394     """
--> 395     reply, _ = self.katcprequest(
    396         name='read', request_timeout=self._timeout, require_ok=True,
    397         request_args=(device_name, str(offset), str(size)))
    398     return reply.arguments[1]

File ~/souk/env/lib/python3.8/site-packages/casperfpga-0.4.4.dev1349+head.d3674b5-py3.8-linux-x86_64.egg/casperfpga/transport_katcp.py:292, in KatcpTransport.katcprequest(self, name, request_timeout, require_ok, request_args, max_retries, retry_delay_s)
    290 if (reply.arguments[0] != katcp.Message.OK) and require_ok:
    291     if reply.arguments[0] == katcp.Message.FAIL:
--> 292         raise KatcpRequestFail(
    293             'Request %s on host %s failed.\n\t'
    294             'Request: %s\n\tReply: %s' %
    295             (request.name, self.host, request, reply))
    296     elif reply.arguments[0] == katcp.Message.INVALID:
    297         raise KatcpRequestInvalid(
    298             'Invalid katcp request %s on host %s.\n\t'
    299             'Request: %s\n\tReply: %s' %
    300             (request.name, self.host, request, reply))

KatcpRequestFail: Request read on host rfsoc4x2 failed.
        Request: ?read p0_mix_lo2_scale 0 0
        Reply: !read fail

```

jack-h, 2023-06-26:
This is strange. The 100tone_test.py script (in `control_sw/test_scripts`)
uses the same `set_multi_tone` method without issue.
Investigating now

jack-h, 2023-06-26:
Bug affects trying to load < 8 tones in one go. Fixed in 853f7b29c533