realtimeradio/souk-firmware #18: Why is accumulator.get_new_spectra() always equal to zero up until element 32?

sr-cdf, 2023-06-27:
Random question about reading out from the accumulators...

Trying this:

```
import numpy as np

import souk_mkid_readout

r=souk_mkid_readout.SoukMkidReadout('rfsoc4x2',configfile='/home/sam/souk/souk-firmware/software/control_sw/config/souk-single-pipeline-4x2.yaml')

if not r.fpga.is_programmed():
    r.program()
r.initialize()

r.fpga.print_status()

r.output.use_psb()

freqs_hz= [ 10e6, 15e6]

r.set_multi_tone(freqs_hz)

a1,a2=r.accumulators

print(np.nonzero(a1.get_new_spectra()))

```

Gives `(array([32, 33]),)`.

Why doesn't it start from zero?

Full session:
```
Python 3.8.16 (default, Dec  7 2022, 01:12:06) 
Type 'copyright', 'credits' or 'license' for more information
IPython 8.12.2 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import numpy as np
   ...: 
   ...: import souk_mkid_readout
   ...: 
   ...: r=souk_mkid_readout.SoukMkidReadout('rfsoc4x2',configfile='/home/sam/souk/souk-firmware/software/contro
   ...: l_sw/config/souk-single-pipeline-4x2.yaml')
   ...: 
   ...: if not r.fpga.is_programmed():
   ...:     r.program()
   ...: r.initialize()
   ...: 
   ...: r.fpga.print_status()
   ...: 
   ...: r.output.use_psb()
   ...: 
   ...: freqs_hz= [ 10e6, 15e6]
   ...: 
   ...: r.set_multi_tone(freqs_hz)
   ...: 
   ...: a1,a2=r.accumulators
   ...: 
   ...: print(np.nonzero(a1.get_new_spectra()))
   ...: 
2023-06-27 16:57:26,766 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): fpga
2023-06-27 16:57:26,766 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): rfdc
2023-06-27 16:57:30,623 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): sync
2023-06-27 16:57:30,639 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): input
2023-06-27 16:57:30,642 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): pfb
2023-06-27 16:57:30,647 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): pfbtvg
2023-06-27 16:57:30,712 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): chanselect
2023-06-27 16:57:30,819 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): zoomfft
2023-06-27 16:57:30,823 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): zoomacc
2023-06-27 16:57:30,824 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): mixer
2023-06-27 16:57:30,949 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): psb_chanselect
2023-06-27 16:57:31,014 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): psb_offset_chanselect
2023-06-27 16:57:31,083 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): psb
2023-06-27 16:57:31,086 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): psboffset
2023-06-27 16:57:31,089 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): autocorr
2023-06-27 16:57:31,090 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): accumulator0
2023-06-27 16:57:31,105 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): accumulator1
2023-06-27 16:57:31,121 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): output
2023-06-27 16:57:31,122 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): gen_cordic
2023-06-27 16:57:31,143 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Initializing block (writable): gen_lut
2023-06-27 16:57:32,218 - souk_mkid_readout.souk_mkid_readout:rfsoc4x2:0 - INFO - Performing software global reset
2023-06-27 16:57:33,423 - souk_mkid_readout.blocks.block:rfsoc4x2:p0_sync - WARNING - Timed out waiting for sync pulse
antname: None
fw_build_time: 2023-06-05T15:04:00
fw_supported: True
fw_type: 2
fw_version: 5.5.0.0
host: astrog119:rfsoc4x2
programmed: True
sw_version: 0.0.1-853f7b29
timestamp: 2023-06-27T16:57:33.837713
2023-06-27 16:57:34,271 - souk_mkid_readout.blocks.block:rfsoc4x2:p0_acc0 - WARNING - Accumulation counter changed while reading data!
(array([32, 33]),)
```

```

jack-h, 2023-06-27:
That would be consistent with a firmware bug yielding an off-by-one ram
addressing error.
I've fixed some bugs in this area of the code relatively recently, but
perhaps they didn't make it into rev 5.5. I'll investigate.
Probably the data are correct if you offset the accumulator indexing by 32.

jack-h, 2023-06-30:
Not yet tested, but possibly/hopefully fixed in firmware [souk_single_pipeline_4x2_2023-06-30_1710.fpg](https://github.com/realtimeradio/souk-firmware/blob/main/firmware/src/souk_single_pipeline_4x2/outputs/souk_single_pipeline_4x2_2023-06-30_1710.fpg)

sr-cdf, 2023-07-06:
Tested and seems to be fixed.

```
In [6]: np.nonzero(r.accumulators[0].get_new_spectra())
2023-07-06 15:33:24,706 - souk_mkid_readout.blocks.block:rfsoc4x2:p0_acc0 - WARNING - Accumulation counter changed while reading data!

Out[6]: (array([0]),)

```

And the values returned now have average magnitude much greater than zero when connected in loopback (and also with `input.enable_loopback`) , suggesting the correct channel is now being delivered.

Nice one :)

jack-h, 2023-07-06:
Hurrah!

closed 2023-07-06 by jack-h