realtimeradio/souk-firmware #19: Mixer output is at 2*f_input, not DC.

sr-cdf, 2023-07-10:
Hey @jack-h 

Upon inspecting the accumulator outputs for a single tone with internal loopback enabled and 'mixer power mode' off, it appears the accumulated signal always contains a component at twice the frequency of the input signal (ie 2x the cordic LO), unless the accumulation just happens to alias this down to DC.

Looking at the mixer block in simulink, it seems this is present because the mixer is multiplying the LO signal ( A * ( cos(wt) + j sin(wt) ) ) directly with the PFB output ( B * ( cos(wt + phi) + j sin(wt + phi) ) ), giving LO * PSB = 2AB ( cos (2wt + phi) + j sin(2wt + phi) ).

One of the terms need to be complex conjugated to remove this component and leave just a DC term, eg, LO*conj(PSB) = AB*exp(j phi), or conj(LO) * PSB  = AB*conj(exp(j phi))

Can you add a complex conjugation to a mixer input of your choice?

jack-h, 2023-07-28:
Addressed in 5.5.1

jack-h, 2023-07-31:
tested OK in `5.5.4.0`