realtimeradio/souk-firmware #21: Increase LO scale factor number of bits

sr-cdf, 2023-07-21:
The scale factor is currently 8-bits, which gives quite coarse steps at low scales.

12-bit would be better, 16-bit is probably overkill.

jack-h, 2023-07-28:
rev 5.5.3 uses 12 bits

jack-h, 2023-07-31:
@sr-cdf I've just realised I probably did the wrong thing here -- Am I right in thinking you wanted 12 bits but with 4 fractional bits. I.e., keep the 1/256 maximum attenuation, but set the step size to

closed 2023-07-31 by jack-h

sr-cdf, 2023-07-31:
> *External email to Cardiff University - *Take care when replying/opening
> attachments or links.
> *Nid ebost mewnol o Brifysgol Caerdydd yw hwn - *Cymerwch ofal wrth
> ateb/agor atodiadau neu ddolenni.
>
> @sr-cdf
> <https://eur03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fsr-cdf&data=05%7C01%7Csam.rowe%40astro.cf.ac.uk%7C8ad4bca3875e487d481f08db91a8b0b0%7Cbdb74b3095684856bdbf06759778fcbc%7C1%7C0%7C638263925919526162%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=4t99%2B%2FrxiFQnjZ01cOB0S10LzY0UGnACsATnFGM6yMc%3D&reserved=0>
> I've just realised I probably did the wrong thing here -- Am I right in
> thinking you wanted 12 bits but with 4 fractional bits. I.e., keep the
> 1/256 maximum attenuation, but set the step size to

Your message got cut off at the end but I think just 12 total bits going
down from 1 to 1 / 2**12 is fine. Do you mean there is an option to add
additional gain by adjusting the fixed point decimal place? I don't think
that's necessary.

jack-h, 2023-07-31:
Ignore, I misunderstood how the gain factor was being applied :)