realtimeradio/souk-firmware #22: ADC input spectrum appears reversed if not using internal loopback

sr-cdf, 2023-07-24:
Hey @jack-h,

There is an unexpected difference between measurements at the ADC input with the input loopback enabled and disabled.

Using the PSB with F_out=250MHz, so that F_baseband = -978.8MHz with F_DUC = 1228.8 MHz, the tone appears in the right place when the internal input loopback is enabled, but is at the negative frequency when the loopback is disabled and a cable connects the DAC to the ADC.

One explanation is that I and Q are being swapped somewhere, possibly coming out of the rfdc block?

I first noticed this when looking at the accumulator outputs, but have traced it all the way back to the input snapshot.

It does not depend on the generator type.

It may mean the channel select before the mixer is picking the wrong channels.

Plot and python session below:

[image]

```
IPython 8.12.2 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import matplotlib.pyplot as plt; plt.ion()
Installed tk event loop hook.
Out[1]: <contextlib.ExitStack at 0xffffb7175760>

In [2]: import numpy as np

In [3]: import souk_mkid_readout

In [4]: r=souk_mkid_readout.SoukMkidReadout('localhost',configfile='/home/casper/git/souk-firmware/software/control_sw/config/souk-single-
   ...: pipeline-4x2.yaml')

In [5]: r.program(fpgfile='./git/souk-firmware/firmware/src/souk_single_pipeline_4x2/outputs/souk_single_pipeline_4x2_2023-06-30_1710.fpg'
   ...: )
2023-07-24 05:46:50,414 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Programming with ./git/souk-firmware/firmware/src/souk_single_pipeline_4x2/outputs/souk_single_pipeline_4x2_2023-06-30_1710.fpg

In [6]: r.initialize()
2023-07-24 05:46:59,280 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Initializing block (writable): fpga
2023-07-24 05:46:59,280 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Initializing block (writable): rfdc
2023-07-24 05:47:03,400 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Initializing block (writable): sync
2023-07-24 05:47:03,423 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Initializing block (writable): input
2023-07-24 05:47:03,429 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Initializing block (writable): pfb
2023-07-24 05:47:03,438 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Initializing block (writable): pfbtvg
2023-07-24 05:47:03,735 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Initializing block (writable): chanselect
2023-07-24 05:47:04,206 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Initializing block (writable): zoomfft
2023-07-24 05:47:04,218 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Initializing block (writable): zoomacc
2023-07-24 05:47:04,221 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Initializing block (writable): mixer
2023-07-24 05:47:04,580 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Initializing block (writable): psb_chanselect
2023-07-24 05:47:04,972 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Initializing block (writable): psb_offset_chanselect
2023-07-24 05:47:05,364 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Initializing block (writable): psb
2023-07-24 05:47:05,373 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Initializing block (writable): psboffset
2023-07-24 05:47:05,382 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Initializing block (writable): autocorr
2023-07-24 05:47:05,385 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Initializing block (writable): accumulator0
2023-07-24 05:47:05,470 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Initializing block (writable): accumulator1
2023-07-24 05:47:05,514 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Initializing block (writable): output
2023-07-24 05:47:05,518 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Initializing block (writable): gen_cordic
2023-07-24 05:47:05,586 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Initializing block (writable): gen_lut
2023-07-24 05:47:10,349 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Performing software global reset
2023-07-24 05:47:11,561 - souk_mkid_readout.blocks.block:localhost:p0_sync - WARNING - Timed out waiting for sync pulse

In [7]: r.fpga.print_status()
antname: None
fw_build_time: 2023-06-30T11:09:23
fw_supported: True
fw_type: 2
fw_version: 5.5.0.1
host: localhost:localhost
programmed: True
sw_version: 0.0.1-efdf5d20-dirty
timestamp: 2023-07-24T05:47:18.275856

In [8]: def setfreqpsb(r,freq_hz,phase=0,amplitude=1.0):
   ...:     freq_hz=np.atleast_1d(freq_hz)
   ...:     phase=np.atleast_1d(phase)
   ...:     amplitude=np.atleast_1d(amplitude)
   ...:     n_tones=len(freq_hz)
   ...:     r.output.use_psb()
   ...:     r.reset_psb_outputs()
   ...:     r.set_multi_tone(freq_hz, phase, amplitude)
   ...:     shift_stages = int(np.ceil(np.log2(n_tones)))
   ...:     shift_schedule = 2**(shift_stages) - 1
   ...:     for synth in [r.psb, r.psboffset]:
   ...:         synth.set_fftshift(shift_schedule)
   ...:     r.sync.arm_sync(wait=False)
   ...:     r.sync.sw_sync() # Not necessary if there is a PPS connected
   ...:     return freq_hz
   ...: 

In [9]: setfreqpsb(r,250e6)
Out[9]: array([2.5e+08])

In [10]: r.input.enable_loopback()

In [11]: r.input.plot_adc_spectrum(True)

In [12]: r.input.disable_loopback()

In [13]: r.input.plot_adc_spectrum(True)

In [14]: plt.legend(['Firmware loopback looks OK','Hardware loopback appears reversed.'])
Out[14]: <matplotlib.legend.Legend at 0xffffa0e997c0>

In [15]: plt.suptitle('r.input.plot_adc_spectrum() is flipped if not using input loopback\n PSBset freq is 250MHz')
Out[15]: Text(0.5, 0.98, 'r.input.plot_adc_spectrum() is flipped if not using input loopback\n PSBset freq is 250MHz')

In [16]: plt.savefig('input_spectrum_reversed.png')

```

sr-cdf, 2023-07-25:
Flipping the NCO coarse mixer frequency to -Fs/4 does flip the spectrum as seen in the ADC snapshot:

[image]

jack-h, 2023-07-28:
Yes, this makes sense -- the ADC mixer should be in the opposite direction to the DAC mixer. Fixed in v5.5.4

jack-h, 2023-07-31:
tested OK on hardware v5.5.4

closed 2023-07-31 by jack-h