realtimeradio/souk-firmware #24: Demo maximum tone frequency update rate

jack-h, 2023-08-24:
@sr-cdf -- will post updates here

jack-h, 2023-09-07:
In commit 7baa8d2 added simple code to write to the LO phase accumulator step registers on each new accumulation.

Initial observations:

Using the `mixer.set_freqs(...)` method with a local connection (using the local transport which writes straight to `/dev/mem`) it takes ~100ms to write to the phase accumulator step/offset and gain registers.

By pre-computing the memory addresses (in much the same way as the accumulation reading already worked), reading an accumulation and writing phase steps (but not new offsets and gains) takes ~1ms.  Which is to say, over a few runs of 1000 1ms accumulations (`acc_len = 1200`) I see between 0 and 4 cycles taking too long.

This doesn't include:
 1. Any compute to determine what the new phase steps should be
 2. Any handling of a situation where a tone drifts far enough that it needs to be assigned a new bin (this would necessitate reloading all the reorder maps)
 3. Any consideration of changing per-LO gains or offsets (I assume the offsets will never be changed so that phase remains continuous as frequency is updated)
 4. Any acknowledgement of the fact the final system will be dealing with multiple signal paths simultaneously.

I think the bottom line is, if the goal is to meet 1ms loop period probably everything should be written in something pre-compiled (this is traditionally the point where somebody claims there is some python library available to get similar performance). But things seem pretty close even with very simple code, which makes me relatively confident we can get this working without _too much_ pain.

Would be interested in your thoughts @sr-cdf regarding what the most useful thing to do next is, and how the 4 assumptions above marry up with your envisaged requirements.

jack-h, 2023-10-24:
version 6 of the firmware reorganizes the data path so that there are fewer deep rams to read integrations from and write LO frequencies to. With this firmware version, the average loop time to read an accumulation and write new LOs (still with the caveats above) averages, over 10000 loops, 0.61ms per loop. Maximum observed was 0.8ms per loop.

jack-h, 2023-10-24:
Note part 2:

Changing the looping behaviour so that on each loop the (2048) output frequencies are incremented by a small fraction of a bin -- which doesn't really make any sense, but at least adds some small compute -- doesn't materially change the loop time, which now averages 0.66ms. So, I suspect maybe we might get away with pure python after all.

jack-h, 2023-10-29:
With some basic trigonometry, and the calculation `new_frequencies = old_frequencies + Constant*accumulation_phase` I get:

```
(py3venv) casper@localhost:~/src/souk-firmware/software/control_sw/test_scripts$ sudo /home/casper/py3venv/bin/python ./souk_poll_acc.py --nloop 1000 --update-los --fpgfile ~/src/souk-firmware/firmware/src/souk_single_pipeline_krm/outputs/souk_single_pipeline_krm_2023-10-25_1121.fpg
Accumulation time is approximately 1.0 milliseconds
Initializing all LOs to bin centers
Entering loop
Average read time: 0.81 ms
Max read time: 1.11 ms
Average loop time: 1.01 ms
Number of reads: 1000
Number of too slow reads: 8
```

I.e., we're right on the edge of what a dumb python script can do. In my opinion this is close enough to be relatively confident this is doable with optimization, and I suspect it's not worth worrying about this unless/until there is a clear picture about exactly what behaviour the tone tracking should have, and a sense of how often situations will be encountered where a tone "jumps" frequency channels,

sr-cdf, 2023-10-30:
Seems very promising :)

sr-cdf, 2023-10-30:
@jack-h any chance you can compile for the 4x2 so I can give it a go?

jack-h, 2023-10-30:
sure -- doing so now

jack-h, 2023-10-30:
Just pushed --
https://github.com/realtimeradio/souk-firmware/blob/v6/firmware/src/souk_single_pipeline_4x2/outputs/souk_single_pipeline_4x2_2023-10-30_1139.fpg

jack-h, 2024-04-11:
This has probably now got slower since `v7` firmware makes RX and TX LOs independent. Thus twice as many parameters need to be loaded to set them up.

closed 2026-01-08 by jack-h