realtimeradio/souk-firmware #25: Change PFB window function

jack-h, 2023-08-24:
@sr-cdf -- what window spec to you desire?

sr-cdf, 2023-08-24:
try `scipy.signal.windows.dpss(window_len, NW=2.0)`
your plotting script shows it to be flatter in band with lower lobes than the hanning window:

[image]

jack-h, 2023-08-31:
Updated in `souk_single_pipeline_4x2_2023-08-30_1832.fpg`

closed 2023-08-31 by jack-h

jack-h, 2023-09-05:
> Works a treat!
>
> Figures below show PSB tones measured on spectrum analyser with Hann on
> the left, DPSS on the right.
>
> SFDR reduced by up to 25 dB at the band edges, and 5.5 dB on average.
>
> SNR improved by up to 12 dB or more in places, and 2.5 dB on average.
>
> Also, the tone power is varying slightly more smoothly with the DPSS.
>
> [image: hann_vs_dpss_sweep]
> <https://user-images.githubusercontent.com/17500915/265725775-fe38095f-6a7d-4158-96b6-858ec8f43e42.png>
>
> [image: hann_vs_dpss_reduced]
> <https://user-images.githubusercontent.com/17500915/265725794-2f22e061-5200-4518-93e1-2884efdbef8a.png>

Nice -- I think there is certainly scope to improve the spur level a little
if necessary, either by slightly increasing the number of taps in the
filter and/or tweaking `NW` in the window generation function.