realtimeradio/souk-firmware #27: Symlink to dtbo

sr-cdf, 2023-09-11:
I notice the latest fpg files are symlinked to a more generic filename so the config file doesn't need updating with each new version.

When programming, the system looks for a dtbo with the same name as the link but with .dtbo instead of .fpg. 

Can you add a symlink to the latest dtbo file too?

Or tweak the software somehow so it uses the symlink target instead of the link name when searching for the dtbo file when programming?

jack-h, 2023-09-11:
How's this -- https://github.com/realtimeradio/souk-firmware/commit/1ad1d1db790d43e9923b059ebfefae1c9c95c50d

The symlink is now expanded prior to programming. `self.fpgfile` remains populated with the unexpanded path, which may or may not be a symlink.
Just tested here and it works for me!

closed 2023-09-11 by jack-h

sr-cdf, 2023-09-11:
That ought to do it!