realtimeradio/souk-firmware #33: pip install issues with v6 on new krm board

sr-cdf, 2024-01-15:
Followed the instructions on the README after fresh install of casper krm image on new board.

Hardware setup following `souk_firmware/README.md` worked perfectly.

SSH into board with no trouble.

Software setup following  `souk_firmware/software/README.md` threw a warning on casperfpga `circus requires Python '>=3.7' but the running Python is 3.6.9` and failed on control_sw `Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-y0b4v65s/pyyaml/`

Details:
```
(py3venv) casper@localhost:~/src/souk-firmware/software$ cd casperfpga/
(py3venv) casper@localhost:~/src/souk-firmware/software/casperfpga$ pip install -r requirements.txt

/home/casper/py3venv/lib/python3.6/site-packages/redis/utils.py:13: CryptographyDeprecationWarning: Python 3.6 is no longer supported by the Python core team. Therefore, support for it is deprecated in cryptography and will be removed in a future release.
  import cryptography  # noqa
Requirement already satisfied: cython in /home/casper/py3venv/lib/python3.6/site-packages (from -r requirements.txt (line 1))
Requirement already satisfied: future in /home/casper/py3venv/lib/python3.6/site-packages (from -r requirements.txt (line 2))
Requirement already satisfied: katcp in /home/casper/py3venv/lib/python3.6/site-packages (from -r requirements.txt (line 3))
Requirement already satisfied: numpy in /home/casper/py3venv/lib/python3.6/site-packages (from -r requirements.txt (line 4))
Requirement already satisfied: odict in /home/casper/py3venv/lib/python3.6/site-packages (from -r requirements.txt (line 5))
Requirement already satisfied: tornado in /home/casper/py3venv/lib/python3.6/site-packages (from -r requirements.txt (line 6))
Requirement already satisfied: redis in /home/casper/py3venv/lib/python3.6/site-packages (from -r requirements.txt (line 7))
Requirement already satisfied: tftpy in /home/casper/py3venv/lib/python3.6/site-packages (from -r requirements.txt (line 8))
Requirement already satisfied: progressbar2 in /home/casper/py3venv/lib/python3.6/site-packages (from -r requirements.txt (line 9))
Requirement already satisfied: traitlets in /home/casper/py3venv/lib/python3.6/site-packages (from -r requirements.txt (line 11))
Requirement already satisfied: jedi in /home/casper/py3venv/lib/python3.6/site-packages (from -r requirements.txt (line 12))
Requirement already satisfied: parso in /home/casper/py3venv/lib/python3.6/site-packages (from -r requirements.txt (line 13))
Requirement already satisfied: ipython in /home/casper/py3venv/lib/python3.6/site-packages (from -r requirements.txt (line 14))
Requirement already satisfied: requests in /home/casper/py3venv/lib/python3.6/site-packages (from -r requirements.txt (line 15))
Requirement already satisfied: ply in /home/casper/py3venv/lib/python3.6/site-packages (from katcp->-r requirements.txt (line 3))
Requirement already satisfied: setuptools in /home/casper/py3venv/lib/python3.6/site-packages (from odict->-r requirements.txt (line 5))
Requirement already satisfied: async-timeout>=4.0.2 in /home/casper/py3venv/lib/python3.6/site-packages (from redis->-r requirements.txt (line 7))
Requirement already satisfied: importlib-metadata>=1.0; python_version < "3.8" in /home/casper/py3venv/lib/python3.6/site-packages (from redis->-r requirements.txt (line 7))
Requirement already satisfied: typing-extensions; python_version < "3.8" in /home/casper/py3venv/lib/python3.6/site-packages (from redis->-r requirements.txt (line 7))
Requirement already satisfied: packaging>=20.4 in /home/casper/py3venv/lib/python3.6/site-packages (from redis->-r requirements.txt (line 7))
Requirement already satisfied: python-utils>=2.3.0 in /home/casper/py3venv/lib/python3.6/site-packages (from progressbar2->-r requirements.txt (line 9))
Requirement already satisfied: six in /home/casper/py3venv/lib/python3.6/site-packages (from progressbar2->-r requirements.txt (line 9))
Requirement already satisfied: decorator in /home/casper/py3venv/lib/python3.6/site-packages (from traitlets->-r requirements.txt (line 11))
Requirement already satisfied: ipython-genutils in /home/casper/py3venv/lib/python3.6/site-packages (from traitlets->-r requirements.txt (line 11))
Requirement already satisfied: backcall in /home/casper/py3venv/lib/python3.6/site-packages (from ipython->-r requirements.txt (line 14))
Requirement already satisfied: pygments in /home/casper/py3venv/lib/python3.6/site-packages (from ipython->-r requirements.txt (line 14))
Requirement already satisfied: pexpect; sys_platform != "win32" in /home/casper/py3venv/lib/python3.6/site-packages (from ipython->-r requirements.txt (line 14))
Requirement already satisfied: prompt-toolkit!=3.0.0,!=3.0.1,<3.1.0,>=2.0.0 in /home/casper/py3venv/lib/python3.6/site-packages (from ipython->-r requirements.txt (line 14))
Requirement already satisfied: pickleshare in /home/casper/py3venv/lib/python3.6/site-packages (from ipython->-r requirements.txt (line 14))
Requirement already satisfied: certifi>=2017.4.17 in /home/casper/py3venv/lib/python3.6/site-packages (from requests->-r requirements.txt (line 15))
Requirement already satisfied: charset-normalizer~=2.0.0; python_version >= "3" in /home/casper/py3venv/lib/python3.6/site-packages (from requests->-r requirements.txt (line 15))
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /home/casper/py3venv/lib/python3.6/site-packages (from requests->-r requirements.txt (line 15))
Requirement already satisfied: idna<4,>=2.5; python_version >= "3" in /home/casper/py3venv/lib/python3.6/site-packages (from requests->-r requirements.txt (line 15))
Requirement already satisfied: zipp>=0.5 in /home/casper/py3venv/lib/python3.6/site-packages (from importlib-metadata>=1.0; python_version < "3.8"->redis->-r requirements.txt (line 7))
Requirement already satisfied: pyparsing!=3.0.5,>=2.0.2 in /home/casper/py3venv/lib/python3.6/site-packages (from packaging>=20.4->redis->-r requirements.txt (line 7))
Requirement already satisfied: ptyprocess>=0.5 in /home/casper/py3venv/lib/python3.6/site-packages (from pexpect; sys_platform != "win32"->ipython->-r requirements.txt (line 14))
Requirement already satisfied: wcwidth in /home/casper/py3venv/lib/python3.6/site-packages (from prompt-toolkit!=3.0.0,!=3.0.1,<3.1.0,>=2.0.0->ipython->-r requirements.txt (line 14))
(py3venv) casper@localhost:~/src/souk-firmware/software/casperfpga$ 
(py3venv) casper@localhost:~/src/souk-firmware/software/casperfpga$ pip install .
/home/casper/py3venv/lib/python3.6/site-packages/redis/utils.py:13: CryptographyDeprecationWarning: Python 3.6 is no longer supported by the Python core team. Therefore, support for it is deprecated in cryptography and will be removed in a future release.
  import cryptography  # noqa
Processing /home/casper/src/souk-firmware/software/casperfpga
Requirement already satisfied: IPython in /home/casper/py3venv/lib/python3.6/site-packages (from casperfpga==0.0+unknown.202401151412)
Requirement already satisfied: future in /home/casper/py3venv/lib/python3.6/site-packages (from casperfpga==0.0+unknown.202401151412)
Requirement already satisfied: numpy in /home/casper/py3venv/lib/python3.6/site-packages (from casperfpga==0.0+unknown.202401151412)
Collecting katcp==0.9.1 (from casperfpga==0.0+unknown.202401151412)
  Downloading https://files.pythonhosted.org/packages/e1/15/40494b7a1a39d381c9834e4ea070c45133e6964417d0ce51cdd314e17541/katcp-0.9.1-py3-none-any.whl (214kB)
    100% |████████████████████████████████| 215kB 1.5MB/s 
Collecting katversion (from casperfpga==0.0+unknown.202401151412)
  Using cached https://files.pythonhosted.org/packages/c2/a8/e0dffadc4f735363595bef94f5dda46ba3e9a8961c6c4cb0c63df5aec2e3/katversion-1.2-py2.py3-none-any.whl
Requirement already satisfied: odict in /home/casper/py3venv/lib/python3.6/site-packages (from casperfpga==0.0+unknown.202401151412)
Requirement already satisfied: setuptools in /home/casper/py3venv/lib/python3.6/site-packages (from casperfpga==0.0+unknown.202401151412)
Requirement already satisfied: tornado in /home/casper/py3venv/lib/python3.6/site-packages (from casperfpga==0.0+unknown.202401151412)
Requirement already satisfied: redis in /home/casper/py3venv/lib/python3.6/site-packages (from casperfpga==0.0+unknown.202401151412)
Requirement already satisfied: tftpy in /home/casper/py3venv/lib/python3.6/site-packages (from casperfpga==0.0+unknown.202401151412)
[... 36 lines cut ...]
      File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
[... cut ...]
      File "/tmp/pip-build-y0b4v65s/pyyaml/setup.py", line 201, in get_source_files
        self.cython_sources(ext.sources, ext)
      File "/usr/lib/python3.6/distutils/cmd.py", line 103, in __getattr__
        raise AttributeError(attr)
    AttributeError: cython_sources
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-y0b4v65s/pyyaml/
```

My first google search suggested dropping pyyaml back to 5.3.1, but  the pip requirements explicitly requires version==5.4.1.

sr-cdf, 2024-01-16:
@jack-h, is version 6 even ready for testing yet?

jack-h, 2024-01-16:
Having a look now.

jack-h, 2024-01-16:
I removed the pyyaml version req in commit 15da6136e88 and that seems to get past this. Experimenting on a board now.
Be aware that you need to apply 10MHz to the reference port of the clock card to make things clock.
You can check if a board has locked OK:

```
(py3venv) casper@localhost:~$ sudo krc-utils status
[sudo] password for casper: 
[lmk04208.0] status: locked
[ lmx2594.0] status: locked
[ lmx2594.1] status: locked
[ lmx2594.2] status: locked
```

See if you can at least get to the stage of initializing ok with `souk_single_pipeline_krm_2023-10-25_1355.fpg` then by our meeting on Thurs I'll verify what level of testing should be working.

In ipython you should at least be able to verify after initialization:
```
In [4]: r.fpga.get_status()
Out[4]: 
({'programmed': True,
  'timestamp': '2024-01-16T13:10:06.658232',
  'host': 'localhost.localdomain:localhost',
  'antname': None,
  'sw_version': '0.0.1+839a556d-dirty',
  'fw_version': '6.1.0.0',
  'fw_type': 2,
  'fw_build_time': '2023-10-25T14:54:58',
  'fw_supported': True},
 {'sw_version': 2})

In [5]: r.fpga.get_fpga_clock()
Out[5]: 307888501.0
```

sr-cdf, 2024-01-16:
Ok I'll give it a go. 

Does the using the external 10mhz mean that the onboard OCXO is not in use?

sr-cdf, 2024-01-16:
OK.

Will the same LMX/LMK files for 4x2 work on this board? They'll need copying over to /lib/firmware, I suppose?

jack-h, 2024-01-16:
Unfortunately not. They are loaded at boot my a tool provided by Knowres.
The files I have (and the ones used are those symlinked) are:
```
(py3venv) casper@localhost:~/src/souk-firmware/software/control_sw$ ls /etc/krc-utils.d/clock.d/ -lah
total 120K
drwxr-xr-x 2 casper casper 4.0K Oct 26 12:02 .
drwxr-xr-x 3 root   root   4.0K Aug 17 10:19 ..
lrwxrwxrwx 1 casper casper   35 Oct 26 12:02 lmk04208.txt -> lmk04208_in_10M_clk0_out_122M88.txt
-rw-r--r-- 1 root   root    412 Oct 12 18:55 lmk04208_in_10M_clk0_out_122M88.txt
-rw-r--r-- 1 casper casper  412 Aug 17 10:17 lmk04208_in_12M8_out_122M88.txt
lrwxrwxrwx 1 casper casper   32 Oct 12 18:56 lmx2594.txt -> lmx2594_in_122M88_out_0409M6.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_0102M4.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_0204M8.txt
-rw-r--r-- 1 casper casper 1.6K Aug 17 10:17 lmx2594_in_122M88_out_0245M76.txt
-rw-r--r-- 1 casper casper 1.6K Aug 17 10:17 lmx2594_in_122M88_out_0409M6.txt
-rw-r--r-- 1 casper casper 1.6K Aug 17 10:17 lmx2594_in_122M88_out_0491M52.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_0737M28.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_1474M56.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_1966M08.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_2048M.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_2457M6.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_2949M12.txt
-rw-r--r-- 1 casper casper 1.6K Aug 17 10:17 lmx2594_in_122M88_out_3072M.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_3194M88.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_3276M8.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_3686M4.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_3923M16.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_4096M.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_4423M68.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_4669M44.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_4915M2.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_5734M4.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_5898M24.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_6144M.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_6389M76.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_6400M.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_6553M6.txt
```

If you are missing any of these in the OS image I can add them to the repo. If you switch to `lmk04208_in_12M8_out_122M88.txt` I believe that is all that is required to go back to the onboard oscillator

jack-h, 2024-01-16:
To be more clear -- the LMX / LMK files aren't required to be loaded via casperfpga on the KRM board. This saves a config step, but also means you can't change them on the fly to mess with clock rates like you can with the 4x2.

sr-cdf, 2024-01-16:
lmk04208_in_10M_clk0_out_122M88.txt is missing on the image.

I'll make the symlinks and point the config.yaml to them...

jack-h, 2024-01-17:
So, the clock programming is dubiously handled by the `~casper/onboot.sh` script, which has to program the FPGA prior to loading the clock config.

sr-cdf, 2024-01-18:
Mine doesn't seem to be using the symlinks at all and I can't find where it is getting the paths from.

```(py3venv) casper@localhost:~$ sudo /usr/bin/krc-utils init
[lmk04208.0] Loading: /etc/krc-utils.d/clock.d/lmk04208_in_12M8_out_122M88.txt
[ lmx2594.0] Loading: /etc/krc-utils.d/clock.d/lmx2594_in_122M88_out_0409M6.txt
[ lmx2594.1] Loading: /etc/krc-utils.d/clock.d/lmx2594_in_122M88_out_0409M6.txt
[ lmx2594.2] Loading: /etc/krc-utils.d/clock.d/lmx2594_in_122M88_out_0409M6.txt
[lmk04208.0] status: locked
[ lmx2594.0] status: locked
[ lmx2594.1] status: locked
[ lmx2594.2] status: locked
(py3venv) casper@localhost:~$ 
```

The directory seems identical to yours:
```
(py3venv) casper@localhost:~$ ls /etc/krc-utils.d/clock.d/ -lah
total 120K
drwxr-xr-x 2 casper casper 4.0K Jan 17 12:46 .
drwxr-xr-x 3 root   root   4.0K Jan 17 12:28 ..
lrwxrwxrwx 1 casper casper   35 Jan 17 12:28 lmk04208.txt -> lmk04208_in_10M_clk0_out_122M88.txt
-rw-r--r-- 1 root   root    412 Jan 17 12:28 lmk04208_in_10M_clk0_out_122M88.txt
-rw-r--r-- 1 casper casper  412 Aug 17 10:17 lmk04208_in_12M8_out_122M88.txt
lrwxrwxrwx 1 casper casper   32 Jan 17 12:46 lmx2594.txt -> lmx2594_in_122M88_out_0409M6.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_0102M4.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_0204M8.txt
-rw-r--r-- 1 casper casper 1.6K Aug 17 10:17 lmx2594_in_122M88_out_0245M76.txt
-rw-r--r-- 1 casper casper 1.6K Aug 17 10:17 lmx2594_in_122M88_out_0409M6.txt
-rw-r--r-- 1 casper casper 1.6K Aug 17 10:17 lmx2594_in_122M88_out_0491M52.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_0737M28.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_1474M56.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_1966M08.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_2048M.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_2457M6.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_2949M12.txt
-rw-r--r-- 1 casper casper 1.6K Aug 17 10:17 lmx2594_in_122M88_out_3072M.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_3194M88.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_3276M8.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_3686M4.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_3923M16.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_4096M.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_4423M68.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_4669M44.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_4915M2.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_5734M4.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_5898M24.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_6144M.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_6389M76.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_6400M.txt
-rw-r--r-- 1 casper casper 1.5K Aug 17 10:17 lmx2594_in_122M88_out_6553M6.txt
```

jack-h, 2024-01-18:
> (py3venv) ***@***.***:~$ ls /etc/krc-utils.d/clock.d/ -lah

Interesting. Let me see if I changed the krc code.

jack-h, 2024-01-18:
> I see in clock_spi_interface.h:
>
> #define LMK_CFG_FILE "lmk04208_in_12M8_out_122M88.txt"
> #define LMX_CFG_FILE "lmx2594_in_122M88_out_0409M6.txt"
>
> did you change these and recompile?

Yes!

sr-cdf, 2024-01-18:
Can you send me a copy of your '/home/casper/src/krc-utils' directory and your '/usr/bin/krc-utils` 

And do you want to update the image with the recompiled version?

sr-cdf, 2024-01-18:
No need to copy to me now, it built and works after replacing /usr/bin/krc-utils.

```
sudo krc-utils init
[lmk04208.0] Loading: /etc/krc-utils.d/clock.d/lmk04208.txt
Timeout: lmk04208.0 not locked
[ lmx2594.0] Loading: /etc/krc-utils.d/clock.d/lmx2594.txt
[ lmx2594.1] Loading: /etc/krc-utils.d/clock.d/lmx2594.txt
[ lmx2594.2] Loading: /etc/krc-utils.d/clock.d/lmx2594.txt
[lmk04208.0] status: unlocked
[ lmx2594.0] status: unlocked high
[ lmx2594.1] status: unlocked high
[ lmx2594.2] status: unlocked high
(py3venv) casper@localhost:~/src/krc-

sr-cdf, 2024-01-18:
And with the 10MHz plugged in, it locks:
```
(py3venv) casper@localhost:~/src/krc-utils$ sudo krc-utils init
[lmk04208.0] Loading: /etc/krc-utils.d/clock.d/lmk04208.txt
[ lmx2594.0] Loading: /etc/krc-utils.d/clock.d/lmx2594.txt
[ lmx2594.1] Loading: /etc/krc-utils.d/clock.d/lmx2594.txt
[ lmx2594.2] Loading: /etc/krc-utils.d/clock.d/lmx2594.txt
[lmk04208.0] status: locked
[ lmx2594.0] status: locked
[ lmx2594.1] status: locked
[ lmx2594.2] status: locked
```

sr-cdf, 2024-01-18:
The pip issue seems to be solved and the clocking is working now.

closed 2024-01-18 by sr-cdf