realtimeradio/souk-firmware #35: tone frequency offset wrong by a half

sr-cdf, 2024-03-08:
Hey @jack-h ,

I'm trying to get the sweeping working but something appears to be up with the tone frequencies, as viewed on a spectrum analyser.

If the tone offset from the bin center is zero, all is good, but for a given offset, the measured tone only has half the offset.

Eg:
Set to center of bin 7428 (999.6 MHz), measured tone at 999.6 MHz.
Set to +50Khz offset from bin 7428 (999.65 MHz), measured tone at 999.625MHz.

This is version 6.3.0.0 with the single real ADC input.

In `r.mixer.set_chan_freq` and `r.mixer.set_freqs`, the phase step for the LO is calculated as:
```
        fft_period_s = self.n_chans / sample_rate_hz
        fft_rbw_hz = 1./fft_period_s # FFT channel width, Hz
        phase_steps = freqs_hz / fft_rbw_hz * 2 * np.pi
```

This gives 1.2MHz for the channel width, where `mixer.n_chans == 2048` and `sample_rate_hz == r.adc_clk_hz == 2457.6e6`, regardless of the actual FFT size, which i believe is 8192 in this version. (The RX PFB FFT is 8192, the 2x TX PSB FFT are 4096, giving 8192 in total. So I'd expect the channel width to be 300 khz. But this is a factor of 4 off! 

If I add a factor of 2 here, I can get the tones back to the expected places:
<pre><code>
        fft_period_s = self.n_chans / sample_rate_hz <b>* 2</b>
        fft_rbw_hz = 1./fft_period_s # FFT channel width, Hz
        phase_steps = freqs_hz / fft_rbw_hz * 2 * np.pi
</code></pre>

But this is a hack and I have no idea if this is likely to screw up a bunch of other stuff down stream.

jack-h, 2024-03-08:
Sorry, this should have been caught earlier.

Should be fixed in https://github.com/realtimeradio/souk-firmware/commit/109a515184c8824b6a7b2edbd944b53eb1cf3011

Testing with:
```
python souk_load_tones.py 10.10.10.106 ../config/souk-single-pipeline-krm.yaml 999050000,999350000,999650000,999950000
```

I see 4 tones at the correct frequencies, with 2 sent from each DAC (999.05 & 999.65 MHz on the first DAC, 999.35 & 999.95MHz on the second DAC)

sr-cdf, 2024-03-08:
Sorry it took me so long to spot it!

All good now, thanks!

closed 2024-03-08 by sr-cdf