realtimeradio/souk-firmware #38: KRM DAC1 is the active output when 'p0_use_dual_dac' is off, not DAC0

sr-cdf, 2024-03-21:
Only an issue if you have terrible OCD.

DAC 1 cable loopback sweep:

[image]

DAC 0 cable loopback sweep:

[image]

jack-h, 2024-03-21:
You know, as I built the firmware I really didn't think I had better than a
50:50 chance of getting this right.

jack-h, 2024-03-21:
This bothers me, so building the fix now.

jack-h, 2024-03-21:
Fixed in `v6.3.1.1`
Also (added bonus) handled the errors you normally get when instantiating a connection to a just-booted unprogrammed board

closed 2024-03-21 by jack-h