realtimeradio/souk-firmware #39: Switch to 2nd Nyquist zone?

sr-cdf, 2024-04-02:
How easy is it to add a switch to set the converters to use the 2nd nyquist zone?

mitchburnett, 2024-04-03:
It should be available in the recent casperfpga `rfdc` class object as [`get`/`set_nyquist_zone()`](https://github.com/realtimeradio/casperfpga/blob/453f556a99417bfc7d9899411548cc08da1b74b1/src/rfdc.py#L602) with the dump of other control features like DSA/VOP.

As an aside, I am curious to know how the rest of the control changes have worked out for you? If you have tested them or noticed any issues.

sr-cdf, 2024-04-03:
Aha, thanks Mitch, the RFDC class object is called `SoukMkidReadout.rfdc.core` and I see the recent functions in there but it looks like i need to update my tcpborphserver as calling the function fails with an `unknown command` error:

```
In [4]: r.rfdc.core.get_nyquist_zone(0,0,'dac')
---------------------------------------------------------------------------
KatcpRequestInvalid                       Traceback (most recent call last)
<ipython-input-4-fce370b42f90> in <module>
----> 1 r.rfdc.core.get_nyquist_zone(0,0,'dac')

~/py3venv/lib/python3.6/site-packages/casperfpga/rfdc.py in get_nyquist_zone(self, ntile, nblk, converter_type)
    590 
    591     args = (ntile, nblk, "adc" if converter_type == self.ADC_TILE  else "dac")
--> 592     reply, informs = t.katcprequest(name='rfdc-get-nyquist-zone', request_timeout=t._timeout, request_args=args)
    593 
    594     info = informs[0].arguments[0].decode().split(' ')

~/py3venv/lib/python3.6/site-packages/casperfpga/transport_katcp.py in katcprequest(self, name, request_timeout, require_ok, request_args, max_retries, retry_delay_s)
    298                     'Invalid katcp request %s on host %s.\n\t'
    299                     'Request: %s\n\tReply: %s' %
--> 300                     (request.name, self.host, request, reply))
    301             else:
    302                 raise KatcpRequestError(

KatcpRequestInvalid: Invalid katcp request rfdc-get-nyquist-zone on host localhost.
        Request: ?rfdc-get-nyquist-zone 0 0 dac
        Reply: !rfdc-get-nyquist-zone invalid unknown\_command

```

[tries to update tcpborphserver]

At first i tried following the instructions at https://casper.groups.et.byu.net/souk-updates/ as mentioned in https://github.com/realtimeradio/souk-firmware/issues/1#issuecomment-1833652732 but this led to issues programming the fpga. 

Then tried again working from https://github.com/realtimeradio/souk-firmware/tree/main/software/zynq, but it failed to program again:

```
In [1]: import souk_mkid_readout
r =/home/casper/py3venv/lib/python3.6/site-packages/redis/utils.py:13: CryptographyDeprecationWarning: Python 3.6 is no longer supported by the Python core team. Therefore, support for it is deprecated in cryptography and will be removed in a future release.
  import cryptography  # noqa
^[[A
In [2]: r = souk_mkid_readout.SoukMkidReadout('localhost',fpgfile='/home/casper/src/souk-firmware/firmware/src/souk_sing
   ...: le_pipeline_krm/outputs/souk_single_pipeline_krm.fpg', local=False)
r.program()2024-04-03 17:09:54,992 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Board is not programmed with valid firmware. Skipping block initialization

In [3]: r.program()
2024-04-03 17:09:57,492 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Programming with /home/casper/src/souk-firmware/firmware/src/souk_single_pipeline_krm/outputs/souk_single_pipeline_krm_2024-03-21_1105.fpg
---------------------------------------------------------------------------
KatcpRequestFail                          Traceback (most recent call last)
<ipython-input-3-5a4876cc90ca> in <module>
----> 1 r.program()

~/py3venv/lib/python3.6/site-packages/souk_mkid_readout/souk_mkid_readout.py in program(self, fpgfile)
    204         realpath = path.realpath(self.fpgfile)
    205         self.logger.info(f"Programming with {realpath}")
--> 206         self._cfpga.upload_to_ram_and_program(realpath)
    207         self._initialize_blocks()
    208 

~/py3venv/lib/python3.6/site-packages/casperfpga/casperfpga.py in upload_to_ram_and_program(self, filename, wait_complete, initialise_objects, **kwargs)
    357                 if self.bitstream[-3:] == 'fpg':
    358                     self.get_system_information(filename,
--> 359                                                 initialise_objects=initialise_objects)
    360 
    361 

~/py3venv/lib/python3.6/site-packages/casperfpga/casperfpga.py in get_system_information(self, filename, fpg_info, initialise_objects, **kwargs)
    817         self._create_other_devices(device_dict, initialise=initialise_objects)
    818         self.logger.info("Creating ADC devices")
--> 819         self._create_casper_adc_devices(device_dict, initialise=initialise_objects)
    820         self.transport.memory_devices = self.memory_devices
[... 106 lines cut ...]
-rw-rw-r--  1 casper casper  169720 Mar 21 08:47 librfdc.so.8.1
-rw-rw-r--  1 casper casper    3992 Mar 21 08:47 readme
-rw-rw-r--  1 casper casper 2521848 Mar 21 08:47 tcpborphserver3_v8.1
```

```
casper@localhost:~$ ls -l ~/souk-updates/
total 3004
-rw-rw-r-- 1 casper casper  169720 Nov 30 06:10 librfdc.so.8.1
-rw-rw-r-- 1 casper casper  169720 Nov 30 06:10 librfdc.so.8.1.1
-rw-rw-r-- 1 casper casper    6830 Jan 31 03:22 readme
-rw-rw-r-- 1 casper casper 2720904 Jan 31 05:03 tcpborphserver3_v8.1
```

```(py3venv) casper@localhost:~$ ls -l /opt/local/lib/
total 552
-rwxr-xr-x 1 root root 193280 Aug 18  2023 librfdc.so
lrwxrwxrwx 1 root root     14 Apr  3 16:43 librfdc.so.1 -> librfdc.so.8.1
-rwxr-xr-x 1 root root 193280 Aug 18  2023 librfdc.so.1.1
-rwxr-xr-x 1 root root 169720 Apr  3 16:43 librfdc.so.8.1
```

I note that the link in https://github.com/realtimeradio/souk-firmware/issues/1#issuecomment-1834080672 is dead.

Maybe you or @jack-h could point me to a version that will work?

mitchburnett, 2024-04-05:
@sr-cdf, sorry about the issues. Can I get some clarification on the details when programming after making the update per the instructions in the `readme`? Is the provided block of text from the `telnet` session the result from working from the souk-firmware `zynq/` or is it from when you updated `tcpborphserver3` directly from my ftp share? Did you reboot the board after replacing it? Or just restart the `tcpborphserver3` service?

The sizes look right for what I am expecting. Further more the md5sum for the version I have here is: `bf43a248d632dc06ce4fad25794a9ae1  tcpborphserver3_v8.1`. There is no update to be made for `librfdc.so.8.1`

The reason that it is failing in that `telnet` session is because the device tree overlay (mapping the rfdc into the linux device tree) is not having any success, which is rather unexpected. It looks to be hung at the same spot in `casperfpga`. Is the `telnet` session block the output from calling `r.program()`?  Regardless, that portion of the programming sequence is unchanged from even your current version of TBS that works.

I had pinged @jack-h about the updates to pass them onto you. I cannot say for sure though if he has a slightly different process to translate my instructions to your system.

mitchburnett, 2024-04-08:
@sr-cdf I took a look over the weekend. It doesn't look to me like there is anything out of the ordinary to update TBS for you. So, let me know about these other questions about restarting TBS and we can go from there working this out.

jack-h, 2024-04-08:
The md5sum of the tcpborphserver at https://github.com/realtimeradio/souk-firmware/tree/main/software/zynq is `acd906bb9be1ea8c0f40b8a50a317af6`.
There is a newer version on my board which seems to have the relevant commands. I have uploaded this -- https://github.com/realtimeradio/souk-firmware/blob/main/software/zynq/tcpborphserver3_v8.1 -- it looks like I merged the source code but didn't update the binary in git. Sorry about that.

Try copying this new binary to `/usr/local/sbin/tcpborphserver3`

sr-cdf, 2024-04-08:
Thanks guys,

@jack-h I just tried this updated version and I get the same dto error when attempting to program. Did you change anything else in the time between sharing the initial KRM CASPER image with me, and updating to the latest TBS?

The only version that works for me is the 'stock' version with `4d4884978c393186c21dbd025d72b46e`. All other versions fail to program.

@mitchburnett thanks for taking a look, i think the telnet session above was with the souk-firmware/software/sync version.

I did reboot after replacing it (with `sudo reboot` as opposed to a full power cycle).

And yes, I called `r.program()` while the telnet session was open, and that's what popped up.

jack-h, 2024-04-08:
hmmm. I'll have a look with my test setup here

jack-h, 2024-04-09:
On my RFSoC board:

```
(py3venv) casper@localhost:~/src/souk-firmware/software/zynq$ md5sum *.1
c54b30b0cbe82f8dbb95fa138ee62dfb  librfdc.so.8.1
263baa56b050f8ed7fe5d495e6f4814f  tcpborphserver3_v8.1
(py3venv) casper@localhost:~/src/souk-firmware/software/zynq$ md5sum /usr/local/sbin/tcpborphserver3
263baa56b050f8ed7fe5d495e6f4814f  /usr/local/sbin/tcpborphserver3
(py3venv) casper@localhost:~/src/souk-firmware/software/zynq$ ldd /usr/local/sbin/tcpborphserver3
	linux-vdso.so.1 (0x0000ffff9941b000)
	libz.so.1 => /lib/aarch64-linux-gnu/libz.so.1 (0x0000ffff9933b000)
	librfdc.so.1 => /opt/local/lib/librfdc.so.1 (0x0000ffff99301000)
	libmetal.so.0 => /usr/local/lib/libmetal.so.0 (0x0000ffff992b7000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000ffff9915e000)
	/lib/ld-linux-aarch64.so.1 (0x0000ffff993ef000)
	libsysfs.so.2 => /lib/aarch64-linux-gnu/libsysfs.so.2 (0x0000ffff99145000)
	libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000ffff99119000)
	librt.so.1 => /lib/aarch64-linux-gnu/librt.so.1 (0x0000ffff99100000)
(py3venv) casper@localhost:~/src/souk-firmware/software/zynq$ md5sum /opt/local/lib/librfdc.so.1
c54b30b0cbe82f8dbb95fa138ee62dfb  /opt/local/lib/librfdc.so.1
(py3venv) casper@localhost:~/src/souk-firmware/software/zynq$ md5sum /usr/local/lib/libmetal.so.0
b928a41f2811ad292eeb17b46924e241  /usr/local/lib/libmetal.so.0
```

I.e., both rfdc library and borphserver binary match those in the repo.

I have no problems doing:

```
In [4]: import souk_mkid_readout

In [5]: r = souk_mkid_readout.SoukMkidReadout('10.10.10.127', fpgfile='/home/jackh/src/souk-firmware2/souk-firmware/firmware/src/souk_single_pipeline_krm/outputs/souk_single_pipeline_kr
   ...: m_2024-03-21_1105.fpg')
2024-04-09 11:30:39,000 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Board is not programmed with valid firmware. Skipping block initialization

In [6]: r.initialize()
2024-04-09 11:30:45,153 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Board is _NOT_ programmed
2024-04-09 11:30:45,154 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Programming with /home/jackh/src/souk-firmware2/souk-firmware/firmware/src/souk_single_pipeline_krm/outputs/souk_single_pipeline_krm_2024-03-21_1105.fpg
2024-04-09 11:30:46,366 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Initializing block (writable): fpga
2024-04-09 11:30:46,366 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Initializing block (writable): rfdc
2024-04-09 11:30:46,441 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Initializing block (writable): sync
2024-04-09 11:30:46,450 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Initializing block (writable): input
2024-04-09 11:30:46,452 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Initializing block (writable): pfb
2024-04-09 11:30:46,456 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Initializing block (writable): pfbtvg
2024-04-09 11:30:46,457 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Initializing block (writable): autocorr
2024-04-09 11:30:46,458 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Initializing block (writable): gen_cordic
2024-04-09 11:30:46,469 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Initializing block (writable): gen_lut
2024-04-09 11:30:46,611 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Initializing block (writable): psbscale
2024-04-09 11:30:46,612 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Initializing block (writable): zoomfft
2024-04-09 11:30:46,616 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Initializing block (writable): zoomacc
2024-04-09 11:30:46,618 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Initializing block (writable): chanselect
2024-04-09 11:30:46,673 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Initializing block (writable): mixer
2024-04-09 11:30:46,799 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Initializing block (writable): psb_chanselect
2024-04-09 11:30:46,927 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Initializing block (writable): psb
2024-04-09 11:30:46,932 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Initializing block (writable): accumulator0
2024-04-09 11:30:46,951 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Initializing block (writable): accumulator1
2024-04-09 11:30:46,972 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Initializing block (writable): output
2024-04-09 11:30:46,973 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Initializing block (writable): out_delay
2024-04-09 11:30:46,975 - souk_mkid_readout.souk_mkid_readout:10.10.10.127:0 - INFO - Performing software global reset
2024-04-09 11:30:48,183 - souk_mkid_readout.blocks.block:10.10.10.127:p0_sync - WARNING - Timed out waiting for sync pulse

In [7]: r.rfdc.core.get_nyquist_zone(0,0,'adc')
Out[7]: 1
```

Anything different in the libraries used by your tcpborphserver, @sr-cdf ?

sr-cdf, 2024-04-09:
No difference in those libraries:
```(py3venv) casper@localhost:~$ cd src/souk-firmware/software/zynq
(py3venv) casper@localhost:~/src/souk-firmware/software/zynq$ md5sum *.1
c54b30b0cbe82f8dbb95fa138ee62dfb  librfdc.so.8.1
263baa56b050f8ed7fe5d495e6f4814f  tcpborphserver3_v8.1
(py3venv) casper@localhost:~/src/souk-firmware/software/zynq$ md5sum /usr/local/sbin/tcpborphserver3
263baa56b050f8ed7fe5d495e6f4814f  /usr/local/sbin/tcpborphserver3
(py3venv) casper@localhost:~/src/souk-firmware/software/zynq$ ldd /usr/local/sbin/tcpborphserver3
        linux-vdso.so.1 (0x0000ffff8c7f3000)
        libz.so.1 => /lib/aarch64-linux-gnu/libz.so.1 (0x0000ffff8c713000)
        librfdc.so.1 => /opt/local/lib/librfdc.so.1 (0x0000ffff8c6d9000)
        libmetal.so.0 => /usr/local/lib/libmetal.so.0 (0x0000ffff8c68f000)
        libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000ffff8c536000)
        /lib/ld-linux-aarch64.so.1 (0x0000ffff8c7c7000)
        libsysfs.so.2 => /lib/aarch64-linux-gnu/libsysfs.so.2 (0x0000ffff8c51d000)
        libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000ffff8c4f1000)
        librt.so.1 => /lib/aarch64-linux-gnu/librt.so.1 (0x0000ffff8c4d8000)
(py3venv) casper@localhost:~/src/souk-firmware/software/zynq$ md5sum /opt/local/lib/librfdc.so.1
c54b30b0cbe82f8dbb95fa138ee62dfb  /opt/local/lib/librfdc.so.1
(py3venv) casper@localhost:~/src/souk-firmware/software/zynq$ md5sum /usr/local/lib/libmetal.so.0
b928a41f2811ad292eeb17b46924e241  /usr/local/lib/libmetal.so.0
```

I've even tried starting a new python venv and reinstalling casperfpga and souk_mkid_readout from scratch using the latest commit (needed to pip install -upgrade pip first). But to no avail.

```
(py3venv) casper@localhost:~$ ipython
/home/casper/.local/lib/python3.6/site-packages/IPython/core/interactiveshell.py:935: UserWarning: Attempting to work in a virtualenv. If you encounter problems, please install IPython inside the virtualenv.
  warn("Attempting to work in a virtualenv. If you encounter problems, please "
Python 3.6.9 (default, Jun 29 2022, 11:45:57) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.16.3 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import souk_mkid_readout

In [2]: r = souk_mkid_readout.SoukMkidReadout('localhost',fpgfile='/home/casper/src/souk-firmware/firmware/src/souk_single_pipeline_krm/outputs/souk_single_pipeline_krm_2024-03-21_1105.fpg')

In [3]: r.fpga.print_status()
antname: None
fw_build_time: 2024-03-21T11:04:14
fw_supported: True
fw_type: 2
fw_version: 6.3.1.1
host: localhost.localdomain:localhost
programmed: True
sw_version: 0.0.1+v6.3.1.1-2-g81677baf
timestamp: 2024-04-09T15:51:15.991238

In [4]: r.program()
2024-04-09 15:51:26,409 - souk_mkid_readout.souk_mkid_readout:localhost:0 - INFO - Programming with /home/casper/src/souk-firmware/firmware/src/souk_single_pipeline_krm/outputs/souk_single_pipeline_krm_2024-03-21_1105.fpg
---------------------------------------------------------------------------
KatcpRequestFail                          Traceback (most recent call last)
<ipython-input-4-5a4876cc90ca> in <module>
----> 1 r.program()

~/py3venv/lib/python3.6/site-packages/souk_mkid_readout/souk_mkid_readout.py in program(self, fpgfile)
    204         realpath = path.realpath(self.fpgfile)
    205         self.logger.info(f"Programming with {realpath}")
--> 206         self._cfpga.upload_to_ram_and_program(realpath)
    207         self._initialize_blocks()
    208 

~/py3venv/lib/python3.6/site-packages/casperfpga/casperfpga.py in upload_to_ram_and_program(self, filename, wait_complete, initialise_objects, **kwargs)
    357                 if self.bitstream[-3:] == 'fpg':
    358                     self.get_system_information(filename,
--> 359                                                 initialise_objects=initialise_objects)
    360 
    361 

~/py3venv/lib/python3.6/site-packages/casperfpga/casperfpga.py in get_system_information(self, filename, fpg_info, initialise_objects, **kwargs)
[... 251 lines cut ...]
--> 295                     (request.name, self.host, request, reply))
    296             elif reply.arguments[0] == katcp.Message.INVALID:
    297                 raise KatcpRequestInvalid(

KatcpRequestFail: Request dto on host localhost failed.
        Request: ?dto apply
        Reply: !dto fail

In [5]: r.fpga.print_status()
2024-04-09 16:07:14,200 - souk_mkid_readout.blocks.block:localhost: - WARNING - Software supports FW rev 6.3, but not 1.1.3.6
antname: None
fw_build_time: 1994-07-02T06:34:29
fw_supported: False
fw_type: 33554432
fw_version: 1.1.3.6
host: localhost.localdomain:localhost
programmed: True
sw_version: 0.0.1+v6.3.1.1-2-g81677baf
timestamp: 2024-04-09T16:07:14.193391

```

Note that the `fw_version` is in reverse: 1.1.3.6 instead of 6.3.1.1.

I suppose it usually gets corrected after the DTO is loaded.

jack-h, 2024-04-09:
> Does katcp-library `AVN_legacy-307-gc646b44-dirty 2024-02-01T12:47:02` agree with your system?

Yes.

> I see this: `#log info 1712674559.102 unknown Writing\_'tcpborphserver.bin'\_to\_fpga_manager` 

I see this too.

I see the same tcpborphserver binary and dtbo files so I think the files being copied to the board are correct.

```
(py3venv) casper@localhost:~$ sudo md5sum /lib/firmware/*
abc208e043a4b280f0a2c93e5974e95a  /lib/firmware/spi_3wire.bit
a11b79a421e774eb257e02d54dd2c5f8  /lib/firmware/system-led.bit
675a7fbebc2455883202be806ae12ea6  /lib/firmware/system.bit
24f5a5bec8d7349f845eb4f907ace88e  /lib/firmware/tcpborphserver.bin
c2466eb6312ebf1fff4fe7c95b20f5f7  /lib/firmware/tcpborphserver.dtbo
839b9fe15cae68d302dde26ec336fea0  /lib/firmware/top.bin
411ff23115e830c2017fe27aa4d6821d  /lib/firmware/top.bit
```

```
jackh@rtr-desktop2:~$ telnet 10.10.10.151 7147
Trying 10.10.10.151...
Connected to 10.10.10.151.
Escape character is '^]'.
#version-connect katcp-library AVN_legacy-307-gc646b44-dirty 2024-02-01T12:47:02
#version-connect katcp-protocol 5.0-M
#version-connect kernel 5.10.0-xilinx-v2021.2 #1\_SMP\_Tue\_Oct\_12\_09:30:57\_UTC\_2021
#log info 1712678753.844 raw new\_client\_connection\_10.10.10.154:54414
#client-connected 10.10.10.154:54414
#log error 1712678753.847 raw fpga\_not\_programmed
#log error 1712678754.348 raw fpga\_not\_programmed
#log error 1712678754.850 raw fpga\_not\_programmed
#log error 1712678754.851 raw fpga\_not\_programmed
#log error 1712678755.353 raw fpga\_not\_programmed
#log error 1712678755.855 raw fpga\_not\_programmed
#log error 1712678755.874 raw fpga\_not\_programmed
#log error 1712678756.375 raw fpga\_not\_programmed
#log error 1712678756.877 raw fpga\_not\_programmed
#log error 1712678756.879 raw fpga\_not\_programmed
#log error 1712678757.381 raw fpga\_not\_programmed
#log error 1712678757.883 raw fpga\_not\_programmed
#log info 1712678761.867 raw awaiting\_transfer\_on\_port\_2446
#log info 1712678761.937 raw new\_client\_connection\_127.0.0.1:59856
#client-connected 127.0.0.1:59856
#log info 1712678762.445 unknown Writing\_'tcpborphserver.bin'\_to\_fpga_manager
#log info 1712678762.672 unknown FPGA\_state\_register\_suggests\_programming\_was\_OK
#fpga loaded
#log info 1712678762.921 raw mapping\_area\_0x00073000\_at\_0xa0000000
#fpga mapped
#fpga ready
#log info 1712678761.939 kcpfpg retrying\_connect\_to\_port\_7146
#log info 1712678762.673 kcpfpg fpga\_programmed
#log info 1712678762.921 raw received\_end\_of\_file\_from\_127.0.0.1:59856
#client-disconnected 127.0.0.1:59856
#log info 1712678762.942 raw awaiting\_transfer\_on\_port\_2340
#log info 1712678763.734 raw will\_apply\_overlay
#log info 1712678763.735 raw creating\_dto\_casper
#log info 1712678763.735 raw writing\_dto

#log info 1712678816.616 raw about\_to\_check\_metadata
#log info 1712678816.619 raw current\_metal\_loglevel=6\n
#log info 1712678816.619 raw updated\_metal\_loglevel=7\n
#log info 1712678816.642 raw looking\_up\_rfdc\_device\_configuration
#log info 1712678816.646 raw registering\_rfdc\_with\_metal
#log info 1712678816.647 raw starting\_xrfdc\_driver\_initialization
#log info 1712678816.647 raw rfdc\_configured\_successfuly
```

What's in your `/boot/` partition. Mine is:

```
(py3venv) casper@localhost:~$ sudo md5sum /boot/*
42120053bd949cc0045b6cebc319ba3c  /boot/BOOT.BIN
6139b67a21165152ec3198705328925b  /boot/boot.scr
2998b52455ec9b28a0802f0ac995faa8  /boot/image.ub
```

The reversing of the `fw_version` is, I think, endianness being detected incorrectly, I'd guess because the `program` command in python is not succeeding and so is bailing out before completing detection.

mitchburnett, 2024-04-09:
> I see this: #log info 1712674559.102 unknown Writing\_'tcpborphserver.bin'\_to\_fpga_manager, maybe this needs updating too?

There isn't an update that needs to happen here. Under the hood, the bitstream used to program the board is sent from the client to TBS becoming `tcpborphserver.bin`, and is then sent off to `fpga_manager` for programming.

Testing how far the DTO overlay gets is a bit sticky, admittedly the log messages could be more helpful to know what about applying the DTO failed. If the result of reading `/configfs/device-tree/overlays/casper/status` is anything other than `applied` will cause "applying the dto" to fail. This typically happens if the dtbo doesn't make sense to `fpga_manager`. We could debug this further, although, part of me is suspect that this is a byproduct of some other issue. Some quick things though would be to `cat  /configfs/device-tree/overlays/casper/status` to get what the status is. I want to recall there are 4 potential values? it may give more info. You can also open a serial console at the board and look at the kernel messages that pop up on a session there. Or, tail `dmesg` as you program. There will be some messages from `fpga_manager` about the status of the process.

After copying the updated library to the new location, what are the ownership/permissions set on the server?

sr-cdf, 2024-04-10:
@jack-h My `/boot` is the same as yours:

```(py3venv) casper@localhost:~$ sudo md5sum /boot/*
[sudo] password for casper: 
42120053bd949cc0045b6cebc319ba3c  /boot/BOOT.BIN
6139b67a21165152ec3198705328925b  /boot/boot.scr
2998b52455ec9b28a0802f0ac995faa8  /boot/image.ub
```

Could this be a python module version issue? Here are my pip installed packages and versions:

```
(py3venv) casper@localhost:~$ pip list
Package            Version
------------------ --------------------------
async-timeout      4.0.2
backcall           0.2.0
casperfpga         0.4.4.dev1371+head.453f556
certifi            2024.2.2
charset-normalizer 2.0.12
circus             0.16.1
crcmod             1.7
cycler             0.11.0
Cython             3.0.10
decorator          5.1.1
future             1.0.0
httplib2           0.22.0
idna               3.6
importlib-metadata 4.8.3
ipython            7.16.3
ipython-genutils   0.2.0
jedi               0.17.0
katcp              0.9.3
katversion         1.2
kiwisolver         1.3.1
Mako               1.1.6
Markdown           3.3.7
MarkupSafe         2.0.1
matplotlib         3.3.4
numpy              1.19.5
odict              1.9.0
packaging          21.3
parso              0.8.4
pdoc3              0.10.0
pexpect            4.9.0
pickleshare        0.7.5
Pillow             8.4.0
pip                21.3.1
pkg_resources      0.0.0
ply                3.11
progressbar        2.5
progressbar2       3.55.0
prompt-toolkit     3.0.36
psutil             5.9.8
ptyprocess         0.7.0
Pygments           2.14.0
pyparsing          3.1.2
python-dateutil    2.9.0.post0
python-utils       3.5.2
PyYAML             6.0.1
pyzmq              25.1.2
redis              4.3.6
remoteobjects      1.2.1
requests           2.27.1
scipy              1.5.4
setuptools         39.0.1
simplejson         3.19.2
six                1.16.0
souk-mkid-readout  0.0.1+v6.3.1.1.2.g81677baf
termcolor          1.1.0
tftpy              0.8.2
tornado            4.5.3
traitlets          4.3.3
typing_extensions  4.1.1
urllib3            1.26.18
waitress           2.0.0
wcwidth            0.2.13
wheel              0.37.1
zipp               3.6.0

```

 

@mitchburnett I do not have `/configfs`:
```
(py3venv) root@localhost:~# ls -l /
total 68
drwxr-xr-x   2 root root  4096 Aug 17  2023 bin
drwxrwxr-x   2 root root  4096 Jan  1  1970 boot
drwxr-xr-x  13 root root 13900 Apr  9 16:05 dev
drwxr-xr-x  78 root root  4096 Feb 29 16:13 etc
drwxr-xr-x   3 root root  4096 Jul 13  2022 home
drwxr-xr-x   2 root root  4096 Jul 13  2022 include
drwxr-xr-x  14 root root  4096 Jul 18  2022 lib
drwx------   2 root root  4096 Mar  5 14:45 lost+found
drwxr-xr-x   2 root root  4096 Aug  6  2020 media
drwxr-xr-x   2 root root  4096 Aug  6  2020 mnt
drwxr-xr-x   3 root root  4096 Aug 16  2023 opt
dr-xr-xr-x 157 root root     0 Apr  9 16:05 proc
drwx------   7 root root  4096 Feb 29 15:35 root
drwxr-xr-x  14 root root   480 Apr 10 09:02 run
drwxr-xr-x   2 root root  4096 Aug 16  2023 sbin
drwxr-xr-x   2 root root  4096 Aug  6  2020 srv
dr-xr-xr-x  13 root root     0 Apr  9 16:05 sys
-rwx------   1 root root  2308 Aug 17  2023 tcpborphserver.dtbo
drwxrwxrwt  11 root root  4096 Apr 10 09:02 tmp
drwxr-xr-x  11 root root  4096 Aug 16  2023 usr
drwxr-xr-x  11 root root  4096 Aug  6  2020 var

```
 but under `/sys/modules/configfs` there is: 

```
(py3venv) root@localhost:~# ls -l /sys/module/configfs/
total 0
--w------- 1 root root 4096 Apr  9 16:05 uevent
-r--r--r-- 1 root root 4096 Apr 10 09:06 version
(py3venv) root@localhost:~# cat /sys/module/configfs/version
0.0.2
(py3venv) root@localhost:~# cat /sys/module/configfs/uevent 
cat: /sys/module/configfs/uevent: Permission denied
```

`dmesg` only reports one line when programming:
```
[61722.966149] fpga_manager fpga0: writing tcpborphserver.bin to Xilinx ZynqMP FPGA Manager
```
```
(py3venv) root@localhost:~# dmesg | grep fpga
[    2.970469] fpga_manager fpga0: Xilinx ZynqMP FPGA Manager registered
[    3.131334] of-fpga-region fpga-full: FPGA Region probed
[    7.722533] fpga_manager fpga0: writing top.bit to Xilinx ZynqMP FPGA Manager
[  126.060922] fpga_manager fpga0: writing tcpborphserver.bin to Xilinx ZynqMP FPGA Manager
[61064.321824] fpga_manager fpga0: writing tcpborphserver.bin to Xilinx ZynqMP FPGA Manager
[61722.966149] fpga_manager fpga0: writing tcpborphserver.bin to Xilinx ZynqMP FPGA Manager
```

Persmissions for everything in tcpborphserver3 are:
```
(py3venv) casper@localhost:~$ ls -l /usr/local/sbin/
total 6248
-rwxr-xr-x 1 root root 2070232 Aug 16  2023 kcs
**-rwxr-xr-x 1 root root 2527280 Apr  9 15:43 tcpborphserver3**
-rwxr-xr-x 1 root root 1788144 Aug 16  2023 tmon
-rwxr-xr-x 1 root root    3863 Aug  6  2020 unminimize

```

jack-h, 2024-04-10:
From my board:
```
(py3venv) casper@localhost:~/src/souk-firmware/software/zynq/katcp_devel/tcpborphserver3$ ls -l /configfs
lrwxrwxrwx 1 root root 19 Feb  1 13:25 /configfs -> /sys/kernel/config/
```

sr-cdf, 2024-04-10:
That symlink is not generated on my board with the newer versions of tcpborphserver.

It is generated if I reboot with the older version.

If I create it myself, the board programs nicely.

I wonder why it isn't generated automatically on my board with the newest version.

sr-cdf, 2024-04-10:
Interestingly, the symlink now persists after reboots :-/

jack-h, 2024-04-10:
> I just need to set the mixer NCO frequency to Fs*3/4...

There is a `rfdc-update-nco` katcp command, but I don't think it has propagated to casperfpga. I'm also not sure if the behaviour will actually change since the ADC/DAC are set up to use the coarse (multiples of fs/4) LOs rather than the general NCO.

Recompiling using the NCO (but setting this to fs/4 at build time) would probably make this interface start working. And it could be propagated to casperfpga.

Could probably do this today if you really want it?

sr-cdf, 2024-04-10:
Yeah, I tried that but the mixer is in coarse mode, so the fine frequency setting doesn't apply.

The API in the Pynq image offers on the fly adjustment of the mixer settings, including the switch to fine nco mode, so it should be possible to implement this here.

I might give it a go unless @mitchburnett knows of any reason why it should not work.

In the mean time, if you wanted to compile a version with fine mixer nco as default, I would definitely find it useful.

jack-h, 2024-04-10:
> Yeah, I tried that but the mixer is in coarse mode, so the fine frequency
> setting doesn't apply.
>
> The API in the Pynq image offers on the fly adjustment of the mixer
> settings, including the switch [...]

I'll let Mitch comment, but I'd be very surprised if this was difficult.
Seems like chatgpt should be able to expose the whole API via katcp
commands....

> In the mean time, if you wanted to compile a version with fine mixer nco
> as default, I would definitely find it useful.

I'll do this .

jack-h, 2024-04-11:
Just uploaded this to the `main` branch. Tagged `v6.3.2`

jack-h, 2024-04-11:
Also present in the new `v7.1`-tagged firmware

sr-cdf, 2024-04-15:
With `v6.3.2`, the fine mixer setting is now working. 

However, I cannot set the DAC mixer frequency as this is not implemented in tcpborphserver.

Telnet log:
```
?rfdc-update-nco 0 0 1328.8 adc
#log info 1713193026.364 raw request\_set\_nco\_freq:\_1328.800\_MHz
!rfdc-update-nco ok

?rfdc-update-nco 0 0 1328.8 dac
#log info 1713192885.755 raw request\_set\_nco\_freq:\_1328.800\_MHz
#log error 1713192885.755 raw setting\_dac\_mixer\_nco\_not\_implemented
!rfdc-update-nco invalid
```

The fix is to implement the dac mixer nco setting in tcpborphserver, I'll take a look this week.

mitchburnett, 2024-04-15:
@sr-cdf and @jack-h I can get DAC mixer control implemented. More generally, I will clean up everything related to controlling the mixer in both fine and coarse modes. There has been previous work done already for toggling between fine and coarse mixer. It had fallen by the wayside and delayed when testing effects on maintaining multi-chip synchronization (multiple rfsocs working together) and working out nuances to support that use case. We can forgo that here and just get what has been done pulled in.

jack-h, 2024-04-15:
If you can point me to what I should merge I can do that and rebuild (or
Sam can)

mitchburnett, 2024-04-22:
@jack-h and @sr-cdf that mixer control for both [`tcpborphserver` ](https://gitlab.ras.byu.edu/alpaca/casper/katcp/-/tree/rfsoc/rfdc-dsa-vop?ref_type=heads) and [`casperfpga`](https://github.com/mitchburnett/casperfpga/tree/rfsocs/rfdc-dsa-vop) control has been merged into my branches.

sr-cdf, 2024-04-23:
Thanks @mitchburnett, I'll take a look now...

sr-cdf, 2024-04-23:
`casperfpga` installs fine, but `tcpborphserver` fails:

```

[sam@sam1 souk]$ git clone https://gitlab.ras.byu.edu/alpaca/casper/katcp.git
Cloning into 'katcp'...
remote: Enumerating objects: 10556, done.
remote: Counting objects: 100% (56/56), done.
remote: Compressing objects: 100% (51/51), done.
remote: Total 10556 (delta 38), reused 10 (delta 5), pack-reused 10500
Receiving objects: 100% (10556/10556), 3.44 MiB | 3.43 MiB/s, done.
Resolving deltas: 100% (7776/7776), done.

[sam@sam1 souk]$ cd katcp/

[sam@sam1 katcp]$ git checkout rfsoc/rfdc-dsa-vop
branch 'rfsoc/rfdc-dsa-vop' set up to track 'origin/rfsoc/rfdc-dsa-vop'.
Switched to a new branch 'rfsoc/rfdc-dsa-vop'

[sam@sam1 katcp]$ make
make -C katcp KATCP=../katcp all
make[1]: Entering directory '/home/sam/souk/katcp/katcp'
gcc -Wall -O2 -DKATCP_STRICT_CONFORMANCE -DKATCP_USE_FLOATS -DKATCP_ENABLE_LLINT -DKATCP_CONSISTENCY_CHECKS -DKATCP_MEMSTATS -DKATCP_STDERR_ERRORS -DKATCP_LOG_REQUESTS -DKATCP_UNIQUE_CLIENT_NAMES -DKATCP_SUBPROCESS -DKATCP_EXPERIMENTAL -DKATCP_DEPRECATED -DKATCP_HEAP_TIMERS -DKATCP_TRAP_TERM -DKATCP_PROTOCOL_MAJOR_VERSION=5 -DKATCP_PROTOCOL_MINOR_VERSION=0 -ggdb -fPIC  -DVERSION=\"v0.2.0-298-gdc92d39\" -DBUILD=\"2024-04-23T12:01:54\" -c line.c -o line.o -I.
gcc -Wall -O2 -DKATCP_STRICT_CONFORMANCE -DKATCP_USE_FLOATS -DKATCP_ENABLE_LLINT -DKATCP_CONSISTENCY_CHECKS -DKATCP_MEMSTATS -DKATCP_STDERR_ERRORS -DKATCP_LOG_REQUESTS -DKATCP_UNIQUE_CLIENT_NAMES -DKATCP_SUBPROCESS -DKATCP_EXPERIMENTAL -DKATCP_DEPRECATED -DKATCP_HEAP_TIMERS -DKATCP_TRAP_TERM -DKATCP_PROTOCOL_MAJOR_VERSION=5 -DKATCP_PROTOCOL_MINOR_VERSION=0 -ggdb -fPIC  -DVERSION=\"v0.2.0-298-gdc92d39\" -DBUILD=\"2024-04-23T12:01:54\" -c netc.c -o netc.o -I.
gcc -Wall -O2 -DKATCP_STRICT_CONFORMANCE -DKATCP_USE_FLOATS -DKATCP_ENABLE_LLINT -DKATCP_CONSISTENCY_CHECKS -DKATCP_MEMSTATS -DKATCP_STDERR_ERRORS -DKATCP_LOG_REQUESTS -DKATCP_UNIQUE_CLIENT_NAMES -DKATCP_SUBPROCESS -DKATCP_EXPERIMENTAL -DKATCP_DEPRECATED -DKATCP_HEAP_TIMERS -DKATCP_TRAP_TERM -DKATCP_PROTOCOL_MAJOR_VERSION=5 -DKATCP_PROTOCOL_MINOR_VERSION=0 -ggdb -fPIC  -DVERSION=\"v0.2.0-298-gdc92d39\" -DBUILD=\"2024-04-23T12:01:54\" -c dispatch.c -o dispatch.o -I.
dispatch.c: In function ‘append_end_katcp’:
dispatch.c:2304:12: warning: implicit declaration of function ‘append_end_flat_katcp’; did you mean ‘append_args_flat_katcp’? [-Wimplicit-function-declaration]
 2304 |     return append_end_flat_katcp(d);
      |            ^~~~~~~~~~~~~~~~~~~~~
      |            append_args_flat_katcp
gcc -Wall -O2 -DKATCP_STRICT_CONFORMANCE -DKATCP_USE_FLOATS -DKATCP_ENABLE_LLINT -DKATCP_CONSISTENCY_CHECKS -DKATCP_MEMSTATS -DKATCP_STDERR_ERRORS -DKATCP_LOG_REQUESTS -DKATCP_UNIQUE_CLIENT_NAMES -DKATCP_SUBPROCESS -DKATCP_EXPERIMENTAL -DKATCP_DEPRECATED -DKATCP_HEAP_TIMERS -DKATCP_TRAP_TERM -DKATCP_PROTOCOL_MAJOR_VERSION=5 -DKATCP_PROTOCOL_MINOR_VERSION=0 -ggdb -fPIC  -DVERSION=\"v0.2.0-298-gdc92d39\" -DBUILD=\"2024-04-23T12:01:54\" -c loop.c -o loop.o -I.
gcc -Wall -O2 -DKATCP_STRICT_CONFORMANCE -DKATCP_USE_FLOATS -DKATCP_ENABLE_LLINT -DKATCP_CONSISTENCY_CHECKS -DKATCP_MEMSTATS -DKATCP_STDERR_ERRORS -DKATCP_LOG_REQUESTS -DKATCP_UNIQUE_CLIENT_NAMES -DKATCP_SUBPROCESS -DKATCP_EXPERIMENTAL -DKATCP_DEPRECATED -DKATCP_HEAP_TIMERS -DKATCP_TRAP_TERM -DKATCP_PROTOCOL_MAJOR_VERSION=5 -DKATCP_PROTOCOL_MINOR_VERSION=0 -ggdb -fPIC  -DVERSION=\"v0.2.0-298-gdc92d39\" -DBUILD=\"2024-04-23T12:01:54\" -c log.c -o log.o -I.
gcc -Wall -O2 -DKATCP_STRICT_CONFORMANCE -DKATCP_USE_FLOATS -DKATCP_ENABLE_LLINT -DKATCP_CONSISTENCY_CHECKS -DKATCP_MEMSTATS -DKATCP_STDERR_ERRORS -DKATCP_LOG_REQUESTS -DKATCP_UNIQUE_CLIENT_NAMES -DKATCP_SUBPROCESS -DKATCP_EXPERIMENTAL -DKATCP_DEPRECATED -DKATCP_HEAP_TIMERS -DKATCP_TRAP_TERM -DKATCP_PROTOCOL_MAJOR_VERSION=5 -DKATCP_PROTOCOL_MINOR_VERSION=0 -ggdb -fPIC  -DVERSION=\"v0.2.0-298-gdc92d39\" -DBUILD=\"2024-04-23T12:01:54\" -c time.c -o time.o -I.
gcc -Wall -O2 -DKATCP_STRICT_CONFORMANCE -DKATCP_USE_FLOATS -DKATCP_ENABLE_LLINT -DKATCP_CONSISTENCY_CHECKS -DKATCP_MEMSTATS -DKATCP_STDERR_ERRORS -DKATCP_LOG_REQUESTS -DKATCP_UNIQUE_CLIENT_NAMES -DKATCP_SUBPROCESS -DKATCP_EXPERIMENTAL -DKATCP_DEPRECATED -DKATCP_HEAP_TIMERS -DKATCP_TRAP_TERM -DKATCP_PROTOCOL_MAJOR_VERSION=5 -DKATCP_PROTOCOL_MINOR_VERSION=0 -ggdb -fPIC  -DVERSION=\"v0.2.0-298-gdc92d39\" -DBUILD=\"2024-04-23T12:01:54\" -c shared.c -o shared.o -I.
gcc -Wall -O2 -DKATCP_STRICT_CONFORMANCE -DKATCP_USE_FLOATS -DKATCP_ENABLE_LLINT -DKATCP_CONSISTENCY_CHECKS -DKATCP_MEMSTATS -DKATCP_STDERR_ERRORS -DKATCP_LOG_REQUESTS -DKATCP_UNIQUE_CLIENT_NAMES -DKATCP_SUBPROCESS -DKATCP_EXPERIMENTAL -DKATCP_DEPRECATED -DKATCP_HEAP_TIMERS -DKATCP_TRAP_TERM -DKATCP_PROTOCOL_MAJOR_VERSION=5 -DKATCP_PROTOCOL_MINOR_VERSION=0 -ggdb -fPIC  -DVERSION=\"v0.2.0-298-gdc92d39\" -DBUILD=\"2024-04-23T12:01:54\" -c misc.c -o misc.o -I.
gcc -Wall -O2 -DKATCP_STRICT_CONFORMANCE -DKATCP_USE_FLOATS -DKATCP_ENABLE_LLINT -DKATCP_CONSISTENCY_CHECKS -DKATCP_MEMSTATS -DKATCP_STDERR_ERRORS -DKATCP_LOG_REQUESTS -DKATCP_UNIQUE_CLIENT_NAMES -DKATCP_SUBPROCESS -DKATCP_EXPERIMENTAL -DKATCP_DEPRECATED -DKATCP_HEAP_TIMERS -DKATCP_TRAP_TERM -DKATCP_PROTOCOL_MAJOR_VERSION=5 -DKATCP_PROTOCOL_MINOR_VERSION=0 -ggdb -fPIC  -DVERSION=\"v0.2.0-298-gdc92d39\" -DBUILD=\"2024-04-23T12:01:54\" -c server.c -o server.o -I.
gcc -Wall -O2 -DKATCP_STRICT_CONFORMANCE -DKATCP_USE_FLOATS -DKATCP_ENABLE_LLINT -DKATCP_CONSISTENCY_CHECKS -DKATCP_MEMSTATS -DKATCP_STDERR_ERRORS -DKATCP_LOG_REQUESTS -DKATCP_UNIQUE_CLIENT_NAMES -DKATC [... cut ...] 2  -c main.c -o main.o -I../katcp -I../rfdc_v8.1
main.c:4:10: fatal error: stdio.h: No such file or directory
    4 | #include <stdio.h>
      |          ^~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:36: main.o] Error 1
make[1]: Leaving directory '/home/sam/souk/katcp/tcpborphserver3'
make: *** [Makefile:28: tcpborphserver3-all] Error 2
[sam@sam1 katcp]$ 

```

`./tcpborphserver3/Makefile` refers to `../sysroot2020_2` and `../rfdc_v8.1`, which I don't have.

@jack-h do you have these?

sr-cdf, 2024-04-23:
Clearly this was doomed to fail as I was compiling it on my laptop and not on the rfsoc.

I'll try again on the rfsoc but using jacks katcp_devel branch, with the new additions merged in. 

That should work.

sr-cdf, 2024-04-23:
It worked.

For `casperfpga`, I installed the latest version of `souk-firmware/software/casperfpga`, but replacing the `rfdc.py` wit the one from @mitchburnett. Followed by the usual `pip install --upgrade .`.

For `tcpborphserver`, I built the latest version of `souk-firmware/software/zync/katcp_devel`, replacing `raw.c`, `rfsoc.h`, and `rfsoc.c` in `tcpborphserver3`. Then I had to set `IS_RFSOC ?= 1` in `tcpborphserver3/Makefile` and had to copy `souk-firmware/software/zync/librdfc.so.8.1` to `souk-firmware/software/zync/katcp_devel/rfdc/librfdc.a`. After running make in `katcp_devel/katcp` and in `katcp_devel/tcpborphserver3`, I copied the new binary to `/usr/local/sbin` and rebooted.

I'm certain there is a better way to do this, but for now its doing the job and I can set the DAC and ADC fine mixer frequencies, and the Nyquist zone with `souk-firmware` using `casperfpga`.

Thanks @mitchburnett, I'll let you if I run into any issues with the new API commands.

@jack-h you may want to go ahead and incorporate these changes this in a more sophisticated way.

sr-cdf, 2024-04-24:
@mitch, FYI, 

In `casperfpga/src/rfdc.py`, lines similar to this crop up a lot:

`args = (ntile, "adc" if converter_type == self.ADC_TILE  else "dac")`

which works fine if `converter_type` is the integer `self.ADC_TILE` or `self.DAC_TILE` but not if its a string of `'adc'` or `'dac'` as described in the various docstrings.

sr-cdf, 2024-06-14:
Closing this as the basic functionality is there and it works.

closed 2024-06-14 by sr-cdf

jack-h, 2024-07-16:
@sr-cdf FYI, with the latest `software/zynq/katcp_devel` submodule you _should_ be able to simply do (on the rfsoc):
```
cd tcpborphserver
IS_RFSOC=1 make
sudo make install
```

and reboot to get the library compiled and installed