realtimeradio/souk-firmware #45: measued phases not consistent in v7.1

sr-cdf, 2024-04-29:
The average value of the phase of a tone measured in the accumulator (loopback enabled) was repeatable in v6.3 but now appears to change randomly in v7.1.

Eg:

v6:
1. program the board, set a tone, read the accumulator, get a value for phase.
2. change the tone, read the accumulator, get a different value of phase.
3. change back to previous tone (sample amplitude and offset), and get back the original value again.

v7:
1. program the board, set a tone, read the accumulator, get a value for phase
2. change the tone, read the accumulator, get a different value of phase,
3. change back to previous tone (sample amplitude and offset), and get a completely different value for the phase. 

This is problematic when performing frequency sweeps and resonator tracking as we depend on the system not adding arbitrary random phase offsets when setting tones.

Note that the process of setting a new tone that is identical in every way to a tone that is already set does not change the phase. 

Test code:

```import sys,os,time
import numpy as np

import souk_mkid_readout
r = souk_mkid_readout.SoukMkidReadout('krm4',fpgfile='/home/sam/souk/souk-firmware/firmware/src/souk_single_pipeline_krm/outputs/souk_single_pipeline_krm.fpg')

r.program()
r.initialize()
r.adc_clk_hz =  2457600000
r.input.enable_loopback()
r.output.use_psb()
r.psb.set_fftshift(0b1)
r.pfb.set_fftshift(0b11111111)
r.accumulators[0].set_acc_len(1000)

ch     = 0
freqs  = np.array([1e9])
amps   = np.array([1.0])
phases = np.array([0.0])

#set tone and immediately measure get_new_spectra
r.set_multi_tone(np.atleast_1d(freqs),amplitudes=np.atleast_1d(amps),phase_offsets_rads=np.atleast_1d(phases))
z = r.accumulators[0].get_new_spectra()
print(freqs[ch],np.abs(z[ch]),np.angle(z[ch]))

#wait a bit and measure again
time.sleep(2)
z = r.accumulators[0].get_new_spectra()
print(freqs[ch],np.abs(z[ch]),np.angle(z[ch]))

#set the tone again and measure
r.set_multi_tone(np.atleast_1d(freqs),amplitudes=np.atleast_1d(amps),phase_offsets_rads=np.atleast_1d(phases))
z = r.accumulators[0].get_new_spectra()
print(freqs[ch],np.abs(z[ch]),np.angle(z[ch]))

#wait a bit and measure again
time.sleep(2)
z = r.accumulators[0].get_new_spectra()
print(freqs[ch],np.abs(z[ch]),np.angle(z[ch]))

#set an new tone and measure:
freqs+=1000
r.set_multi_tone(np.atleast_1d(freqs),amplitudes=np.atleast_1d(amps),phase_offsets_rads=np.atleast_1d(phases))
z = r.accumulators[0].get_new_spectra()
print(freqs[ch],np.abs(z[ch]),np.angle(z[ch]))

#wait a bit and measure again
time.sleep(2)
z = r.accumulators[0].get_new_spectra()
print(freqs[ch],np.abs(z[ch]),np.angle(z[ch]))
[... 114 lines cut ...]
2024-04-29 11:00:02,641 - souk_mkid_readout.blocks.block:krm4:p0_acc0 - WARNING - Accumulation counter changed while reading data!
1000000000.0 261969813.65526235 <b>0.9520980236587934</b>
2024-04-29 11:00:02,911 - souk_mkid_readout.blocks.block:krm4:p0_acc0 - WARNING - Accumulation counter changed while reading data!
1000000000.0 261970545.51222906 <b>0.9520664243310474</b>
2024-04-29 11:00:04,934 - souk_mkid_readout.blocks.block:krm4:p0_acc0 - WARNING - Accumulation counter changed while reading data!
1000000000.0 261971118.5526834 <b>0.9520851367434449</b>
2024-04-29 11:00:05,202 - souk_mkid_readout.blocks.block:krm4:p0_acc0 - WARNING - Accumulation counter changed while reading data!
1000001000.0 261984568.22841656 <b>-2.1162183570156787</b>
2024-04-29 11:00:07,225 - souk_mkid_readout.blocks.block:krm4:p0_acc0 - WARNING - Accumulation counter changed while reading data!
1000001000.0 261984380.90371767 <b>-2.1162154422879165</b>
2024-04-29 11:00:07,384 - souk_mkid_readout.blocks.block:krm4:p0_acc0 - WARNING - Accumulation counter changed while reading data!
1000000000.0 261969784.09465718 <b>-1.5192828183601694</b>
2024-04-29 11:00:09,406 - souk_mkid_readout.blocks.block:krm4:p0_acc0 - WARNING - Accumulation counter changed while reading data!
1000000000.0 261967662.00210956 <b>-1.5192836544268882</b>
antname: None
fw_build_time: 2024-04-11T14:59:19
fw_supported: True
fw_type: 2
fw_version: 7.1.0.0
host: sam1:krm4
programmed: True
sw_version: 0.0.1+v7.1-dirty
timestamp: 2024-04-29T11:00:09.410890

</pre>

sr-cdf, 2024-05-10:
@jack-h thanks for the suggestion to arm and trigger the sync, this fixes this issue.

The only remaining minor inconvenience is that `sync.arm_sync()` always waits over a second for a pulse, even when we are going to be performing a software controlled sync. When trying to sweep over many frequencies quickly, we will need to disable this wait time.

jack-h, 2024-05-10:
oh, that is annoying and easy to fix. Standby

jack-h, 2024-05-10:
Does `arm_sync(wait=False)` save you a ~second.

I also just commited a tweak 770e404bc7 so `sw_sync()` also takes a `wait` argument which, if False, will save a little time.

sr-cdf, 2024-05-10:
Oh, pip is failing with 
```
(venv-remote-control) [sam@sam1 control_sw]$ pip install --upgrade .
Processing /home/sam/souk/souk-firmware/software/control_sw
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: souk-mkid-readout
  Building wheel for souk-mkid-readout (setup.py) ... done
  Created wheel for souk-mkid-readout: filename=souk_mkid_readout-v7.1.1_9_g770e404b_dirty-py3-none-any.whl size=64127 sha256=a6ed2f25550bb57928664674cf7c3b6d9a6d595233be14d3442f58b9138b95a8
  Stored in directory: /tmp/pip-ephem-wheel-cache-t29d1xmv/wheels/89/d7/71/b3ef402edb163dd096f266591a56b9893e7c69f7d163e66c62
  WARNING: Built wheel for souk-mkid-readout is invalid: Metadata 1.2 mandates PEP 440 version, but 'v7.1.1-9-g770e404b-dirty' is not
Failed to build souk-mkid-readout
ERROR: Could not build wheels for souk-mkid-readout, which is required to install pyproject.toml-based projects
```

jack-h, 2024-05-10:
I get a failure to build the wheel, but installation still works. What versions are you on?

```
(py3-venv) jackh@rtr-dev3:~/src/souk-firmware2/souk-firmware/software/control_sw$ pip --version
pip 20.0.2 from /home/jackh/py3-venv/lib/python3.8/site-packages/pip (python 3.8)
(py3-venv) jackh@rtr-dev3:~/src/souk-firmware2/souk-firmware/software/control_sw$ python --version
Python 3.8.10
```

sr-cdf, 2024-05-10:
It fails to build at all with these versions:
```(venv-remote-control) [sam@sam1 control_sw]$ pip --version
pip 24.0 from /home/sam/souk/venv-remote-control/lib64/python3.10/site-packages/pip (python 3.10)
(venv-remote-control) [sam@sam1 control_sw]$ python --version
Python 3.10.11
```
It succeeds after downgrading pip to 20.0.2.

sr-cdf, 2024-05-10:
wait=False on arm and sync speeds things up nicely.

Still a bit slower than without the sync, but I can live with it for now.

jack-h, 2024-05-10:
I think doing better probably needs a cleaner picture of the requirements of the TX vs RX LOs.
For example, one way to do this would be to have, rather than a TX and RX LO, an RX LO and an "offset" LO, which shifts the TX LO. This way you could sweep by only manipulating a single LO.

sr-cdf, 2024-05-10:
> I think doing better probably needs a cleaner picture of the requirements of the TX vs RX LOs.

Sure, and in my head there are different use cases with separate requirements.

> For example, one way to do this would be to have, rather than a TX and RX LO, an RX LO and an "offset" LO, which shifts the TX LO. This way you could sweep by only manipulating a single LO.

I imagine this would be ok for the case of small offsets of much less than a filterbank channel width, but would fall over when trying to operate between channels.

And it wouldn't be so good if for example if the rx spectrum is flipped wrt the tx spectrum.

Shall we leave it for now until the actual requirements are more well defined?

jack-h, 2024-05-10:
It will not shock you to learn that is my preference :)

closed 2024-05-23 by jack-h