realtimeradio/souk-firmware #48: mixer.set_dependent_tx failing to give warning

sr-cdf, 2024-05-16:
```
In [19]: r.mixer.set_dependent_tx(sync=True, offset=False, scale=False, step=False)
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[19], line 1
----> 1 r.mixer.set_dependent_tx(sync=True, offset=False, scale=False, step=False)

File ~/souk/venv-remote-control/lib64/python3.10/site-packages/souk_mkid_readout/blocks/mixer.py:370, in Mixer.set_dependent_tx(self, sync, offset, scale, step)
    354 def set_dependent_tx(self, sync=True, offset=True, scale=True, step=True):
    355     """
    356     Set the TX LOs to use dependent control signals from the RX LOs.
    357 
   (...)
    368     :type step: bool
    369     """
--> 370     self.set_independent_tx(sync=not sync, offset=not offset, scale=not scale, step=not step)

File ~/souk/venv-remote-control/lib64/python3.10/site-packages/souk_mkid_readout/blocks/mixer.py:345, in Mixer.set_independent_tx(self, sync, offset, scale, step)
    343 if not sync:
    344     if (offset or scale or step):
--> 345         self.warning('Using non-shared sync, but sharing another control signal. This is unlikely to be what you want.')
    346 ctrl = 0
    347 ctrl += (int(sync) << self._IND_SYNC_OFFSET)

AttributeError: 'Mixer' object has no attribute 'warning'
```

sr-cdf, 2024-05-16:
should be `self.logger.warning`

jack-h, 2024-05-23:
fixed (avoided) in v7.2

closed 2024-05-23 by jack-h