realtimeradio/souk-firmware #49: Setting dependent tx kills the output

sr-cdf, 2024-05-16:
@jack-h 

Calling `r.mixer.set_dependent_tx()` kills the output as viewed on a spectrum analyser.

I've narrowed it down to the `scale` parameter as the culprit.

Any chance of a quick fix?

```
r.mixer.set_independent_tx()
# output tones are ok

r.mixer.set_independent_tx(sync=True, offset=True, scale=True, step=True)
# output tones are ok

r.mixer.set_independent_tx(sync=True, offset=True, scale=True, step=False)
# output tones are ok

r.mixer.set_independent_tx(sync=True, offset=True, scale=False, step=True)
# output tones disappear

r.mixer.set_independent_tx(sync=True, offset=False, scale=True, step=True)
# output tones are ok

```

Setting and resetting the either or both of the TX/RX LOs with `mixer.set_freqs`, and/or `mixer.set_amplitude_scale` does not bring the tones back when independent tx scale is False.

Neither does arming and triggering a software sync.

jack-h, 2024-05-16:
Think I have this one pinned down, just need to figure out what the fix is.

I think what is happening is that if the sync is shared between paths, this ends up being inconsistent with the channel selection IDs, to the point that the scale value gets assigned to the wrong bin.

jack-h, 2024-05-23:
fixed in v7.2

closed 2024-05-23 by jack-h