realtimeradio/souk-firmware #50: dependent tx is not equivalent to independent tx despite equal settings

sr-cdf, 2024-05-16:
With everything set to independent, behaviour is as expected.

With everything set to dependent, the all outputs are equal to zero.

With `step` and `offset` set to dependent, but 'sync' and 'scale' independent, there is a large drop in accumulated magnitude when the LO frequency is not zero.

Test script and console output below:

```
import sys,os,time
import numpy as np

import souk_mkid_readout
r = souk_mkid_readout.SoukMkidReadout('krm4',configfile='/home/sam/souk/souk-firmware/software/control_sw/config/souk-single-pipeline-krm-2G.yaml')

r.program()
r.initialize()
r.adc_clk_hz =  r.config['adc_clk_hz']

import logging
r.accumulators[0].logger.setLevel(logging.ERROR)

r.fpga.print_status()
print(r.config)

r.output.use_psb()
r.psb.set_fftshift(0b1)
r.pfb.set_fftshift(0b11111111)
r.accumulators[0].set_acc_len(1000)

chans  = [0,1]
freqs  = np.array([1e9,1.5555555555e9])
amps   = np.array([1.0,1.0])
phases = np.array([0.0,np.pi])

print()
print('Set independent tx:')
r.mixer.set_independent_tx(sync=True, offset=True, scale=True, step=True)
print('r.mixer.get_tx_independence():',r.mixer.get_tx_independence())

print('\tEnable loopback')
r.input.enable_loopback()
#set tone and immediately measure get_new_spectra
r.set_multi_tone(np.atleast_1d(freqs),amplitudes=np.atleast_1d(amps),phase_offsets_rads=np.atleast_1d(phases))
r.sync.arm_sync(wait=False)
r.sync.sw_sync(wait=False)
time.sleep(0.1)
z = r.accumulators[0].get_new_spectra()
for ch in chans:
    print('\tChannel %d:'%ch)
    print('\t\t','freq:',freqs[ch])
    print('\t\t','tx mixer lo (step,offset,scale)',r.mixer.get_phase_offset(ch,'tx'))
    print('\t\t','rx mixer lo (step,offset,scale)',r.mixer.get_phase_offset(ch,'rx'))
    print('\t\t','accumulated magnitude:',np.abs(z[ch]))
    print('\t\t','accumulated phase:',np.angle(z[ch]))

print('\tDisable loopback')
r.input.disable_loopback()
#set tone and immediately measure get_new_spectra
z = r.accumulators[0].get_new_spectra()
for ch in chans:
    print('\tChannel %d:'%ch)
    print('\t\t','freq:',freqs[ch])
    print('\t\t','tx mixer lo (step,offset,scale)',r.mixer.get_phase_offset(ch,'tx'))
    print('\t\t','rx mixer lo (step,offset,scale)',r.mixer.get_phase_offset(ch,'rx'))
    print('\t\t','accumulated magnitude:',np.abs(z[ch]))
    print('\t\t','accumulated phase:',np.angle(z[ch]))

print()
print('Set dependent tx:')
[... 236 lines cut ...]
                 rx mixer lo (step,offset,scale) (0.0, 0.0, 0.999755859375)
                 accumulated magnitude: 262045667.64211157
                 accumulated phase: 0.5154138787106566
        Channel 1:
                 freq: 1555555555.5
                 tx mixer lo (step,offset,scale) (0.6981310023356222, -3.141592653589793, 0.999755859375)
                 rx mixer lo (step,offset,scale) (0.6981310023356222, -3.141592653589793, 0.999755859375)
                 accumulated magnitude: 261800.6509254704
                 accumulated phase: 2.2358231085412297
        Disable loopback
        Channel 0:
                 freq: 1000000000.0
                 tx mixer lo (step,offset,scale) (0.0, 0.0, 0.999755859375)
                 rx mixer lo (step,offset,scale) (0.0, 0.0, 0.999755859375)
                 accumulated magnitude: 23883871.194216482
                 accumulated phase: 1.0576653946357848
        Channel 1:
                 freq: 1555555555.5
                 tx mixer lo (step,offset,scale) (0.6981310023356222, -3.141592653589793, 0.999755859375)
                 rx mixer lo (step,offset,scale) (0.6981310023356222, -3.141592653589793, 0.999755859375)
                 accumulated magnitude: 11940.151129696811
                 accumulated phase: 3.0206968823429974

In [2]: 
```

jack-h, 2024-05-23:
Is this still an issue with `v7.2` or was this all related to the other problems with independent tx/rx which `v7.2` attempts to fix?

sr-cdf, 2024-05-24:
This was dealt with by the changes in 7.2 that removed the dependent/independent modes of operation.

jack-h, 2024-05-24:
Fabulous

closed 2024-05-24 by jack-h