realtimeradio/souk-firmware #52: cannot get r.adc_clk_hz in 'local' mode

sr-cdf, 2024-05-17:
the new code in commit 5c315e2 fails when souk_mkid_readout is instantiated with local=True.

"""
In [5]: fpga_clk_hz = int(r.rfdc.core.get_fabric_clk_freq(0, 'adc') * 1e6)
   ...: 
   ...: 
   ...: 
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-5-d9fb810cb9ef> in <module>
----> 1 fpga_clk_hz = int(r.rfdc.core.get_fabric_clk_freq(0, 'adc') * 1e6)
      2 
      3 

~/py3venv/lib/python3.6/site-packages/casperfpga/rfdc.py in get_fabric_clk_freq(self, ntile, converter_type)
    461 
    462     args = (ntile, "adc" if converter_type == self.ADC_TILE  else "dac")
--> 463     reply, informs = t.katcprequest(name='rfdc-get-fab-clk-freq', request_timeout=t._timeout, request_args=args)
    464 
    465     info = informs[0].arguments[0].decode()

AttributeError: 'LocalMemTransport' object has no attribute 'katcprequest'
"""

sr-cdf, 2024-05-17:
Actually, it doesnt really fail, it catches the exception and it can get the adc_clk_hz from the config file so its.

So maybe not an issue.

jack-h, 2024-05-17:
> the adc_clk_hz from the config file so its fine.

Ah, yes, there's an edge case.

closed 2024-05-23 by jack-h