realtimeradio/souk-firmware #54: windowed accumulator, some kind of coefficient indexing bug?

sr-cdf, 2024-06-13:
Hey @jack-h 

Something appears to be up with the windowed accumulator.

I was investigating the attenuation of aliased frequencies with different window functions and everything sort of made sense with a rectangular window (all ones), but any other kind of window gave unexpected results.

I devised a test to see if all the coefficients were being applied correctly and found that for an accumulation length of 1000, only the first 250 (exactly n/4) have any effect on the output:

Setting the window to all zeroes [0,0,0,0...]  and measuring a single channel of the accumulated spectra gives exactly zero. 

Setting the window to [1,0,0,0...]  and measuring a single channel of the accumulated spectra gives some value. 

Setting the window to [1,1,0,0...] gives a value 2x higher, as i would expect. 

Setting [1,1,1,0...] gives a 3x higher value etc. 

After 250 ones, the value stops increasing, despite the window length being 1000.

When I apply a window function with coefficients [a,b,c,d,e.....], the actual measured frequency response of the accumulator appears consistent with a window that looks like either [a,0,0,0,b,0,0,0,c,0,0,0.....] or [a,a,a,a,b,b,b,b,....] , I'd need to look a bit closer to tell exactly which it is.

If I double the acc_len to 2000, and try again, it's the first 500 coefficients that are in play, confirming the n/4 dependence.

Could this be a new bug or is it likely to have been with us from the off?

This is tested on v7.2, souk_single_pipeline_krm/outputs/souk_single_pipeline_krm_2024-05-16_1629.fpg, the 2G version.

Example code to reproduce:

```

import sys,os,time
from numpy import *
from matplotlib.pyplot import *; ion()
from scipy import signal
import souk_mkid_readout

def set_dac_first_nyquist_zone(use_inv_sinc_filter=False):
    #disable dac mix-mode
    r.rfdc.core.set_nyquist_zone(0,0,r.rfdc.core.DAC_TILE,1)

    #use high pass image reject filer. Has no effect?
    r.rfdc.core.set_imr_mode(0,0,0)

    #mix baseband to center of 1st zone (+Fs*1/4, where Fs=2*r.adc_clk_hz)
    r.rfdc.core.set_fine_mixer_freq(0,0,r.rfdc.core.DAC_TILE,2*r.adc_clk_hz*1/4/1e6)

    #optionally set inverse sinc filter to level off the 0.5-1dB drop at the edges
    if use_inv_sinc_filter:
        r.rfdc.core.set_invsinc_fir(0,0,r.rfdc.core.INVSINC_FIR_NYQUIST1)
    else:
        r.rfdc.core.set_invsinc_fir(0,0,r.rfdc.core.INVSINC_FIR_DISABLED)

    return

def set_dac_second_nyquist_zone(use_inv_sinc_filter=False):
    #enable dac mix-mode
    r.rfdc.core.set_nyquist_zone(0,0,r.rfdc.core.DAC_TILE,2)

    #use high pass image reject filer. Has no effect?
    r.rfdc.core.set_imr_mode(0,0,1)

    #mix baseband to center of 2nd zone and flip (-Fs*3/4, where Fs=2*r.adc_clk_hz)
    r.rfdc.core.set_fine_mixer_freq(0,0,r.rfdc.core.DAC_TILE,-2*r.adc_clk_hz*3/4/1e6)

    #optionally set inverse sinc filter to level off the 0.5-1dB drop at the edges
    if use_inv_sinc_filter:
        r.rfdc.core.set_invsinc_fir(0,0,r.rfdc.core.INVSINC_FIR_NYQUIST2)
    else:
        r.rfdc.core.set_invsinc_fir(0,0,r.rfdc.core.INVSINC_FIR_DISABLED)

    return

def set_adc_first_nyquist_zone():
    #specify operation in first nyquist zone
    r.rfdc.core.set_nyquist_zone(0,0,r.rfdc.core.ADC_TILE,1)

    #mix center of 1st zone down to baseband (-Fs*1/4, where Fs=2*r.adc_clk_hz)
    r.rfdc.core.set_fine_mixer_freq(0,0,r.rfdc.core.ADC_TILE,-2*r.adc_clk_hz*1/4/1e6)

    assert int(r.rfdc.core.get_mixer_settings(0,0,r.rfdc.core.ADC_TILE)['Freq']*1e6) == int(-2*r.adc_clk_hz*1/4)

    return

def set_adc_second_nyquist_zone():
    #specify operation in second nyquist zone
    r.rfdc.core.set_nyquist_zone(0,0,r.rfdc.core.ADC_TILE,2)

    #mix center of 2nd zone down to baseband and flip (+Fs*3/4, where Fs=2*r.adc_clk_hz)
    r.rfdc.core.set_fine_mixer_freq(0,0,r.rfdc.core.ADC_TILE,+2*r.adc_clk_hz*3/4/1e6)

    assert int(r.rfdc.core.get_mixer_settings(0,0,r.rfdc.core.ADC_TILE)['Freq']*1e6) == int(2*r.adc_clk_hz*3/4)

    return

ACCLEN  = 1000 # 500 sps
ACCNUM = 0 #accumulator to read from
PFBLEN = 4096

# PSBSHIFT=2**5-1
PSBSHIFT=2**4-1

# PFBSHIFT=2**5-1
PFBSHIFT=2**4-1
PFBSHIFT=2**7-1

r = souk_mkid_readout.SoukMkidReadout('krm4',configfile='/home/sam/souk/souk-firmware/software/control_sw/config/souk-single-pipeline-krm-2G.yaml')

r.program()
r.initialize()
acc=r.accumulators[0]
r.sync.set_delay(6225)
r.sync.arm_sync(wait=False)
r.sync.sw_sync(wait=False)

r.output.use_psb()
r.accumulators[ACCNUM].set_acc_len(ACCLEN)
# r.psb.set_fftshift(2**2-1)
r.psb.set_fftshift(PSBSHIFT)
r.pfb.set_fftshift(PFBSHIFT)

r.rfdc.core.set_vop(0,0,20000)
r.rfdc.core.set_dsa(0,0,0)

accfreq = r.adc_clk_hz/(r.accumulators[ACCNUM]).get_acc_len()/PFBLEN
accfreqoff=accfreq

ch=0 #channel number for plotting etc

set_dac_first_nyquist_zone()
set_adc_first_nyquist_zone()

f1 = 1.0e9
freqs = np.array([f1])
num_tones = len(freqs)
amps = np.ones(num_tones)
np.random.seed(1)
phases = np.random.uniform(0,2*np.pi,num_tones)

r.set_multi_tone(np.atleast_1d(freqs),amplitudes=np.atleast_1d(amps),phase_offsets_rads=np.atleast_1d(phases))
time.sleep(0.1)
r.sync.arm_sync(wait=False)
time.sleep(0.1)
r.sync.sw_sync()

time.sleep(10)

ch=0
z=zeros(ACCLEN,dtype=complex)

for j in range(ACCLEN):
    print(j,ACCLEN)
    def ws(n):
        tmp=zeros(n)
        tmp[:j+1]=1
        return tmp

    r.accumulators[0].set_window(ws)
    time.sleep(0.01)
    z[j] = r.accumulators[0].get_new_spectra()[0]

figure()
plot(z.real)
figure()
plot(z.imag)

```

jack-h, 2024-06-14:
Thanks for the clear description and instructions -- I'll look at this next week and hopefully have an answer before you are back.

jack-h, 2024-07-23:
I think the issue here was all about the software mis-handling parallel samples.

With the above commit, the following stripped down version of your test:

```
import sys,os,time
from numpy import *
from matplotlib.pyplot import *;
import souk_mkid_readout

ACCLEN  = 128
ACCNUM = 0 # accumulator to read from
HOST = '10.11.0.217'
CONFIG = '/home/jackh/src/souk-firmware2/souk-firmware/software/control_sw/config/souk-single-pipeline-krm-2G.yaml'

r = souk_mkid_readout.SoukMkidReadout(HOST, configfile=CONFIG)

r.program()
r.initialize()
acc=r.accumulators[0]
r.sync.set_delay(6225)

r.pfbtvg.tvg_enable()
r.chanselect.set_channel_outmap(np.arange(r.chanselect.n_chans_out))
r.accumulators[ACCNUM].set_acc_len(ACCLEN)
r.mixer.enable_power_mode()
r.sync.arm_sync(wait=False)
r.sync.sw_sync(wait=False)

ch=3
z=zeros(ACCLEN,dtype=complex)

for j in range(ACCLEN):
    print(j,ACCLEN)
    def ws(n):
        tmp=zeros(n)
        tmp[:j+1]=1
        return tmp

    r.accumulators[0].set_window(ws)
    time.sleep(0.01)
    z[j] = r.accumulators[0].get_new_spectra()[ch]

figure()
plot(z.real, 'bo-')
plot(z.imag, 'ro-')
show()
```

returns

[image]

which I believe is correct.

I.e. a necessary limitation of the firmware is that every 4 samples (which the firmware processes in parallel) must have the same coefficient. The software previously mishandled this (and no coefficients after the first ACC_LEN/4 were ever used). This update does the closest it can to the Right Thing, and decimates provided coefficients.
Perhaps most importantly, the reported window (obtained with `get_window()`) should now report what coefficients are actually being applied (which will be a decimated, and then repeated, form of those set with `set_window()`)

jack-h, 2024-07-24:
@sr-cdf close at will if you're happy with this

sr-cdf, 2024-07-24:
Happy enough.

I don't really understand why the parallel samples in the accumulator all have to receive the same coefficient in the firmware.

But as it stands now, there is negligible difference between an N-length window and a decimated N/4-length window.

The following plots show a quick test of the measured power of a test signal as a function of its offset from the tone (points) compared with the expected response for the given window function (solid line). 

The first two are from before this fix, where the box-car worked as expected but the sinc was not-quite-right. The second two plots are after this fix and, pleasingly, the sinc looks just about right. 

The noise floor is indicated by the black line, and I dont expect the the data points to fall far below this. 

[image]

[image]

So, I think its fine to close.

jack-h, 2024-07-24:
Excellent.

The parallel sample coefficient sharing isn't fundamental so could
certainly be changed if necessary, it would just cost a little extra RAM
(since multiple coefficients would need to be fetched simultaneously,
requiring more bandwidth). There are minor free optimisations which could
also be made - e.g. taking the average coefficient is probably better than
straight decimation.

I wasn't quite sure this feature would be used at all, so I didn't think
too hard about it.

sr-cdf, 2024-07-24:
Right, makes sense.

If it turns out there is a benefit to be gained then I'll get back to you about it. 

For now I think this is done :)

closed 2024-07-24 by sr-cdf

sr-cdf, 2024-07-25:
For the record, the effect of the decimation can be seen when pushing the accumulator window function to higher bandwidths:

[image]

```
#accumulator window function setting software was updated.
#this is taken using fw 7.2

from numpy import *
from matplotlib.pyplot import *
ion()
from scipy import signal

fig,axes=subplots(3,3,sharey='col',sharex='col')
for i in range(3):
    fcw=[500,5000,50000][i]
    fsw=500000
    def ws(n):
        w=signal.firwin(
        n,
        fcw,
        width=None,
        window='hamming',
        pass_zero=True,
        fs=fsw)
        return w/w.max()
    r.accumulators[0].set_window(ws)
    axes[i,0].plot(ws(1000),alpha=0.5)
    axes[i,0].plot(r.accumulators[0].get_window(),alpha=0.5)
    axes[i,1].plot(ws(1000),alpha=0.5)
    axes[i,1].plot(r.accumulators[0].get_window(),alpha=0.5)
    w=zeros(500000)
    w[500000//2-500:500000//2+500]=ws(1000)
    fw=fw=abs(fft.fftshift(fft.fft(w,len(w))))
    fwf=fft.fftshift(fft.fftfreq(len(fw),1./500000))
    wact=zeros(500000)
    wact[500000//2-500:500000//2+500]= r.accumulators[ACCNUM].get_window()
    fwact=abs(fft.fftshift(fft.fft(wact,len(wact))))
    fwactf=fft.fftshift(fft.fftfreq(len(fwact),1./500000))
    axes[i,2].semilogy(fwf/1e3,fw,alpha=0.5)
    axes[i,2].semilogy(fwactf/1e3,fwact,alpha=0.5)

axes[0,2].set_ylim(1e-7,1e3)
axes[0,1].set_xlim(460,540)
axes[0,0].set_title('Full window')
axes[0,1].set_title('Zoom to center')
axes[0,2].set_title('Spectral response')
axes[0,0].set_ylabel('Sinc BW = 500 Hz')
axes[1,0].set_ylabel('Sinc BW = 5 kHz')
axes[2,0].set_ylabel('Sinc BW = 50 kHz')
[ax.grid(visible=True) for ax in ravel(axes)]
suptitle('Windowed accumulator, effect of N/4 decimation.\nBlue: requested window, Orange: actual window after decimation')
axes[2,0].set_xlabel('index')
axes[2,1].set_xlabel('index')
axes[2,2].set_xlabel('Frequency, kHz')
fig.set_size_inches([11.02,6.06])
tight_layout()
```

jack-h, 2024-07-25:
I suppose this makes sense in that the  coefficients change faster.

I suspect there's room to do something cunning here, so let me know if you
want to revisit this at some point.