realtimeradio/souk-firmware #55: Phase switching

jack-h, 2024-07-16:
Add phase switching!

jack-h, 2024-08-28:
Add blanking

jack-h, 2024-09-10:
@sr-cdf bitstreams now available (with much testing not done) in the v7.4 branch.
These _should_ switch by 180 degrees as discussed earlier in today's meeting.

jack-h, 2024-09-11:
It should be instantaneous, but is applied to the LO before the PSB, so the PSB will have the effect of smoothing the switch (I think).
Perhaps this isn't what is desired and the switch should be downstream of the PSB(?)

I have an electrician here right now and any time I try to do something he trips the RCD, but will get to looking at this soon.

jack-h, 2024-09-11:
To elaborate a little, I think if you set the switch period much longer than 1 spectrum you should at least see consistent power out

sr-cdf, 2024-09-11:
> To elaborate a little, I think if you set the switch period much longer than 1 spectrum you should at least see consistent power out

Yes, I see that, until there is a switch.

> Perhaps this isn't what is desired and the switch should be downstream of the PSB(?)

Maybe that is the way to go.

sr-cdf, 2024-09-11:
How the tone looks:

No switching:
[image]

1 spectra per step:
[image]

64 spectra per step:
[image]

jack-h, 2024-09-17:
Do you know what the target switching rate is likely to be?

I'll think a bit more about placing the switch after the PSB. It's easy in theory, but there needs to be a little thought into how to get the timing of the signals correct. I'll try and make firmware that lets you choose the switching mechanism at runtime, then it should be easy to play around with different configurations and see what works best

sr-cdf, 2024-09-17:
> *External email to Cardiff University - *Take care when replying/opening
> attachments or links.
> *Nid ebost mewnol o Brifysgol Caerdydd yw hwn - *Cymerwch ofal wrth
> ateb/agor atodiadau neu ddolenni.

I wish I could say I do but I don't really know what's going to happen when
the phase modulated signal hits the resonators.

Best guess is anywhere from 100 Hz to 100 kHz, give or take a lot.

Sorry