realtimeradio/souk-firmware #56: Feature request: DAC Snapshot

sr-cdf, 2024-07-17:
Hey @jack-h ,

Can you add a snapshot block right before the DAC, exactly like there is right after the ADC?

jack-h, 2024-07-18:
I can/will do, though of course you can also enable loopback and then use
the ADC snapshot

sr-cdf, 2024-07-18:
Thanks, I thought you might say that, i'd just be happier knowing i'm grabbing the exact signal that is heading to each dac. 

Is it possible to ensure that the snapshots for each of the 2 dacs in a single pipeline are synchronised?

jack-h, 2024-07-23:
The two DACs should always be captured in a synchronized fashion -- have you seen evidence that this isn't the case?

I'll add the snapshot functionality requested - if nothing else, it will mean you can snapshot the DAC without interrupting what the ADC pipeline is processing

jack-h, 2024-07-23:
Ah, I see I am mistaken -- the pair of DACs can currently only be captured via loopback if they are added together in firmware.
The ability to capture them individually really does require extra snapshot capability. That's annoying.

jack-h, 2024-07-23:
Compiling now...

jack-h, 2024-07-23:
Compiled but completely untested in `v7.3` branch

sr-cdf, 2024-07-24:
In [12]: r.adc_snapshot.plot_adc_snapshot()
2024-07-24 10:34:17,931 - souk_mkid_readout.blocks.block:localhost:common_adc_ss - INFO - plot_adc_snapshot is deprecated. Please use plot_snapshot
---------------------------------------------------------------------------
UnboundLocalError                         Traceback (most recent call last)
<ipython-input-12-a6394f48f0d9> in <module>
----> 1 r.adc_snapshot.plot_adc_snapshot()

~/py3venv/lib/python3.6/site-packages/souk_mkid_readout/blocks/adc_snapshot.py in plot_adc_snapshot(self, nsamples)
     66         """
     67         self.logger.info('plot_adc_snapshot is deprecated. Please use plot_snapshot')
---> 68         return self.plot_snapshot(nsamples=nsamples)
     69 
     70     def plot_snapshot(self, nsamples=None):

~/py3venv/lib/python3.6/site-packages/souk_mkid_readout/blocks/adc_snapshot.py in plot_snapshot(self, nsamples)
     77         from matplotlib import pyplot as plt
     78         x2d = np.atleast_2d(self.get_snapshot())
---> 79         for i in range(x.shape[0]):
     80             if nsamples is not None:
     81                 x = x2d[i][0:nsamples]

UnboundLocalError: local variable 'x' referenced before assignment

sr-cdf, 2024-07-24:
With the above typo corrected, I can plot the ADC snapshot using `adc_snapshot.plot_snapshot()`

However, plotting the DAC snapshot fails with: 
```
KatcpRequestFail: Request read on host localhost failed.
        Request: ?read common_dac_ss_ctrl 0 4
        Reply: !read fail
```

`common_dac_ss_ctrl` does not exist in `fgpa.listdev()`

sr-cdf, 2024-07-24:
If i rename the dac snapshot to `adc_ss1` then I can read it, only issue is the samples are out of order and I need to reverse every 8 samples:
[image]

Other than that it seems to be working pretty well :)

jack-h, 2024-07-24:
Ha. Ok, both those are easy to fix

jack-h, 2024-07-24:
import numpy as np, souk_mkid_readout

r = souk_mkid_readout.SoukMkidReadout('10.11.0.217', configfile='/home/jackh/src/souk-firmware2/souk-firmware/software/control_sw/config/souk-single-pipeline-krm-2G.yaml')

r.initialize()
r.output.use_psb()
r.set_tone(0, 1200000*4.623 + r.adc_clk_hz/2.)
r.dac_snapshot.plot_snapshot()
```
![image](https://github.com/user-attachments/assets/5c2280c0-75ec-424f-8102-a055c47c0fd1)

```
r.use_dual_dac()
r.dac_snapshot.plot_snapshot()
```

![image](https://github.com/user-attachments/assets/5d81aabe-8ced-44b6-b33c-74ec181a6e3b)

```
r.set_tone(0, 1200000*3.623 + r.adc_clk_hz/2.)
r.dac_snapshot.plot_snapshot()
```

![image](https://github.com/user-attachments/assets/d2e7ce74-ca65-42c8-8447-ca66254bae81)

```
r.set_tone(1, 1200000*4.623 + r.adc_clk_hz/2.)
r.dac_snapshot.plot_snapshot()
```
![image](https://github.com/user-attachments/assets/0c23c4ea-353a-4403-af56-07eb89a906df)

```
r.use_single_dac()
r.dac_snapshot.plot_snapshot()
```

![image](https://github.com/user-attachments/assets/6355a270-7049-439b-8479-084f87df9c33)

jack-h, 2024-07-24:
Think this is doing as it should now.
Tagging `v7.3.0.1` and merging to main

sr-cdf, 2024-07-24:
Awesome

closed 2024-07-24 by sr-cdf