realtimeradio/souk-firmware #61: timestamping

jack-h, 2024-08-13:
How does timestamp get returned with accumulators?

sr-cdf, 2024-09-13:
Looks like getting the time takes 0.2ms, load_internal_time takes 23ms, sw_sync takes 66ms
```
r.sync.arm_sync(wait=False)
fpga_clk_hz = int(r.rfdc.core.get_fabric_clk_freq(0,0)*1e6)
print(fpga_clk_hz)

ttnow = int( time.time() * fpga_clk_hz)
print(time.time()-ttnow/fpga_clk_hz)

r.sync.load_internal_time( ttnow )
print(time.time()-ttnow/fpga_clk_hz)

r.sync.sw_sync()
print(time.time()-ttnow/fpga_clk_hz)
print(time.time()-ttnow/fpga_clk_hz)

256000000
0.00023674964904785156
0.023051738739013672
0.08935379981994629
0.08971047401428223
```

jack-h, 2024-09-17:
Everything in your post is impressively correct, I believe.

With no PPS supplied, the accuracy of the internal telescope time is limited to how quickly various operations can be completed. What you're seeing -- a few tens of milliseconds -- seems plausible.

Provided the RFSoC is clocking from a GPS-locked reference (rather than the onboard XO) it shouldn't drift vs NTP.

The (my) typical load mechanism is to-be-implemented in https://github.com/realtimeradio/souk-firmware/blob/8dc9c36529073f3e2cf8b86f421a34ea664af0c8/software/control_sw/src/blocks/sync.py#L246

It amounts to:

1. Wait for a PPS to pass
2. Calculate the appropriate TT value at the next PPS
3. Arm sync logic to load this time on the next PPS
4. Check no PPS has passed yet, since step 1.
5. Wait for PPS

Provided steps 2&3 take less than 1 second  this should reliably lock TT to the PPS reference.

jack-h, 2025-07-08:
Present since aff82a6a via accumulator's `get_new_spectra(get_tt=True)` argument