realtimeradio/souk-firmware #62: Dual pipeline questions

sr-cdf, 2024-09-13:
The dual pipeline is working and I can see good DAC and ADC snapshots and good signals on the spectrum analyser on both pipelines.

Apparently `common.ninputs` can go up to 4, is this in case we ever put 4 pipelines on one board?

Given that it seems to be working fine at first glance, is the plan to progress with dual pipeline as the default?

Or does it take so long to compile that its more convenient to continue with the single pipeline? (While development is still ongoing.)

jack-h, 2024-09-17:
I think now development has slowed this is a good idea. I'll rebuild the
latest designs in dual pipeline mode.

common.ninputs of 4 is a bit of a hangover from the aspirational 4
pipelines per board days. As the name suggests, everything in the common
block is shared over pipelines, so really the limit is just the size of the
multiplexer that feeds the block (which currently has 4 inputs, 2 of which
are just repeats of the first input)

jack-h, 2024-09-17:
And now I've seen your edit. Good catch!

jack-h, 2024-09-17:
This is fixed and I've made the dual-pipeline firmware the same rev as the single in https://github.com/realtimeradio/souk-firmware/commit/183ed91c957a21b4667b2a7fb21e9d28859e48fe

closed 2024-09-17 by jack-h