realtimeradio/souk-firmware #64: CPU stops

sr-cdf, 2024-10-02:
Hi @jack-h, 

FYI

I can make the CPU stop at will in python by writing to the mixer lo phase increment shared bram using the axil-mmap (in casperfpga.transport_localmem.LocalMemTransport) and then reading from it straight away in the next line:

This code will crash the CPU:
```
(py3venv) casper@localhost:~$ sudo ~/py3venv/bin/ipython

import souk_mkid_readout
r = souk_mkid_readout.SoukMkidReadout('localhost',configfile='/home/casper/src/souk-firmware/software/control_sw/config/souk-single-pipeline-krm-2G.yaml',local=False)
r.program()
r.initialize()
r_fast = souk_mkid_readout.SoukMkidReadout('localhost',configfile='/home/casper/src/souk-firmware/software/control_sw/config/souk-single-pipeline-krm-2G.yaml',local=True)

i=0
ramname = f'{r.mixer.prefix}tx_lo{i}_phase_inc'  # 'p0_mix_tx_lo0_phase_inc': Sbram:p0_mix_tx_lo0_phase_inc,
phase_addrs_tx = []
phase_addrs_tx += [r_fast.mixer.host.transport._get_device_address(ramname)]

nbytes = r.mixer._n_serial_chans * 4
maxwrite=512
nwrite = (nbytes // maxwrite)

b0=bytes([0]*nbytes) # all 0x00
b1=bytes([255]*nbytes) # all 0xFF

#must write in 512 byte chunks?
for i in range(len(phase_addrs_tx)):
    for j in range(nwrite):
        print(i,j)
        raw1 = b1[j*maxwrite:(j+1)*maxwrite]
        r_fast.mixer.host.transport.axil_mm[phase_addrs_tx[i]+j*maxwrite : phase_addrs_tx[i] +(j+1)*maxwrite] = raw1
        ret1 = r_fast.mixer.host.transport.axil_mm[phase_addrs_tx[i]+j*maxwrite : phase_addrs_tx[i] +(j+1)*maxwrite]
        print(ret1==raw1)

# system prints 0,0 then freezes before the next print
```
There are no kernel messages in dmesg or journal or syslog, nothing printed to the serial konsole, does not respond over network or serial, just everything stops responding and needs a power cycle.

Note that the system does not crash if i put a time.sleep(0) between the read and the write.

What could be the cause of the crash?

jack-h, 2024-10-08:
@mitchburnett Don't suppose you have any words of wisdom? I don't know if this is fundamental, or some error in the way python is memmapping `/dev/mem`. I haven't investigated at all yet, but before I start I wondered if you had ever seen this.

mitchburnett, 2024-10-10:
I have seen the CPU stall when a memory access happens outside a range that the underlying controller is asked to go fetch. So, I actually would not it expect to be an issue with how Python is doing the mapping. It is most likely mapping fine doing everything it thinks it should. The requested address goes out to the controller and stalls waiting for a response. I have seen this in two cases.

The first would be when the available address space has been exhausted. To quickly rule this out I would just look at what the address it wants to go get and compare that to the slave segments in the Vivado address editor. I want to remember that the `M_AXI` interface in `jasper` only has a 256K range starting at `a000_0000` and going to `a003_ffff`. If the requested mixer address you are getting is above that, this could explain for the hang.

The other case is when the address is not accessible to the AXI FIFO interface at the Zynq/fabric boundary. Meaning the loaded Zynq parameters at boot (in the device tree) only has access to one interface like `M_AXI_HPM0_FPD` but the underlying bitstream that was programmed after boot uses two master interfaces and the requested address is accessible through `M_AXI_HPM1_FPD`. To access this at runtime requires new AXI FIFO interface parameters be written into the device tree. I don't suspect this is your issue though if you are still compiling through `jasper`.

If it isn't either of these two things... we have exhausted my wisdom...

sr-cdf, 2024-11-06:
The issue seems to be that the `mmap` in `casperfpga.transport_localmem.py` is mapping the entire system memory from 0x0 to 0xFFFFFFFF.

When a read is called immediately after a write, the system hangs, but not if there is a time.sleep(0) or print statement in between. True for both python and C implementations.

If the 'mmap' is changed to only cover the memory space of the FPGA device, eg 0xa000_000 to 0xa00f_ffff, it never hangs.

Any thoughts on why this might be?

sr-cdf, 2024-11-06:
A wideband sweep with original settings and large 4GB mmap starting at 0x0:

[image]

And again but with a 1MB mmap starting at 0xa000_0000:

[image]

sr-cdf, 2024-11-06:
And interestingly, in the case with the large mmap, where it does crash, it only seems to crash if the length of the write/read is smaller than the page size (4kB). So writing/reading in 4k chunks does actually work, whereas 512 byte chunks will crash it.

jack-h, 2024-11-06:
This is great sleuthing. I have no idea why the behaviour is as you observe -- I could see that if a page straddled the FPGA and system memory space something weird might occur, but that's not the case here.

sr-cdf, 2024-11-06:
So the fix is to add an offset in the mmap call, `axil_mm = mmap(...., offset = 0xa0000000)`, and then to subtract the same offset when accessing the brams. 

And for belt and braces, to set the MAP_SIZE to something like 1MB rather than 4GB.

Out of interest, setting the offset to 0x80000000 (1GB) or higher always prevents the crashes, lower than 0x70000000 always allows the crashes, so somewhere in between there is a memory address that is safe minimum. 

The overall map size doesnt make a difference.

Anyway, that's about as far as I'm going to be looking into it now, as interesting as the problem is.

jack-h, 2024-11-06:
Thanks -- I'll commit a fix to casperfpga

jack-h, 2024-11-08:
Fixed in PR #67

closed 2024-11-08 by jack-h