realtimeradio/souk-firmware #65: local transport mmap write does not complete

sr-cdf, 2024-10-02:
If i write all ones or all zeros to the mixer lo phase increment shared bram and read back after a time.sleep(0), i get back what i set.

But if i write back random bytes (which better represents the actual formatted lo phase increment values) i do not  always get back what i write.

Increasing the sleep time does help. 

The following loop helps but surprisingly, it can take many repetitions before the data is readback correctly.

1. write the bytes
2. sleep
3. read the bytes
4. break if bytes read == bytes written break
5. else go to 1

Some example code:

```
(py3venv) casper@localhost:~$ sudo ~/py3venv/bin/ipython

import souk_mkid_readout
import time
import numpy as np

r = souk_mkid_readout.SoukMkidReadout('localhost',configfile='/home/casper/src/souk-firmware/software/control_sw/config/souk-single-pipeline-krm-2G.yaml',local=False)
r.program()
r.initialize()
r_fast = souk_mkid_readout.SoukMkidReadout('localhost',configfile='/home/casper/src/souk-firmware/software/control_sw/config/souk-single-pipeline-krm-2G.yaml',local=True)

i=0
ramname = f'{r.mixer.prefix}tx_lo{i}_phase_inc'  # 'p0_mix_tx_lo0_phase_inc': Sbram:p0_mix_tx_lo0_phase_inc,
phase_addrs_tx = []
phase_addrs_tx += [r_fast.mixer.host.transport._get_device_address(ramname)]

nbytes = r.mixer._n_serial_chans * 4
maxwrite=512
nwrite = (nbytes // maxwrite)

b0=bytes([0]*nbytes) # all 0x00
b1=bytes([255]*nbytes) # all 0xFF
br=bytes(list(np.floor(np.random.uniform(0,256,nbytes)).astype(int)))

#must write in 512 byte chunks?
for i in range(len(phase_addrs_tx)):
    for j in range(nwrite):
        print(i,j)
        #write all 0x00
        raw0 = b0[j*maxwrite:(j+1)*maxwrite]
        r_fast.mixer.host.transport.axil_mm[phase_addrs_tx[i]+j*maxwrite : phase_addrs_tx[i] +(j+1)*maxwrite] = raw0
        time.sleep(0)
        ret0 = r_fast.mixer.host.transport.axil_mm[phase_addrs_tx[i]+j*maxwrite : phase_addrs_tx[i] +(j+1)*maxwrite]

        #write all 0xff
        raw1 = b1[j*maxwrite:(j+1)*maxwrite]
        r_fast.mixer.host.transport.axil_mm[phase_addrs_tx[i]+j*maxwrite : phase_addrs_tx[i] +(j+1)*maxwrite] = raw1
        time.sleep(0)
        ret1 = r_fast.mixer.host.transport.axil_mm[phase_addrs_tx[i]+j*maxwrite : phase_addrs_tx[i] +(j+1)*maxwrite]

        #write random bytes
        rawr = br[j*maxwrite:(j+1)*maxwrite]
        r_fast.mixer.host.transport.axil_mm[phase_addrs_tx[i]+j*maxwrite : phase_addrs_tx[i] +(j+1)*maxwrite] = rawr
        time.sleep(0)
        retr = r_fast.mixer.host.transport.axil_mm[phase_addrs_tx[i]+j*maxwrite : phase_addrs_tx[i] +(j+1)*maxwrite]

        print(ret0==raw0, ret1==raw1, retr==rawr)

```
The result is not always the same but generally there are some Falses in the third column:
```
0 0
True True False
0 1
True True True
0 2
True True True
0 3
True True True
0 4
True True False
0 5
True True True
0 6
True True True
0 7
True True True
0 8
True True True
0 9
True True True
0 10
True True True
0 11
True True True
0 12
True True True
0 13
True True True
0 14
True True False
0 15
True True True
```

sr-cdf, 2024-10-02:
If i write a ramp of 8bit ints, instead of random bytes, i see this:

```

bramp=bytes((nbytes//256)*list(np.arange(256).astype(int)))
for i in range(len(phase_addrs_tx)):
    for j in range(nwrite):
        print(i,j)
        #write all 0x00
        raw0 = b0[j*maxwrite:(j+1)*maxwrite]
        r_fast.mixer.host.transport.axil_mm[phase_addrs_tx[i]+j*maxwrite : phase_addrs_tx[i] +(j+1)*maxwrite] = raw0
        time.sleep(0)
        ret0 = r_fast.mixer.host.transport.axil_mm[phase_addrs_tx[i]+j*maxwrite : phase_addrs_tx[i] +(j+1)*maxwrite]

        #write all 0xff
        raw1 = b1[j*maxwrite:(j+1)*maxwrite]
        r_fast.mixer.host.transport.axil_mm[phase_addrs_tx[i]+j*maxwrite : phase_addrs_tx[i] +(j+1)*maxwrite] = raw1
        time.sleep(0)
        ret1 = r_fast.mixer.host.transport.axil_mm[phase_addrs_tx[i]+j*maxwrite : phase_addrs_tx[i] +(j+1)*maxwrite]

        #write random bytes
        rawramp = bramp[j*maxwrite:(j+1)*maxwrite]
        r_fast.mixer.host.transport.axil_mm[phase_addrs_tx[i]+j*maxwrite : phase_addrs_tx[i] +(j+1)*maxwrite] = rawramp
        time.sleep(0)
        retramp = r_fast.mixer.host.transport.axil_mm[phase_addrs_tx[i]+j*maxwrite : phase_addrs_tx[i] +(j+1)*maxwrite]
        if rawramp!=retramp:
            break
            
        print(ret0==raw0, ret1==raw1, retramp==rawramp)

raw_vals=np.frombuffer(rawramp,dtype=np.uint8)
ret_vals=np.frombuffer(retramp,dtype=np.uint8)

print(raw_vals)
print(ret_vals)
print(ret_vals!=raw_vals)

[  0   1   2   3   4   5   6   7   8   9  10  11  12  13  14  15  16  17
  18  19  20  21  22  23  24  25  26  27  28  29  30  31  32  33  34  35
  36  37  38  39  40  41  42  43  44  45  46  47  48  49  50  51  52  53
  54  55  56  57  58  59  60  61  62  63  64  65  66  67  68  69  70  71
  72  73  74  75  76  77  78  79  80  81  82  83  84  85  86  87  88  89
  90  91  92  93  94  95  96  97  98  99 100 101 102 103 104 105 106 107
 108 109 110 111 112 113 114 115 116 117 118 119 120 121 122 123 124 125
 126 127 128 129 130 131 132 133 134 135 136 137 138 139 140 141 142 143
 144 145 146 147 148 149 150 151 152 153 154 155 156 157 158 159 160 161
 162 163 164 165 166 167 168 169 170 171 172 173 174 175 176 177 178 179
 180 181 182 183 184 185 186 187 188 189 190 191 192 193 194 195 196 197
 198 199 200 201 202 203 204 205 206 207 208 209 210 211 212 213 214 215
 216 217 218 219 220 221 222 223 224 225 226 227 228 229 230 231 232 233
 234 235 236 237 238 239 240 241 242 243 244 245 246 247 248 249 250 251
 252 253 254 255   0   1   2   3   4   5   6   7   8   9  10  11  12  13
  14  15  16  17  18  19  20  21  22  23  24  25  26  27  28  29  30  31
  32  33  34  35  36  37  38  39  40  41  42  43  44  45  46  47  48  49
  50  51  52  53  54  55  56  57  58  59  60  61  62  63  64  65  66  67
  68  69  70  71  72  73  74  75  76  77  78  79  80  81  82  83  84  85
  86  87  88  89  90  91  92  93  94  95  96  97  98  99 100 101 102 103
 104 105 106 107 108 109 110 111 112 113 114 115 116 117 118 119 120 121
 122 123 124 125 126 127 128 129 130 131 132 133 134 135 136 137 138 139
 140 141 142 143 144 145 146 147 148 149 150 151 152 153 154 155 156 157
 158 159 160 161 162 163 164 165 166 167 168 169 170 171 172 173 174 175
 176 177 178 179 180 181 182 183 184 185 186 187 188 189 190 191 192 193
 194 195 196 197 198 199 200 201 202 203 204 205 206 207 208 209 210 211
 212 213 214 215 216 217 218 219 220 221 222 223 224 225 226 227 228 229
 230 231 232 233 234 235 236 237 238 239 240 241 242 243 244 245 246 247
 248 249 250 251 252 253 254 255]

[  0   1   2   3   4   5   6   7   8   9  10  11  12  13  14  15  16  17
  18  19  20  21  22  23  24  25  26  27  28  29  30  31  32  33  34  35
  36  37  38  39  40  41  42  43  44  45  46  47  48  49  50  51  52  53
[... 48 lines cut ...]
 False False False False False False False False False False False False
 False False False False False False False False False False False False
 False False False False False False False False False False False False
 False False False False False False False False False False False False
 False False False False False False False False False False False False
 False False False False False False False False False False False False
 False False False False False False False False False False False False
 False False False False False False False False False False False False
 False False False False False False False False False False False False
 False False False False False False False False False False False False
 False False False False False False False False False False False False
 False False False False False False False False False False False False
 False False False False False False False False False False False False
 False False False False False False False False False False False False
 False False False False False False False False False False False False
 False False False False False False False False False False False False
 False False False False False False False False False False False False
 False False False False False False False False False False False False
 False False False False False False False False False False False False
 False False False False False False False False False False False False
 False False False False False False False False False False False False
 False False False False False False False False]

```
Any thoughts?

sr-cdf, 2024-10-02:
Seems to be a python problem as the following c-code (which pretty much does the same thing) does not crash the cpu and always reads back the data that was written:

```
test_mmap.c 

#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include <sys/mman.h>
#include <sys/types.h>
#include <sys/stat.h>
#include <fcntl.h>
#include <string.h>

#define MAP_SIZE 0x100000000 // 4GB
#define ADDR 0xa00aa000
#define NBYTES 8192
#define BLOCK_SIZE 4096

int main() {
  int fd;
  unsigned char *axil_mm;
  unsigned char *data;
  int size, written, n_bytes, i;
  int mismatch = 0; // Flag to track  mismatches
  // Open /dev/mem
  fd = open("/dev/mem", O_RDWR | O_SYNC);
  if (fd == -1) {
    perror("open");
    exit(1);
  }

  // Create the memory mapping
  axil_mm = mmap(NULL, MAP_SIZE, PROT_READ | PROT_WRITE, MAP_SHARED, fd, 0);
  if (axil_mm == MAP_FAILED) {
    perror("mmap");
    exit(1);
  }

  // Allocate memory for data
  size = NBYTES; 
  data = malloc(size);
  if (data == NULL) {
    perror("malloc");
    exit(1);
  }

  // Generate the repeating pattern of 0-255
  for (i = 0; i < size; i++) {
    data[i] = i % 256;
  }

  // Write data to memory
  size = NBYTES;
  written = 0;
  for (i = 0; i < size / BLOCK_SIZE + 1; i++) {
    n_bytes = size - written < BLOCK_SIZE ? size - written : BLOCK_SIZE;
    if (n_bytes > 0) {
      memcpy(axil_mm + ADDR + written, data + written, n_bytes);
    }
    written += n_bytes;
  }

  // Read data from memory and compare
  printf("Data read from memory:\n");
  for (i = 0; i < size; i++) {
    printf("%03d ", axil_mm[ADDR + i]);
    if ((i + 1) % 16 == 0) {
      printf("\n");
    }
    if (axil_mm[ADDR + i] != data[i]) { // Check for mismatch
      mismatch = 1;
    }
  }
  printf("\n");

  // Report mismatches
  if (mismatch) {
    printf("Error: Mismatch detected between written and read data!\n");
  } else {
    printf("Success: All written bytes match read bytes.\n");
  }

  printf("\n");

  // Clean up
  free(data);
  munmap(axil_mm, MAP_SIZE);
  close(fd);

  return 0;
}

```

jack-h, 2024-10-08:
Does the C code also solve https://github.com/realtimeradio/souk-firmware/issues/64 ?

It also occurs to me that reading back the value and seeing it not update doesn't necessarily imply that the value didn't make it to the FPGA.
Since you use the same options to open the file and map its contents I don't have any immediate suggestions. Do you think it is worth abandoning the `LocalMemTransport` interface at this point (I assume the C will also be much faster for things like tone tracking)? Some time ago I added https://github.com/realtimeradio/souk-firmware/blob/main/software/control_sw/test_scripts/fpg2header.py to scrape the fpg file for memory addresses so you could access them by name.

jack-h, 2024-10-08:
Possible related (or at least a potentially useful comparison): https://github.com/realtimeradio/casperfpga/blob/c65b856531f7572ac8e43494499272e2948e498a/src/transport_localpcie.py#L55

sr-cdf, 2024-10-29:
> Does the C code also solve #64 ?

Yes and no. Yes, that C code does not immediately crash the CPU. But, if I remove the print statements from the read-loop, then it does :-/

> It also occurs to me that reading back the value and seeing it not update doesn't necessarily imply that the value didn't make it to the FPGA.

I was trying to find the source of some occasional glitches in the accumulator output streams, which lead me to notice these write issues. So I'm pretty sure that unwanted values are getting to the FPGA and the mixer.

It seems that even the C code occasionally reads back different values to what was written. Running it once it seemed fine, but after running it like 10 or 20 times it does report a mismatch. Usually just one 16 byte chunk, aligned to a 16 byte boundary reports back different to what was written. These can be rewritten immediately until the read back is correct.

jack-h, 2024-10-29:
interesting, this seems like there is some Linux/C-fu required to force some sort of non-caching / `volatile` behaviour.
I don't know if the reason this has never been seen on the remote interface is because it doesn't happen, or there is just such a long delay between write and read that things have propagated.
When the writes don't take, do they work eventually if you keep waiting and reading?

jack-h, 2024-10-29:
I'm not sure if stuff like this is of any relevance -- https://stackoverflow.com/questions/29102530/usage-of-volatile-in-case-of-memory-mapped-devices

sr-cdf, 2024-10-29:
Delaying after the write makes no difference. It looks like it immediately and successfully writes the wrong value.

`memcpy` doesnt work with `volatile` but replacing `memcpy` with byte-by-byte copies reduces the error rate from one 16 byte chunk every few runs to one or two consecutive 4 byte chunks every few hundred runs. So, cacheing of sorts does seem to be part of the problem there.

Specifying the mmap and other relevant variables as `volatile` doesn't make any difference.

I need to test the time taken to write all the values in C with re-writes when necessary. Things may be fast enough for us.

And let me figure out the time delay required to prevent the CPU stopping.

Does katcp do any validation of writes?

jack-h, 2024-10-29:
katcp itself doesn't, but by default the casperfpga client will read back any writes (there is actually a `blindwrite` command to avoid this, which is almost never used). I have never seen a write throw an error in the way you are seeing.

The underlying data bus on the FPGA is 32-bits wide, so I guess it doesn't surprise me that when things go wrong they are in 4-byte chunks. But I'd also be wary about writing anything other than 4-byte chunks at once (i.e., writing anything other than `volatile uint32_t`

sr-cdf, 2024-10-30:
I read the correct data back every time if I get rid of the O_SYNC option when opening /dev/mem. 

Need to check if the data is getting through to the FPGA though...

jack-h, 2024-10-31:
I'd worry that might be the ignorance-is-bliss option(!), but I'm interested to see what you see in hardware.

sr-cdf, 2024-11-06:
As with #64, setting the mmap to map only the FPGA register/bram memory space - as opposed to mapping the entire 4GB of system memory - fixes this.

I'm not going to pretend to understand how memory works with paging and cacheing, in user space, in the kernel or in the cpu, but at some point the data gets corrupted during the write process when the entire system memory is covered with the memory map, but not when only the FPGA devices are.

sr-cdf, 2024-11-06:
See comments on #64

jack-h, 2024-11-08:
Fixed in PR #67

closed 2024-11-08 by jack-h