realtimeradio/souk-firmware #70: Odd bins off by pi radians

sr-cdf, 2024-12-02:
Hi @jack-h,

I'm hoping there is a simple fix for this...

In the accumulated data, the phase of the complex values changes by pi radians when ever a tone crosses to the next filterbank channel. 

Not sure if this is on the Tx or Rx path.

It's simple enough to correct in software, but can it be corrected in firmware?

sr-cdf, 2024-12-02:
> *External email to Cardiff University - *Take care when replying/opening
> attachments or links.
> *Nid ebost mewnol o Brifysgol Caerdydd yw hwn - *Cymerwch ofal wrth
> ateb/agor atodiadau neu ddolenni.
>
> Do you know if this has always been a "feature"?

It's been with us at least since I started sweeping, possibly since the off.

sr-cdf, 2024-12-02:
The software correction is to shift the odd or even bin data by pi radians,
or multiply by exp(i pi), which is the same as multiplying real and imag
both by -1.

I guess the worst case scenario would be to add a set of multipliers to
either the odd or even bins going into the PSB or out of the PFB?

jack-h, 2024-12-03:
OK - this is easily fixed in the synth bank (where there is already a half-bin mix). I'll do a few tests and try to figure out if this is actually coming from the RX or TX path before I implement.

jack-h, 2025-01-18:
Some investigation notes for future me:

First, try to figure out whether the offset is incurred in the RX or TX sides. Simulation suggests the RX side shouldn't do this. but try to verify:

```
s = souk_mkid_readout.SoukMkidReadout('10.11.0.130')
s.program('./souk_single_pipeline_krm_2024-10-08_1526.fpg')
s.initialize()
s.input.enable_loopback()
a = s.accumulators[0]
freqs = np.array([5e6, 5.25e6, 5.5e6, 5.75e6, 6e6]) + s.adc_clk_hz/2
s.set_multi_tone(freqs)
s.sync.arm_sync(False); s.sync.sw_sync()
s.output.use_psb()

# Here's the reported pi radian offset
In[235]: print(np.angle(a.get_new_spectra()[0][0:5]))
[ 0.01563518 -3.12595365  0.01661626 -3.12399916  0.01857069]

# Now use the LUT generator to remove the PSB from the chain.
# Abuse the existing code to generate the waveforms
In [236]: for i, f in enumerate(freqs):
     ...:     s.gen_lut.set_output_freq(0, f + s.adc_clk_hz/2, sample_rate_hz=s.adc_clk_hz)
     ...:     if i == 0:
     ...:         x = s.gen_lut.get_lut_output(0)
     ...:     else:
     ...:         x += s.gen_lut.get_lut_output(0)
     ...: s.gen_lut.set_lut_output(0,x)

# But the LUT gen isn't sync'ed to anything so the phases are all over the place!
# Scan to find a good phase....

In [222]: out = []
     ...: for i in range(0,8192,30):
     ...:     s.gen_lut.set_lut_output(0,np.roll(x,i))
     ...:     a.get_new_spectra()
     ...:     phases = np.angle(a.get_new_spectra()[0][0:5])
     ...:     out += [phases]
     ...:     print(i)
     ...: outa = np.array(out)

# The following plots show zero of phase at approx 150 * 30 samples
In [223]: plt.plot(np.unwrap(outa[:,0] - outa[:,2]) % (2*np.pi)); plt.show()
In [224]: plt.plot(np.unwrap(outa[:,0] - outa[:,4]) % (2*np.pi)); plt.show()

# Set the phase of the input signal here
In [225]: s.gen_lut.set_lut_output(0,np.roll(x,150*30))

# Verify!
In [226]: print(np.angle(a.get_new_spectra()[0][0:5]))
[-0.92045939 -0.96634954 -1.01240325 -1.05852117 -1.10453843]

# And phase looks consistent, just a small residual delay in RX pipeline vs DDC generators
```

So, correct place to fix this issue is in the TX pipeline. Implementing now....

jack-h, 2025-01-18:
@sr-cdf Do you want this fix backported to v7.4 or can I just do it in 7.5?

sr-cdf, 2025-01-19:
Nicely done...

If it's not too much trouble, I still haven't had a go with 7.5 yet.

If it is too much trouble, I'll push on 7.5 sooner rather than later.

jack-h, 2025-01-19:
No probs. I'll verify the fix in the 7.5 branch as soon as it builds and then port it back

jack-h, 2025-01-19:
Fixed in v7.5.
v7.4 rebuild underway

jack-h, 2025-01-19:
And now fixed in v7.4 / main

closed 2025-03-19 by sr-cdf