realtimeradio/souk-firmware #72: v7.5 double buffered mixer.set_freqs not using buf offset

sr-cdf, 2025-03-13:
souk-firmware/software/control_sw/src/blocks/mixer.py

set_freqs method ignores the buf value, but replacing the last line with:

```
                offset = 4 * self._CONTROL_N_WORDS * (buf * self._n_serial_chans + i)
                self.write(reg, v.tobytes(),offset=offset)

```
seems to fix it. 

can you confirm this is the correct fix?

Full corrected version of set_freqs:

```
    def set_freqs(self, freqs_hz, phase_offsets, scaling=1.0, sample_rate_hz=2500000000, los=['rx', 'tx'], next_buf=False):
        """
        Configure the amplitudes, phases, and frequencies of multiple tones.

        :param freqs_hz: The frequencies, in Hz, to emit.
        :type freqs_hz: numpy.ndarray

        :param phase_offsets: The phase offsets at which oscillators should start,
            in units of radians.
        :type phase_offsets: np.ndarray

        :param scaling: optional scaling (<=1) to apply to the output tone
            amplitudes. If a single number, apply this scale to all tones.
        :type scaling: np.ndarray

        :param sample_rate_hz: DAC sample rate, in Hz
        :type sample_rate_hz: float

        :param los: List of LOs to write to. Can be ['rx'], ['tx'] or ['rx', 'tx']
        :type los: list

        :param next_buf: If False, write directly to the buffer currently being
            read by the firmware. If True, write to the next buffer, leaving
            the new parameters inactive until the buffer is switched
            (eg. by `switch_current_buffer`). If 0 or 1, write to that buffer.
        :type next_buf: bool or int

        """
        if next_buf in [0, 1]:
            buf = next_buf
        else:
            buf = self.get_current_buffer()
            if next_buf:
                buf = (buf + 1) % 2
        freqs_hz = np.array(freqs_hz, dtype=float)
        n_tone = len(freqs_hz)
        phase_offsets = np.array(phase_offsets, dtype=float)
        assert len(phase_offsets) == n_tone
        try:
            assert len(scaling) == n_tone
            scaling = np.array(scaling)
        except TypeError:
            scaling = scaling * np.ones(n_tone, dtype=float)
        
        fft_period_s = self._n_upstream_chans / self._upstream_oversample_factor / sample_rate_hz
        fft_rbw_hz = 1./fft_period_s # FFT channel width, Hz
        phase_steps = freqs_hz / fft_rbw_hz * 2 * np.pi
        ri_steps = np.cos(phase_steps) + 1j*np.sin(phase_steps)
        phase_steps, phase_offsets = self._format_phase_step(phase_steps, phase_offsets)
        scaling = self._format_amp_scale(scaling)
        ri_steps = [cplx2uint(ri_step, self._n_ri_step_bits) for ri_step in ri_steps]
        # format appropriately
        phase_steps_u = np.array(phase_steps, dtype='>i4').view('>u4')
        phase_offsets_u = np.array(phase_offsets, dtype='>i4').view('>u4')
        scaling_u = np.array(scaling, dtype='>u4')
        ri_steps_u = np.array(ri_steps, dtype='>u4')
        # create a new array to put all control variable into in parallel
        v = np.zeros(int(np.ceil(n_tone / self._n_parallel_chans)) * self._CONTROL_N_WORDS, dtype='>u4')
        for i in range(min(self._n_parallel_chans, n_tone)):
            v[self._SCALE_WORD_OFFSET :: self._CONTROL_N_WORDS] = scaling_u[i::self._n_parallel_chans]
            v[self._PHASE_INC_WORD_OFFSET :: self._CONTROL_N_WORDS] = phase_steps_u[i::self._n_parallel_chans]
            v[self._PHASE_OFFSET_WORD_OFFSET :: self._CONTROL_N_WORDS] = phase_offsets_u[i::self._n_parallel_chans]
            v[self._RI_STEP_WORD_OFFSET :: self._CONTROL_N_WORDS] = ri_steps_u[i::self._n_parallel_chans]
            for lo in los:
                if lo not in ['rx', 'tx']:
                    raise ValueError(f"Only LOs 'rx' and 'tx' are understood. Not {lo}.")
                reg = f'{lo}_lo{i}_control'
                offset = 4 * self._CONTROL_N_WORDS * (buf * self._n_serial_chans + i)
                self.write(reg, v.tobytes(),offset=offset)

```

jack-h, 2025-03-13:
hopefully hastily fixed

sr-cdf, 2025-03-14:
Sorted

closed 2025-03-14 by sr-cdf