realtimeradio/souk-firmware #74: Residual per-bin phase offset in PSB

sr-cdf, 2025-04-03:
@jack-h you mentioned a residual delay in [Odd bins off by pi radians #70](https://github.com/realtimeradio/souk-firmware/issues/70#issuecomment-2599756925)

This is actually something of a problem.

The phases still jump when stepping across PSB bins, albeit by less, but still enough to cause issues when analysing sweep data and tracking resonances.

As with the original issue, this can be calibrated out in post processing, but this is relatively CPU heavy, so a firmware fix is preferred.

It looks like the exact value of the offset required to zero the phase on any bin is B*2*pi/4096, where B is the bin index in the odd or even PSB. For example if the combined final bin indices are b=[-3,-2,-1,0,1,2,3] for the central 7 bins, corresponding to center frequencies of 1024e6 + [-750000,-500000,-250000,0,250000,500000,750000], then B = [-2,-1,-1,0,0,1,1]

Any thought on the source of the phase offsets, and how to correct it?

Some plots (using the older firmware version as i switched back to that in the proicess of debugging):

Raw sweep:

[image]

Zoom:

[image]

Odd bin flip:

[image]

Zoom:

[image]

Fixed:

[image]

Zoom:

[image]

And the vcariation is now close to zero, but note there are some residual junk around bins close to any sixteenth of the adc clock: 2048MHz /16 = 128MHz, which may be unrelated, but is also a problem that would be good to solve as it is enough to interfere with the resonator readout if the resonators hit these exact frequencies.

Code (v7.5 software and firmware):
```
import numpy as np
import time
import souk_mkid_readout
r = souk_mkid_readout.SoukMkidReadout(host, configfile='/home/casper/src/souk-firmware/software/control_sw/config/souk-single-pipeline-krm-2G.yaml')

#must ensure krc-utils is using the 0409M6 LMX clock

r.program('/home/casper/src/souk-firmware/firmware/src/souk_single_pipeline_krm/outputs/souk_single_pipeline_krm_2024-10-08_1526.fpg')

r.initialize()
r.sync.set_delay(6225)
r.sync.arm_sync(False); r.sync.sw_sync()
r.input.enable_loopback()
r.output.use_psb()
r.psb.set_fftshift(127);  r.psb.reset_overflow_count(); print(r.psb.get_overflow_count())
r.pfb.set_fftshift(127);  r.pfb.reset_overflow_count(); print(r.pfb.get_overflow_count())

f0 = np.linspace(-1000e6,1000e6,1000) + r.adc_clk_hz/2 +np.random.uniform(-1000,1000,1000)
amps=np.ones_like(f0)
phases=np.zeros_like(f0)
r.set_multi_tone(f0,phases,amps)

span=2e6
steps=101
sweep_offsets= np.linspace(-span/2,span/2,steps)
results=zeros((len(sweep_offsets),len(f0)),dtype=complex)

for i in range(len(sweep_offsets)):
    freqs = f0 + sweep_offsets[i]
    r.set_multi_tone(freqs, phases,amps)
    r.sync.arm_sync(False); r.sync.sw_sync()
    time.sleep(0.01)
    results[i]=r.accumulators[0].get_new_spectra()[0][0:1000]
    print(i,'/', steps)

np.save('results.npy',results)
```

jack-h, 2025-04-03:
Is this different to the phase slope which would indicate a residual skew between the RX and TX paths?
Specifically, is it correct that the group delay issue https://github.com/realtimeradio/souk-firmware/issues/37 is different to this, where in that case the residual phase is smooth but here it is stepped as

[image]

sr-cdf, 2025-04-03:
Yes, I think this is different.

The sync.sync_delay is effective in removing the skew between the Rx and Tx LOs such that across any single filterbank channel the slope of the accumulated phase vs frequency is exactly zero iff the sync_delay value == 6225. 

The issue here is that each PSB channel appears to add its own seperate delay/skew/phase-offset, varying linearly from precisely -pi to +pi radians at the lowest/highest bins and exactly zero in the central (DC) bin of each PSB (even and odd). The central bin of the odd PSB is not really DC as it is shifted but the phase offset in that bin is zero.

And

jack-h, 2025-04-03:
Sorry if I'm being dense, but if it is a phase per channel, why are there multiple steps in each coloured bin in the plots?

jack-h, 2025-04-03:
> Ah sorry, the colours change with tone, and each tone sweeps across many
> channels.

Ah ok, got it.

So, I have no idea what causes this, but I'll have a head scratch.

sr-cdf, 2025-04-03:
I could have a look at the design and try to model it in python, but it might help if you walked me through it.

Are you free for a call tomorrow?

jack-h, 2025-04-03:
Definitely. Free basically all day except 1130-1230

sr-cdf, 2025-04-03:
Morning time is good, I'll email you then

sr-cdf, 2025-04-04:
And confirming the same is true after the fix for issue #70.

Applying the same fix, don't need to flip the sign of the odd bins first:

[image]

Note the magntiude has increased by 6dB.

sr-cdf, 2025-04-04:
And with 32x more averaging, the offsets near multiples of Fs/8 are more clear:

[image]

zoom:

[image]

jack-h, 2025-04-04:
Perhaps a red herring, or maybe I haven't had enough coffee yet, but I can't see how the FFT stage 2 butterfly, which I believe should multiply by 1 for the first N/2 samples and -1 for the second N/2, does the right thing

[image]

This looks to me like half the time it takes `a+ib` to `a+ib` (multiplies by 1) but the other half it takes `a+ib` to `b-ia` (multiply by -i). I'll continue with our plan to compile the PSB without the FIR, but I'm also going to simulate this "optimised" stage 2 twiddle vs the normal general-stage twiddle to see if I'm going mad.

sr-cdf, 2025-04-04:
OK, are the details of the inner workings of the FFT are documented somewhere?

In v7.5.3.2 (single pipeline) the result is the same:

[image]

And again the same in the dual pipeline version.

[image]

So any updates can proceed from the latest commit.

jack-h, 2025-04-04:
> OK, are the details of the inner workings of the FFT are documented somewhere?

😆 

> So any updates can proceed from the latest commit.

Great, I'll let you know when I have something to run.

sr-cdf, 2025-04-04:
With the higher averaging, all the measured values are literally identical between the single and dual pipelines. 

I had to run it twice to make sure I took the data correctly!

jack-h, 2025-04-04:
> Perhaps a red herring, or maybe I haven't had enough coffee yet, but I can't see how the FFT stage 2 butterfly, which I believe should multiply by 1 for the first N/2 samples and -1 for the second [...]

I think this is me being dumb, and these are the right coefficients (the second stage draws two coefficients from _half_ a CW period)

jack-h, 2025-04-04:
Build sans FIR complete. Give https://github.com/realtimeradio/souk-firmware/commit/dbadb0e7023863d52ce2a916a7940aee65e9cc73 a go

sr-cdf, 2025-04-04:
So, dbadb0e gives more variation in magnitude across the channels (as expected) and there is a constant delay in phase with jumps at bin edges:

[image]

And if I remove the average group delay we see steps of -pi/2 radians at each bin edge:

[image]

And if I zoom out, we see a trend spanning from -pi to +pi at the band edges.

[image]

So, the likely culprit is somewhere in the FFT?

jack-h, 2025-04-04:
OK, that's something. Putting a tone in the center of a PFS bin is super easy to simulate (it's just a 1 cycle impulse into the FFT). So hopefully I can figure out what's going on.

Also, my (uncommented, probably useless) script for poking around the PFS is https://github.com/realtimeradio/souk-firmware/blob/v7.5-psb-phase-debugging/firmware/src/test_scripts/pfs_test.py

sr-cdf, 2025-04-04:
Thanks for the scipt, I'll take a look in a minute.

I'll look at tones in the PFB sans FIR bin centers now...

sr-cdf, 2025-04-04:
Looking at the already acquired data, with no FIR, when the tones (all with zero phase offset) are at the bin centers, the phases do not equal zero:

[image]

sr-cdf, 2025-04-04:
And they are not all the same value, but taking repeated samples for a bunch of tones, the phase is constant with time:

[image]

They are all flat.

sr-cdf, 2025-04-04:
But as a function of tone frequency:

[image]

sr-cdf, 2025-04-04:
The gradient is 175 nano-radians per hertz, which in time is -27.8 nano seconds

sr-cdf, 2025-04-04:
It's time for me to bail out for the weekend. Looking forward to continuing on Monday :)

jack-h, 2025-04-04:
Ok, thanks for all the info so far.
I'm going to tweak the DAC capture firmware/software so that it's possible to start a capture at a consistent point in an FFT window, in the hope that is easier to interpret....

sr-cdf, 2025-04-04:
Did you want to zoom?

sr-cdf, 2025-04-08:
I'm seeing similar behaviour in simulation when reversing fft output ordering to flip spectra, when in fact, negating the imaginary component (with np.ifft instead of np.fft) seems the way to go:

I know we showed that the FIR filter did not affect this, I think techincally its not the FIR that is doing it, its the FFT ordering:
```
from numpy import *
from matplotlib.pyplot import *; ion()
from scipy import signal

for mode in ['flip_fft','negate_cosine']:
    #system setup
    dac_clk=4096e6
    dac_period = 1/dac_clk
    dac_interpolation = 2
    duc_bw = dac_clk/dac_interpolation
    duc_period = 1./duc_bw
    duc_center = 1024e6

    #tones setup
    random.seed(1)
    many_tones=True
    if many_tones:
        num_tones=1000
        tones_bb = linspace(-1000e6,1000e6,num_tones) + random.uniform(-1000,1000,num_tones)
    else:
        tones_bb = array([1.0e6])
        num_tones=len(tones_bb)
    tones_rf = tones_bb + duc_center
    amps = ones(num_tones,dtype=float)
    phases = zeros(num_tones,dtype=float)

    # #set first quarter of tones amplitudes low to check order
    # if many_tones:
    #     amps[:len(amps)//4]*=0.5

    #psb fft setup
    psb_fft_len = int(2**13)
    psb_fft_freqs = fft.fftfreq(psb_fft_len,duc_period)

    #fir filter setup
    psb_fir_window = lambda x: signal.windows.dpss(x,2.0)
    # psb_fir_window = ones
    # psb_fir_window = zeros
    psb_fir_taps = 8
    # psb_os_factor = 2 # not used?

    def get_coeffs(ntaps, nfft, window_func=np.hanning):
        trange = np.linspace(-ntaps/2., ntaps/2., ntaps*nfft)
        sinc = np.sinc(trange)
        window = window_func(ntaps*nfft)
        coeffs = sinc * window
        # coeffs=window
        return coeffs

    fir_coeffs = get_coeffs(psb_fir_taps,psb_fft_len, psb_fir_window)

    #psb input, canonical frequency ordering
    psb_chans = np.round(tones_bb/duc_bw*psb_fft_len).astype(int)
    psb_input_spec = zeros(psb_fft_len,dtype=complex)
    psb_input_spec[psb_chans]=amps*np.exp(1j*phases)

    # run through
    nruns = 1

    psb_fft_in = array([psb_input_spec for _ in range((nruns+1)*psb_fir_taps)]) #frequncy domain constant input
    if mode =='flip_fft':
        psb_fft_out = array([fft.fft(i) for i in psb_fft_in]).ravel()
    elif mode =='negate_cosine':
        psb_fft_out = array([fft.ifft(i) for i in psb_fft_in]).ravel() # time domain output, ifft flips sign of imaginary part

    psb_fir_out = zeros(psb_fft_len*nruns,dtype=complex)
    for n in range(nruns):
        psb_fir_in = psb_fft_out[n*psb_fft_len*psb_fir_taps:(n+1)*psb_fft_len*psb_fir_taps] #ntaps*nfft
        #weighted overlap add (wola)
        weights = psb_fir_in*fir_coeffs
        overlap = weights.reshape((psb_fir_taps,psb_fft_len))
        add = sum(overlap,axis=0)
        if mode == 'flip_fft':
            add=add[::-1] # reverse order to flip spectra
        elif mode == 'negate_cosine':
            pass
        psb_fir_out[n*psb_fft_len:(n+1)*psb_fft_len] = add

    fig,(s1,s2)=subplots(2,1,sharex=True)

    s1.plot(fft.fftfreq(len(psb_fir_in),duc_period),abs(fft.fft(psb_fir_in)))
    s2.plot(fft.fftfreq(len(psb_fir_out),duc_period),abs(fft.fft(psb_fir_out)))

    s1.title.set_text(mode)

    fig,(s1,s2) = subplots(2,1,sharex=True); s1.grid(True); s2.grid(True)
    s1.set_ylabel('Mag dB'); s2.set_ylabel('Phase rad'); s2.set_xlabel('Frequency Hz')
    s1.plot(psb_fft_freqs[psb_chans],20*log10(abs(fft.fft(psb_fir_in[:psb_fft_len])))[psb_chans],'o')
    s1.plot(psb_fft_freqs[psb_chans],20*log10(abs(fft.fft(add)))[psb_chans],'o')
    s2.plot(psb_fft_freqs[psb_chans],angle(fft.fft(psb_fir_in[:psb_fft_len]))[psb_chans],'o',label='fir input')
    s2.plot(psb_fft_freqs[psb_chans],angle(fft.fft(add))[psb_chans],'o',label='fir output')
    s1.title.set_text(mode)
    s2.legend()

```

[image]

[image]

How are you getting on with the DAC snapshot?

sr-cdf, 2025-04-08:
Apologies, any references to 'cosine' in the above should in fact be 'sine'. Is the imaginary part that should be negated.

jack-h, 2025-04-08:
You HERO.
I have the DAC snapshot firmware compiled, but the start of this week I'm travelling so won't get a chance to test it for a bit.
I'll compiled in your suggestion above, too.

jack-h, 2025-04-12:
To summarize, the issue is an incorrect implementation of an iFFT.

I thought that this could be accomplished by a simple flip of the FFT output samples, but this isn't the case.

```
import numpy as np
from matplotlib import pyplot as plt

N = 32
v = np.random.rand(N) + 1j*np.random.rand(N)

 def plot(v):
     plt.subplot(1,2,1)
     plt.plot(np.fft.fft(v).real, label='fft')
     plt.plot(np.roll(np.fft.fft(v).real[::-1], 1), label='roll(flip(fft))')
     plt.plot(np.fft.fft(np.conjugate(v)).real, label='conj(input)')
     plt.plot(np.conj(np.fft.fft(v)).real, label='conj(output)')
     plt.plot(np.fft.fft(v).real[::-1], label='flip(fft)')
     plt.plot(np.conj(np.fft.fft(np.conj(v))).real, label='conj(input), conj(output)')
     plt.plot(N*np.fft.ifft(v).real, 'o--', label='ifft')
 
     plt.subplot(1,2,2)
     plt.plot(np.fft.fft(v).imag, label='fft')
     plt.plot(np.roll(np.fft.fft(v).imag[::-1], 1), label='roll(flip(fft))')
     plt.plot(np.fft.fft(np.conjugate(v)).imag, label='conj(input)')
     plt.plot(np.conj(np.fft.fft(v)).imag, label='conj(output)')
     plt.plot(np.fft.fft(v).imag[::-1], label='flip(fft)')
     plt.plot(np.conj(np.fft.fft(np.conj(v))).imag, label='conj(input), conj(output)')
     plt.plot(N*np.fft.ifft(v).imag, 'o--', label='ifft')
 
     plt.legend()
     plt.show()

plot(v)
```

yields:

[image]

I.e., the two ways to obtain an ifft with an fft are
  -  to flip and roll, such that the first sample stays the same and the other samples flip (i.e., [0,1,2,3] goes to [0,3,2,1])
  -  to conjugate both input and outputs of the FFT.

Simply flipping (but not rolling) leads to an off-by-one sample error in the output.

I am changing the firmware to implement two conjugations before and after the FFT.

jack-h, 2025-04-13:
@sr-cdf -- have a try with dual pipeline firmware from https://github.com/realtimeradio/souk-firmware/commit/9f67d96d0141c4b23d02da066e1938d991139bda

If this works I'll rebuild the single pipeline and merge this into the v7.5 branch

sr-cdf, 2025-04-13:
Awesome, I'll check it out tomorrow.

Probably a bit late for this now but I just noticed the following, which may be preferable for saving resources:

[image]

jack-h, 2025-04-14:
Clearly I need to spend more time on Wikipedia!
I'll check that out and build it into a future version since it is cheaper
(though conjugating is already much cheaper than the flip which was
previously there)

jack-h, 2025-04-16:
The scaling will change with the removal of the FIR (the FIR  attenuates power, I think by ~6dB) , but the power shouldn't be different between different FFT implementations without the FIR.

Can you try v7.5.5 I have just pushed to `v7.5-psb-phase-debugging` (firmware `2025-04-14_1205`) which includes the FIR and uses the real/imag flip.

sr-cdf, 2025-04-16:
The scaling issue may have just been my choice of fft shift values.

sr-cdf, 2025-04-16:
The magnitude is now skewed, where it had been perfectly smooth and symmetrical before:

[image]

jack-h, 2025-04-16:
Will investigate....

jack-h, 2025-04-17:
I definitely introduced a data vs sync misalignment when switching to the real/imag flip. Just pushed new firmware which fixes this (though stupidly I forgot to increment the bugfix number) so it is still `v7.5.5.3`. Is that any better?

sr-cdf, 2025-04-17:
I'm getting 

`RuntimeError: progremote request(Request to client localhost failed.) on host localhost failed`
when i try to program the device with any firmware.

Will try rebooting...

sr-cdf, 2025-04-17:
Rebooting fixed it

sr-cdf, 2025-04-17:
That fixed it, but still seeing the FFT channel shape as if there was no FIR. 

[image]

sr-cdf, 2025-04-17:
And the phase is still wiggling around when you remove the overall delay:

[image] 

As it did before, so probably aliasing that will go away with a single tone measurement.

jack-h, 2025-04-17:
The FIR is definitely in the design source.  I'll dig around in the compiled products to see if something crazy has happened somewhere. I assume the dual pipeline behaves in the same way?

sr-cdf, 2025-04-17:
> I assume the dual pipeline behaves in the same way?
I havent tried on this latest commit but it does on the commit before, 7.5.5.3

jack-h, 2025-04-17:
I can see all the PFB logic in the compiled design, so I'm very confused.
I'm now out of the office until after Easter, but I'll try to get to the bottom of this as soon as I can.

sr-cdf, 2025-04-17:
Confirmed that dual pipeline looks the same.

And sanity check that something has definitely changed... exactly the same measurement script on 7.5.3.2 (pre-phase-debugging) shows the nice magnitude response (power):

[image]

sr-cdf, 2025-04-17:
I'm also away for a week, but I'll be checking emails intermittently.

Catch you later

sr-cdf, 2025-04-29:
Should these conjugates be after the FFT and before the FIR?

[image]

I'll share the code in moment...

jack-h, 2025-04-29:
I think the two locations are equivalent since the PFB is linear with
pure-real coefficients....

sr-cdf, 2025-04-29:
That makes sense.

On the RFSoC i run this:

```
import souk_mkid_readout
r = souk_mkid_readout.SoukMkidReadout('localhost', configfile='/home/casper/src/souk-firmware/software/control_sw/config/souk-single-pipeline-krm-2G.yaml')
r.initialize()
r.sync.set_delay(6225)
r.sync.arm_sync(False); r.sync.sw_sync()
r.input.enable_loopback()
r.output.use_psb()
r.psb.set_fftshift(511);  r.psb.reset_overflow_count(); print(r.psb.get_overflow_count())
r.pfb.set_fftshift(15);  r.pfb.reset_overflow_count(); print(r.pfb.get_overflow_count())

f0 = np.linspace(-1000e6,1000e6,1000) + r.adc_clk_hz/2 +np.random.uniform(-1000,1000,1000)
amps=np.ones_like(f0)
phases=np.zeros_like(f0)
r.set_multi_tone(f0,phases,amps)
r.sync.arm_sync(False); r.sync.sw_sync();time.sleep(0.01)

span=2e6
steps=101
sweep_offsets= np.linspace(-span/2,span/2,steps)
results=zeros((len(sweep_offsets),len(f0)),dtype=complex)

for i in range(len(sweep_offsets)):
    freqs = f0 + sweep_offsets[i]
    r.set_multi_tone(freqs, phases,amps)
    r.sync.arm_sync(False); r.sync.sw_sync()
    time.sleep(0.01)
    results[i]=r.accumulators[0].get_new_spectra()[0][0:1000]
    print(i,'/', steps)

np.save('results.npy',results)
np.save('f0.npy',f0)

```

Then scp the results and frequencies over to my server and run this on the server:

```
a=np.load('results.npy')
f0=np.load('f0.npy')
span=2e6
steps=101
sweep_offsets= np.linspace(-span/2,span/2,steps)
af=np.array([ f0 + sweep_offsets[i] for i in range(len(sweep_offsets))])

#oneliner to remove the average phase slope 
aa=((a.T.ravel()*np.exp(-1j*af.T.ravel()*np.mean(np.diff(np.unwrap(np.angle(a.T.ravel())))/np.diff(af.T.ravel()))))).reshape(a.T.shape).swapaxes(0,1)

#plot fixed results
fig,(s1,s2)=plt.subplots(2,1,sharex=True); s1.grid(True); s2.grid(True)
s1.plot(af,20*np.log10(np.abs((a))))
#s2.plot(af,np.angle(a))
s2.plot(af,np.angle(aa))
s1.set_ylabel('Mag dB')
s2.set_ylabel('Phase rad')
s1.axvline(1024e6,color='k')
s2.axvline(1024e6,color='k')
#l1=[s1.axvline(1024e6+i,color='k',alpha=0.5,ls='--',zorder=0) for i in np.arange(-32e6+125000,32e6+125000,250000)]
#l2=[s2.axvline(1024e6+i,color='k',alpha=0.5,ls='--',zorder=0) for i in np.arange(-32e6+125000,32e6+125000,250000)]
#l11=[s1.axvline(1024e6+i,color='k',alpha=0.5,ls='--',zorder=0) for i in np.arange(-32e6,32e6,250000)]
#l22=[s2.axvline(1024e6+i,color='k',alpha=0.5,ls='--',zorder=0) for i in np.arange(-32e6,32e6,250000)]
#s1.set_xlim(1022.5e6,1025.5e6)
fig.suptitle('')
plt.tight_layout()
```

jack-h, 2025-04-29:
> I'm seeing similar behaviour in simulation when reversing fft output ordering to flip spectra, when in fact, negating the imaginary component (with np.ifft instead of np.fft) seems the way to go:
>
> I [...]

Looking at this more things don't seem quite as simple as I thought.

In your code for IFFT vs flipped FFT, the phase slope goes away if I use either the IFFT or the FFT, so I don't think the underlying issue is as simple as a slight iFFT mis-implementation.
In the firmware, and simulations of it, the order of operations is, forward-FFT, FIR, spectrum-flip. If I remove the final flip, the FIR doesn't do what it should, regardless of whether the upstream transform is a forward or backward FFT.
Eg, in the simulation, if I replace the [FFT with an IFFT](https://github.com/realtimeradio/souk-firmware/blob/e6b4cf4635ece113825bc286bf5d8db74f4a84a0/firmware/src/test_scripts/pfs_test.py#L37) and remove the [flip ](https://github.com/realtimeradio/souk-firmware/blob/e6b4cf4635ece113825bc286bf5d8db74f4a84a0/firmware/src/test_scripts/pfs_test.py#L50) then the synthesis bank doesn't work -- I can see discontinuities in the time stream.

I think I've got to the point where dConfusion / dt is positive, so I'm going to let this percolate for a bit.

But one thing I think I'm now sure of is that the issue is a real DSP problem, not some bizarre compile issue.

sr-cdf, 2025-04-30:
It's very late and I'm over-tired so dConfusion/dt is tending to infinity, lol, but it seems like your code is flipping the time domain samples after the FFT in 1024 point chunks, so each chunk of the output waveform is just played backwards, leading to apparent discontinuities after every 1024 samples.

Wikipedia says the iFFT(on frequency domain samples) operation == 1/N FFT(on reversed frequency domain samples (excluding DC)), not iFFT == reversed(FFT).

I'm probably missing the point, but I will take a closer look when I'm more alert in the morning.

sr-cdf, 2025-04-30:
What is the frequency ordering of the upstream data into the PSB?

The feed into the PSB FFT (or iFFT) should be in canonical frequency order (0, 1, ... N/2-1, -N/2, ..., -1)

sr-cdf, 2025-04-30:
> The FFT is configured as it would be in the time domain -- i.e with inputs running successive time samples and the output in frequency domain order you describe.

I dont think the FFT really cares what goes into it, but I do think we need to be careful about the order that we put things in.

Can we ensure the first sample into the FFT corresponds to channel 0, then 1, upto N/2-1, and then minus N/2, up to minus 1?

jack-h, 2025-04-30:
This seems to be the order used -- the input bin is chosen based on `np.fft.fftfreq` which reflects the canonical order. See https://github.com/realtimeradio/souk-firmware/blob/e6b4cf4635ece113825bc286bf5d8db74f4a84a0/software/control_sw/src/souk_mkid_readout.py#L483

sr-cdf, 2025-04-30:
OK, so thats how we decide which bin to choose for a given tone frequency. 

```
In [11]: [r._get_closest_psb_bin(i) for i in np.linspace(0,2048e6,10)]
Out[11]: [4096, 5006, 5916, 6827, 7737, 455, 1365, 2276, 3186, 4095]

```
The most negative frequency (-N/2) is at bin 4096 and the most positive (+N/2-1) is at 4095. Thats good.

Does the value at bin 0 feed into the FFT on the first clock of the FFT, and then the value at bin 1 etc ?

jack-h, 2025-04-30:
> Does the value at bin 0 feed into the FFT on the first clock of the FFT, and then the value at bin 1 etc ?

Exactly this

sr-cdf, 2025-04-30:
Even my browser is struggling to keep up with this, i'll post again...

I'm glad I understand.

So, the output of the FFT (which outputs in canonical order) should be consecutive samples representing the time domain waveform running from earliest time to latest time.

sr-cdf, 2025-04-30:
> So, the output of the FFT (which outputs in canonical order) should be consecutive samples representing the time domain waveform running from earliest time to latest time.

No, that is wrong, if the forward FFT operates on data that has been 'fftshifted' then the result is backwards in frequency.

The correct statement is:

The output of an Inverse-FFT which takes a canonically ordered input (representing a frequency domain signal) should be consecutive samples representing the time domain waveform running from earliest time to latest time.

sr-cdf, 2025-04-30:
**conj(FIR(FFT(conj(x))) != FIR(conj(FFT(conj(x))))**

sr-cdf, 2025-04-30:
Conclusion from today (all tones at bin centers):

The correct approach is to take the canonically ordered input signal, do an inverse FFT, and do the FIR.

A forward FFT instead of an IFFT will flip the final spectrum because the input is canonically ordered.

Test for tomorrow: Try again with tones offset from bin centers.

python code from today:

```#sim.py

"""
model psb:

input is frequency domain
runs through inverse fft to get into the time domain
runs through fir
spits out waveform

if inverse fft is replaced by fft the spectrum of the output waveform is reversed (the imaginary part of the time domain signal in negated), because the input is in canonical order

"""

from numpy import *
from matplotlib.pyplot import *; ion()
from scipy import signal

#clock setup
dac_clk=4096e6
dac_period = 1/dac_clk
dac_interpolation = 2
duc_bw = dac_clk/dac_interpolation
duc_period = 1./duc_bw
duc_center = 1024e6

#psb setup
#fft
psb_fft_len = int(2**13)
psb_fft_freqs = fft.fftfreq(psb_fft_len,duc_period)
#fir
psb_fir_window = lambda x: signal.windows.dpss(x,2.0,sym=False)
psb_fir_taps = 8
# psb_os_factor = 2 # not used?
def get_coeffs(ntaps, nfft, window_func=np.hanning):
    trange = np.linspace(-ntaps/2., ntaps/2., ntaps*nfft)
    sinc = np.sinc(trange)
    window = window_func(ntaps*nfft)
    coeffs = sinc * window
    return coeffs
fir_coeffs = get_coeffs(psb_fir_taps,psb_fft_len, psb_fir_window)

#tones setup
random.seed(1)
num_tones=100
tones_bb = linspace(-1000e6,1000e6,num_tones) + random.uniform(-1000,1000,num_tones)
tones_rf = tones_bb + duc_center
phases = zeros(num_tones,dtype=float)
# amps = ones(num_tones,dtype=float)

### set amps to increase with frequency so we spot them i the spectra
amps=(linspace(0.01,0.99,num_tones))

### plot the desired tones
fig,(s1,s2)=subplots(2,1,sharex=True)
fig.suptitle('Desired tones')
s1.plot(tones_bb,amps,'x')
s2.plot(tones_bb,phases,'x')
s1.set_ylabel('abs')
s2.set_ylabel('phase')
[... 105 lines cut ...]
s2.set_ylabel('real/imag')
s1.set_xlabel('fir input sample index')
s2.set_xlabel('fir output sample index')
fig.tight_layout()

###plot the spectra of the inputs and outputs for the FIR
sfin=fft.fftshift(fft.fftfreq(len(fir_in_all),1./2048e6))
szin=fft.fftshift(fft.fft(fir_in_all))
sfout=fft.fftshift(fft.fftfreq(len(fir_out_all),1./2048e6))
szout=fft.fftshift(fft.fft(fir_out_all))
szin[abs(szin)<1e-9]=1e-9+0j #rounding errors again
fig,(s1,s2)=subplots(2,1,sharex=True)
fig.suptitle('FIR input/output signal (frequency domain representation)')
s1.plot(sfin,abs(szin),label='fir in')
s1.plot(sfout,abs(szout),label='fir out')
s2.plot(sfin,angle(szin),label='fir in')
s2.plot(sfout,angle(szout),label='fir out')
s1.legend()
s2.legend()
s1.set_ylabel('abs')
s2.set_ylabel('phase')
s2.set_xlabel('baseband frequency')
fig.tight_layout()

```

[32 more replies]